nmakel/solaredge_modbus #1: Meter registers

bizonxx, 2020-06-18:
Hi,

I would like to read connected Meter registers. Is it possible?

nmakel, 2020-06-18:
I don't have a solaredge meter to test things with, but I've made a start with implementing a meter class. Could you please grab the latest commit from the new meters branch and check whether what's there works?

I'm interested in read_all()'s output from the meter class. You'll want to do the following, assuming you've connected to your inverter named "inv":

```
meters = inv.meters()

for m, mo in meters.items():
    print(m, mo, mo.read_all())
```

Please let me know what the output is, or the error you get.

bizonxx, 2020-06-18:
Not sure I'm doing it corectly but :

```
    inverter = solaredge_modbus.Inverter(
        host=args.host,
        port=args.port,
        timeout=args.timeout,
        unit=args.unit
    )

    meters = inverter.meters()

    for m, mo in meters.items():
       print(m, mo, mo.read_all())
```
```
root@OpenWrt:~/solaredge_modbus# python3 example.py 192.168.1.119 1502
Traceback (most recent call last):
  File "example.py", line 27, in <module>
    print(m, mo, mo.read_all())
  File "/root/solaredge_modbus/solaredge_modbus/__init__.py", line 275, in read_all
    return self._read_all(registers)
  File "/root/solaredge_modbus/solaredge_modbus/__init__.py", line 218, in _read_all
    v_addr = v[0]
KeyError: 0
```

nmakel, 2020-06-18:
Absolutely. I've just submitted another commit that should get you beyond this error.

bizonxx, 2020-06-18:
```
root@OpenWrt:~/solaredge_modbus# python3 example.py 192.168.1.119 1502
1 Meter1(192.168.1.119:1502, connectionType.TCP: timeout=1, unit=0x1) {'c_model': 'PRO380-Mod', 'c_option': 'Export+Import', 'c_version': '2.19', 'c_serialnumber': '19032098', 'c_deviceaddress': 1, 'c_sunspec_did': 203}
```

nmakel, 2020-06-18:
Great! Thanks.

Looks like the mechanism works. I'll need to add the remaining registers so you can try them out.

nmakel, 2020-06-18:
Ok, so I've populated the registers for meter1, and I _think_ I've got the details down right. Could you please run it again and dump the output here?

bizonxx, 2020-06-18:
```
root@OpenWrt:~/solaredge_modbus# python3 example.py 192.168.1.119 1502
1 Meter1(192.168.1.119:1502, connectionType.TCP: timeout=1, unit=0x1)
Traceback (most recent call last):
  File "example.py", line 28, in <module>
    print(f"\t{mo.read_all()}")
  File "/root/solaredge_modbus/solaredge_modbus/__init__.py", line 283, in read_all
    return self._read_all(registers)
  File "/root/solaredge_modbus/solaredge_modbus/__init__.py", line 248, in _read_all
    data = self._read_holding_registers(offset, length)
  File "/root/solaredge_modbus/solaredge_modbus/__init__.py", line 180, in _read_holding_registers
    result = self.client.read_holding_registers(address=address, count=length, unit=self.unit)
  File "/usr/lib/python3.7/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/lib/python3.7/site-packages/pymodbus/client/sync.py", line 107, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.119:1502)]
```

It looks like it is too many registers at ones or something else.
When I removed registers block begining from 'export_energy_apparent' it works:

```
root@OpenWrt:~/solaredge_modbus# python3 example.py 192.168.1.119 1502
1 Meter1(192.168.1.119:1502, connectionType.TCP: timeout=1, unit=0x1)
        {'c_model': 'PRO380-Mod', 'c_option': 'Export+Import', 'c_version': '2.19', 'c_serialnumber': '19032098', 'c_deviceaddress': 1, 'c_sunspec_did': 203, 'current': 42, 'p1_current': 8, 'p2_current': 31, 'p3_current': 2, 'current_scale': -1, 'voltage_ln': 23910, 'p1n_voltage': 23910, 'p2n_voltage': 23930, 'p3n_voltage': 24010, 'voltage_ll': 0, 'p12_voltage': 0, 'p23_voltage': 0, 'p31_voltage': 0, 'voltage_scale': -2, 'frequency': 5000, 'frequency_scale': -2, 'power': -956, 'p1_power': -180, 'p2_power': -745, 'p3_power': -31, 'power_scale': 0, 'power_apparent': 964, 'p1_power_apparent': 186, 'p2_power_apparent': 745, 'p3_power_apparent': 59, 'power_apparent_scale': 0, 'power_reactive': -129, 'p1_power_reactive': -49, 'p2_power_reactive': -29, 'p3_power_reactive': -51, 'power_reactive_scale': 0, 'power_factor': 8299, 'p1_power_factor': 9599, 'p2_power_factor': 10000, 'p3_power_factor': 5299, 'power_factor_scale': -2, 'export_energy_active': 6599, 'p1_export_energy_active': 2900, 'p2_export_energy_active': 2220, 'p3_export_energy_active': 2310, 'import_energy_active': 9850, 'p1_import_energy_active': 4590, 'p2_import_energy_active': 4910, 'p3_import_energy_active': 1170, 'energy_active_scale': 0}
```

nmakel, 2020-06-18:
Odd. I've seen that happen with other types of kwh meters, but SolarEdge hasn't documented a maximum number of registers that can be polled at once. Can you check at which point the error occurs? Can you add any registers beyond export_energy_apparent?

bizonxx, 2020-06-18:
This one is passed:

`"import_energy_reactive_q1": (0x9d44, 2, registerType.HOLDING, registerDataType.UINT32, int, "Total Imported Energy (Reactive) Quadrant 1", "VARh")`

nmakel, 2020-06-18:
Ok, at import_energy_reactive_q1 the connection gets refused? Could very well be that the quadrant values are not supported by non-SolarEdge meters. I'll comment those out for the time being.

I've cleaned things up a bit, please let me know if there's any remaining issues.

bizonxx, 2020-06-18:
```
root@OpenWrt:~/solaredge_modbus# python3 example.py 192.168.1.119 1502
Meter1 Meter1(192.168.1.119:1502, connectionType.TCP: timeout=1, unit=0x1)
Traceback (most recent call last):
  File "example.py", line 28, in <module>
    print(f"\t{mo.read_all()}")
  File "/root/solaredge_modbus/solaredge_modbus/__init__.py", line 289, in read_all
    return self._read_all(registers)
  File "/root/solaredge_modbus/solaredge_modbus/__init__.py", line 266, in _read_all
    results[k] = self._decode_value(data, length, dtype, vtype)
  File "/root/solaredge_modbus/solaredge_modbus/__init__.py", line 200, in _decode_value
    decoded = data.decode_16bit_int()
  File "/usr/lib/python3.7/site-packages/pymodbus/payload.py", line 439, in decode_16bit_int
    return unpack(fstring, handle)[0]
struct.error: unpack requires a buffer of 2 bytes
```

bizonxx, 2020-06-18:
It works if I uncomment first register "import_energy_reactive_q1"
```

root@OpenWrt:~/solaredge_modbus# python3 example.py 192.168.1.119 1502
Meter1 Meter1(192.168.1.119:1502, connectionType.TCP: timeout=1, unit=0x1)
        {'c_model': 'PRO380-Mod', 'c_option': 'Export+Import', 'c_version': '2.19', 'c_serialnumber': '19032098', 'c_deviceaddress': 1, 'c_sunspec_did': 203, 'current': 49, 'p1_current': 11, 'p2_current': 34, 'p3_current': 3, 'current_scale': -1, 'voltage_ln': 24050, 'p1n_voltage': 24050, 'p2n_voltage': 24030, 'p3n_voltage': 24150, 'voltage_ll': 0, 'p12_voltage': 0, 'p23_voltage': 0, 'p31_voltage': 0, 'voltage_scale': -2, 'frequency': 4993, 'frequency_scale': -2, 'power': -1136, 'p1_power': -263, 'p2_power': -831, 'p3_power': -42, 'power_scale': 0, 'power_apparent': 1138, 'p1_power_apparent': 263, 'p2_power_apparent': 831, 'p3_power_apparent': 72, 'power_apparent_scale': 0, 'power_reactive': -76, 'p1_power_reactive': 16, 'p2_power_reactive': -33, 'p3_power_reactive': -59, 'power_reactive_scale': 0, 'power_factor': 8633, 'p1_power_factor': 10000, 'p2_power_factor': 10000, 'p3_power_factor': 5899, 'power_factor_scale': -2, 'export_energy_active': 6599, 'p1_export_energy_active': 2900, 'p2_export_energy_active': 2220, 'p3_export_energy_active': 2310, 'import_energy_active': 10819, 'p1_import_energy_active': 4710, 'p2_import_energy_active': 5740, 'p3_import_energy_active': 1200, 'energy_active_scale': 0, 'export_energy_apparent': 0, 'p1_export_energy_apparent': 0, 'p2_export_energy_apparent': 0, 'p3_export_energy_apparent': 0, 'import_energy_apparent': 0, 'p1_import_energy_apparent': 0, 'p2_import_energy_apparent': 0, 'p3_import_energy_apparent': 0, 'energy_apparent_scale': -32768, 'import_energy_reactive_q1': 0}
```

nmakel, 2020-06-18:
> It works if I uncomment first register "import_energy_reactive_q1"
>
> ```
>
> root@OpenWrt:~/solaredge_modbus# python3 example.py 192.168.1.119 1502
> Meter1 Meter1(192.168.1.119:1502, connectionType.TCP: [...]

How bizarre. Not quite sure what's going on there. In order to figure out where this is going wrong I will need to add a bit of debug code to the _read_all and _decode_value methods. I will post a message once I've done so.

bizonxx, 2020-06-18:
Apparent energy also is not present. I logged by Simply Modbus TCP client and this what I've got:

```
			40243	40245	40247	40249	40251	40253	40255	40257	40259	40260	40262
2020-06-18 23:09:13	0	0	0	0	0	0	0	0	-32768	0	0
```

nmakel, 2020-06-19:
Would you mind running the following? I'm looking for whether this is a too many registers at once issue, or a specific register tripping up the pymodbus library.

```
    meters = inverter.meters()
    
    for m, mo in meters.items():
        for r, v in mo.registers.items():
            print(r, v)
            print(mo.read(r))
        print(m)
        print(mo.read_all())
```

bizonxx, 2020-06-19:
```
c_model (40139, 16, <registerType.HOLDING: 2>, <registerDataType.STRING: 12>, <class 'str'>, 'Model', '')
{'c_model': 'PRO380-Mod'}
c_option (40155, 8, <registerType.HOLDING: 2>, <registerDataType.STRING: 12>, <class 'str'>, 'Configuration', '')
{'c_option': 'Export+Import'}
c_version (40163, 8, <registerType.HOLDING: 2>, <registerDataType.STRING: 12>, <class 'str'>, 'Version', '')
{'c_version': '2.19'}
c_serialnumber (40171, 16, <registerType.HOLDING: 2>, <registerDataType.STRING: 12>, <class 'str'>, 'Serial', '')
{'c_serialnumber': '19032098'}
c_deviceaddress (40187, 1, <registerType.HOLDING: 2>, <registerDataType.UINT16: 3>, <class 'int'>, 'Modbus ID', '')
{'c_deviceaddress': 1}
c_sunspec_did (40188, 1, <registerType.HOLDING: 2>, <registerDataType.UINT16: 3>, <class 'int'>, 'SunSpec DID', {'101': 'Single Phase Inverter', '102': 'Split Phase Inverter', '103': 'Three Phase Inverter', '104': 'Single Phase Meter', '105': 'Three Phase Meter', '201': 'Single Phase Meter', '202': 'Split Phase Meter', '203': '3P1N Three Phase Meter', '204': '3P Three Phase Meter'})
{'c_sunspec_did': 203}
current (40190, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Current', 'A')
{'current': 7}
p1_current (40191, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1 Current', 'A')
{'p1_current': 6}
p2_current (40192, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2 Current', 'A')
{'p2_current': 10}
p3_current (40193, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3 Current', 'A')
{'p3_current': -10}
current_scale (40194, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Current Scale Factor', '')
{'current_scale': -1}
voltage_ln (40195, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Voltage', 'V')
{'voltage_ln': 24010}
p1n_voltage (40196, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1-N Voltage', 'V')
{'p1n_voltage': 24010}
p2n_voltage (40197, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2-N Voltage', 'V')
{'p2n_voltage': 24069}
p3n_voltage (40198, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3-N Voltage', 'V')
{'p3n_voltage': 24039}
voltage_ll (40199, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Voltage Line to Line', 'V')
{'voltage_ll': 0}
p12_voltage (40200, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1-P2 Voltage', 'V')
{'p12_voltage': 0}
p23_voltage (40201, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2-P3 Voltage', 'V')
{'p23_voltage': 0}
p31_voltage (40202, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3-P1 Voltage', 'V')
{'p31_voltage': 0}
voltage_scale (40203, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Voltage Scale Factor', '')
{'voltage_scale': -2}
frequency (40204, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Frequency', 'Hz')
{'frequency': 5000}
frequency_scale (40205, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Frequency Scale Factor', '')
{'frequency_scale': -2}
power (40206, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power', 'W')
{'power': 78}
p1_power (40207, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1 Power', 'W')
{'p1_power': -59}
p2_power (40208, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2 Power', 'W')
{'p2_power': 59}
p3_power (40209, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3 Power', 'W')
{'p3_power': 78}
power_scale (40210, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power Scale Factor', '')
{'power_scale': 0}
power_apparent (40211, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power (Apparent)', 'VA')
{'power_apparent': 536}
p1_power_apparent (40212, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1 Power (Apparent)', 'VA')
{'p1_power_apparent': 123}
p2_power_apparent (40213, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2 Power (Apparent)', 'VA')
{'p2_power_apparent': 208}
p3_power_apparent (40214, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3 Power (Apparent)', 'VA')
{'p3_power_apparent': 235}
power_apparent_scale (40215, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power (Apparent) Scale Factor', '')
{'power_apparent_scale': 0}
power_reactive (40216, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power (Reactive)', 'VA')
{'power_reactive': -531}
p1_power_reactive (40217, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1 Power (Reactive)', 'VA')
{'p1_power_reactive': -109}
p2_power_reactive (40218, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2 Power (Reactive)', 'VA')
[... 11 lines cut ...]
p3_power_factor (40224, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3 Power Factor', '')
{'p3_power_factor': 3300}
power_factor_scale (40225, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power Factor Scale Factor', '')
{'power_factor_scale': -2}
[... cut ...] : -109, 'p2_power_reactive': -200, 'p3_power_reactive': -222, 'power_reactive_scale': 0, 'power_factor': 3633, 'p1_power_factor': 4799, 'p2_power_factor': 2800, 'p3_power_factor': 3300, 'power_factor_scale': -2, 'export_energy_active': 6619, 'p1_export_energy_active': 3070, 'p2_export_energy_active': 2260, 'p3_export_energy_active': 2790, 'import_energy_active': 15640, 'p1_import_energy_active': 6460, 'p2_import_energy_active': 9340, 'p3_import_energy_active': 1340, 'energy_active_scale': 0}
```

bizonxx, 2020-06-19:
Full registers:

```
c_model (40139, 16, <registerType.HOLDING: 2>, <registerDataType.STRING: 12>, <class 'str'>, 'Model', '')
{'c_model': 'PRO380-Mod'}
c_option (40155, 8, <registerType.HOLDING: 2>, <registerDataType.STRING: 12>, <class 'str'>, 'Configuration', '')
{'c_option': 'Export+Import'}
c_version (40163, 8, <registerType.HOLDING: 2>, <registerDataType.STRING: 12>, <class 'str'>, 'Version', '')
{'c_version': '2.19'}
c_serialnumber (40171, 16, <registerType.HOLDING: 2>, <registerDataType.STRING: 12>, <class 'str'>, 'Serial', '')
{'c_serialnumber': '19032098'}
c_deviceaddress (40187, 1, <registerType.HOLDING: 2>, <registerDataType.UINT16: 3>, <class 'int'>, 'Modbus ID', '')
{'c_deviceaddress': 1}
c_sunspec_did (40188, 1, <registerType.HOLDING: 2>, <registerDataType.UINT16: 3>, <class 'int'>, 'SunSpec DID', {'101': 'Single Phase Inverter', '102': 'Split Phase Inverter', '103': 'Three Phase Inverter', '104': 'Single Phase Meter', '105': 'Three Phase Meter', '201': 'Single Phase Meter', '202': 'Split Phase Meter', '203': '3P1N Three Phase Meter', '204': '3P Three Phase Meter'})
{'c_sunspec_did': 203}
current (40190, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Current', 'A')
{'current': -5}
p1_current (40191, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1 Current', 'A')
{'p1_current': -5}
p2_current (40192, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2 Current', 'A')
{'p2_current': 9}
p3_current (40193, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3 Current', 'A')
{'p3_current': -9}
current_scale (40194, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Current Scale Factor', '')
{'current_scale': -1}
voltage_ln (40195, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Voltage', 'V')
{'voltage_ln': 24089}
p1n_voltage (40196, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1-N Voltage', 'V')
{'p1n_voltage': 24089}
p2n_voltage (40197, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2-N Voltage', 'V')
{'p2n_voltage': 24160}
p3n_voltage (40198, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3-N Voltage', 'V')
{'p3n_voltage': 24080}
voltage_ll (40199, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Voltage Line to Line', 'V')
{'voltage_ll': 0}
p12_voltage (40200, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1-P2 Voltage', 'V')
{'p12_voltage': 0}
p23_voltage (40201, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2-P3 Voltage', 'V')
{'p23_voltage': 0}
p31_voltage (40202, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3-P1 Voltage', 'V')
{'p31_voltage': 0}
voltage_scale (40203, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Voltage Scale Factor', '')
{'voltage_scale': -2}
frequency (40204, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Frequency', 'Hz')
{'frequency': 5002}
frequency_scale (40205, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Frequency Scale Factor', '')
{'frequency_scale': -2}
power (40206, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power', 'W')
{'power': 53}
p1_power (40207, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1 Power', 'W')
{'p1_power': 10}
p2_power (40208, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2 Power', 'W')
{'p2_power': 0}
p3_power (40209, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3 Power', 'W')
{'p3_power': 50}
power_scale (40210, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power Scale Factor', '')
{'power_scale': 0}
power_apparent (40211, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power (Apparent)', 'VA')
{'power_apparent': 486}
p1_power_apparent (40212, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1 Power (Apparent)', 'VA')
{'p1_power_apparent': 118}
p2_power_apparent (40213, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2 Power (Apparent)', 'VA')
{'p2_power_apparent': 146}
p3_power_apparent (40214, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3 Power (Apparent)', 'VA')
{'p3_power_apparent': 224}
power_apparent_scale (40215, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power (Apparent) Scale Factor', '')
{'power_apparent_scale': 0}
power_reactive (40216, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power (Reactive)', 'VA')
{'power_reactive': -483}
p1_power_reactive (40217, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1 Power (Reactive)', 'VA')
{'p1_power_reactive': -118}
p2_power_reactive (40218, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2 Power (Reactive)', 'VA')
{'p2_power_reactive': -146}
p3_power_reactive (40219, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3 Power (Reactive)', 'VA')
{'p3_power_reactive': -219}
power_reactive_scale (40220, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power (Reactive) Scale Factor', '')
{'power_reactive_scale': 0}
[... cut ...]
    result = self.client.read_holding_registers(address=address, count=length, unit=self.unit)
  File "/usr/lib/python3.7/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/lib/python3.7/site-packages/pymodbus/client/sync.py", line 107, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.119:1502)]
```

bizonxx, 2020-06-19:
Does it seem timeout? Each time I'm having an error in a different place.

bizonxx, 2020-06-19:
```
root@OpenWrt:~/solaredge_modbus# python3 inverter.py 192.168.1.119 1502
c_model (40139, 16, <registerType.HOLDING: 2>, <registerDataType.STRING: 12>, <class 'str'>, 'Model', '')
{'c_model': 'PRO380-Mod'}
c_option (40155, 8, <registerType.HOLDING: 2>, <registerDataType.STRING: 12>, <class 'str'>, 'Configuration', '')
{'c_option': 'Export+Import'}
c_version (40163, 8, <registerType.HOLDING: 2>, <registerDataType.STRING: 12>, <class 'str'>, 'Version', '')
{'c_version': '2.19'}
c_serialnumber (40171, 16, <registerType.HOLDING: 2>, <registerDataType.STRING: 12>, <class 'str'>, 'Serial', '')
{'c_serialnumber': '19032098'}
c_deviceaddress (40187, 1, <registerType.HOLDING: 2>, <registerDataType.UINT16: 3>, <class 'int'>, 'Modbus ID', '')
{'c_deviceaddress': 1}
c_sunspec_did (40188, 1, <registerType.HOLDING: 2>, <registerDataType.UINT16: 3>, <class 'int'>, 'SunSpec DID', {'101': 'Single Phase Inverter', '102': 'Split Phase Inverter', '103': 'Three Phase Inverter', '104': 'Single Phase Meter', '105': 'Three Phase Meter', '201': 'Single Phase Meter', '202': 'Split Phase Meter', '203': '3P1N Three Phase Meter', '204': '3P Three Phase Meter'})
{'c_sunspec_did': 203}
current (40190, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Current', 'A')
{'current': 5}
p1_current (40191, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1 Current', 'A')
{'p1_current': 5}
p2_current (40192, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2 Current', 'A')
{'p2_current': 9}
p3_current (40193, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3 Current', 'A')
{'p3_current': -9}
current_scale (40194, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Current Scale Factor', '')
{'current_scale': -1}
voltage_ln (40195, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Voltage', 'V')
{'voltage_ln': 24160}
p1n_voltage (40196, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1-N Voltage', 'V')
{'p1n_voltage': 24160}
p2n_voltage (40197, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2-N Voltage', 'V')
{'p2n_voltage': 24169}
p3n_voltage (40198, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3-N Voltage', 'V')
{'p3n_voltage': 24239}
voltage_ll (40199, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Voltage Line to Line', 'V')
{'voltage_ll': 0}
p12_voltage (40200, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1-P2 Voltage', 'V')
{'p12_voltage': 0}
p23_voltage (40201, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2-P3 Voltage', 'V')
{'p23_voltage': 0}
p31_voltage (40202, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3-P1 Voltage', 'V')
{'p31_voltage': 0}
voltage_scale (40203, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Voltage Scale Factor', '')
{'voltage_scale': -2}
frequency (40204, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Frequency', 'Hz')
{'frequency': 4995}
frequency_scale (40205, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Frequency Scale Factor', '')
{'frequency_scale': -2}
power (40206, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power', 'W')
{'power': 128}
p1_power (40207, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1 Power', 'W')
{'p1_power': 23}
p2_power (40208, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2 Power', 'W')
{'p2_power': 19}
p3_power (40209, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3 Power', 'W')
{'p3_power': 86}
power_scale (40210, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power Scale Factor', '')
{'power_scale': 0}
power_apparent (40211, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power (Apparent)', 'VA')
{'power_apparent': 452}
p1_power_apparent (40212, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1 Power (Apparent)', 'VA')
{'p1_power_apparent': 86}
p2_power_apparent (40213, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P2 Power (Apparent)', 'VA')
{'p2_power_apparent': 139}
p3_power_apparent (40214, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P3 Power (Apparent)', 'VA')
{'p3_power_apparent': 229}
power_apparent_scale (40215, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power (Apparent) Scale Factor', '')
{'power_apparent_scale': 0}
power_reactive (40216, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'Power (Reactive)', 'VA')
{'power_reactive': -434}
p1_power_reactive (40217, 1, <registerType.HOLDING: 2>, <registerDataType.INT16: 7>, <class 'int'>, 'P1 Power (Reactive)', 'VA')
{'p1_power_reactive': -83}
[... 77 lines cut ...]
{'export_energy_reactive_q4': 0}
p1_export_energy_reactive_q4 (40286, 2, <registerType.HOLDING: 2>, <registerDataType.UINT32: 4>, <class 'int'>, 'P1 Exported Energy (Reactive) Quadrant 4', 'VARh')
{'p1_export_energy_reactive_q4': 0}
p2_export_energy_reactive_q4 (40288, 2, <registerType.HOLDING: 2>, <registerDataType.UINT32: 4>, <class 'int'>, 'P2 Exported Energy (Reactive) Quadrant 4', 'VARh')
[... cut ...]
    result = self.client.read_holding_registers(address=address, count=length, unit=self.unit)
  File "/usr/lib/python3.7/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/lib/python3.7/site-packages/pymodbus/client/sync.py", line 107, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.119:1502)]
```

bizonxx, 2020-06-19:
print(mo.read_all()) as first makes and error immediately

```
meters = inverter.meters()

for m, mo in meters.items():
	print(mo.read_all())
	for r, v in mo.registers.items():
		print(r, v)
		print(mo.read(r))
	print(m)
```

Without it works fine.

nmakel, 2020-06-19:
Ok, so it seems like every register can be read when queried individually, even the energy_apparent registers at the end. Seems like the read_all query is exceeding the maximum number of registers per query. I will need to modify _read_all() to process the list of registers in chunks.

nmakel, 2020-06-19:
> `print(mo.read_all())` as first makes an error immediately
> 
> ```
> meters = inverter.meters()
> 
> for m, mo in meters.items():
> 	print(mo.read_all())
> 	for r, v in mo.registers.items():
> 		print(r, v)
> 		print(mo.read(r))
> 	print(m)
> ```
> 
> Without `print(mo.read_all())` works fine.

I've added a rudimentary batching system, which breaks up the meter register query into three chunks. Could you please try again? The following should be enough:

```
meters = inverter.meters()

for m, mo in meters.items():
	print(mo.read_all())
```

nmakel, 2020-06-21:
Meter functionality is now in the master branch. Please let me know if there's any further issues.

bizonxx, 2020-06-22:
Now it works perfect. Thank you very much.

bizonxx, 2020-06-22:
I found this. Limit of reading registers is 122.

_SolarEdge has implemented two methods of the MODBUS request procedure:  MODBUS request with explicit register addressing - supported by all communication board CPU versions. For example:•Tx: 01 03 9C 40 00 7A EB AD – **Read 122 registers** starting at address 40001.•Rx: 01 03 F4 53 75 ... [Registers data] ... FF FF 12 1BMODBUS request without explicit addressing – supported by communication board CPU version 2.478 and above. For example:   •Tx: 01 03 00 00 00 7A C4 29 – Read 122 registers starting at offset 0.•Rx: 01 03 F4 53 75 6E 53 ... [Registers data] ... FF FF AE DB_

nmakel, 2020-06-22:
Good to know, thanks for your help!

closed 2020-06-22 by nmakel