nmakel/solaredge_modbus #2: meters() not working

m--k, 2020-06-24:
Hi, I am testing your code on an Raspberry Pi. I have a Solaredge inverter with a Solaredge meter.
This is working:
`meter = solaredge_modbus.Meter(host="192.168.178.81", port=1502)`

This is working and give a correct value:
```
>>> inverter.read("frequency")
{'frequency': 5003}
```

But this gives an error message:
```
>>> inverter.meters()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'Inverter' object has no attribute 'meters'
>>>
```

With a different solution (ioBroker), I can sucessfully read the values of the meter. e.g. registers 40207, 40211
[image]

nmakel, 2020-06-25:
Are you using the latest git version, or the package from pypi? I haven't pushed the changes to a new pypi release, yet.

m--k, 2020-06-25:
Thanks for this hint. I will try it tonight 

Best regards 
Markus

m--k, 2020-06-29:
It works!! Thanks :) 
I plan to use your project together with https://github.com/ngardiner/TWCManager to control a tesla wall charger.

nmakel, 2020-06-29:
Good to hear, thanks for letting me know.

closed 2020-06-29 by nmakel