nmakel/solaredge_modbus #4: able to connect, but cannot read any values

allandaly, 2020-09-14:
Hello Niels,

Thanks a ton for all the work you put into this library. Looks great. I am trying to use it to talk with my SolarEdge 7.6KW HD-Wave with EV Charger, SE7600H. I enabled Modbus TCP on port 502. I can ping the inverter's IP address and I'm able to use your library and also Simply TCP Client to 'connect' to the inverter.

However, when I send any of the 'read' commands using your library and also the Simply TCP Client I do not get any response. I've tried restarting the inverter -- no change.

I realize there's a lot to debug here to get this to work and I'm going to dive into that (upgrade firmware, etc.) but I wonder if you can offer any advice based on your experience with this equipment. Do you have any suggestions of what I should try first? I've been working on this for about 6 hours now and have run out of ideas short of upgrading the firmware and trying again.

By the way, current firmware is as follows. (Also see attached pic.)
DSP1/2:1.000/2.000
CPU: 3.2536

Do you recommend upgrading the firmware?

Thank you for any help or advice you can offer.

-Allan

[image]

nmakel, 2020-09-14:
Hi Allan,

I can't say whether a firmware upgrade will do you any good -- the only experience I have is with the non-LCD unit I have here. I'm guessing you've already read through SolarEdge's document on the modbus implementation? Just to be sure, you can find it [here](https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf). Please check pages 13 and 14, to confirm the inverter is configured properly. The documentation also states -- albeit rather ambiguously -- there is a two minute timeout. Perhaps this is something you're running into?

But it would be helpful to know whether you have any other devices connected on the RS485 bus. You mention an EV charger. Is there also a revenue meter connected?

Also, if you'd like you can paste the code you're using, to be sure it's not an error on that end.

allandaly, 2020-09-15:
Hi Niels,

The version of the SolarEdge SE7600H that I have includes an integrated EV charging accessory. I thought that was a pretty smart technology to integrate even though I don't yet have an electric vehicle. Anyway, maybe that is affecting the Modbus internals. Not sure. I do believe there's a revenue-grade meter in there on the EV charging load path, but it's not explicitly called out in any of the Modbus documentation I've been able to find.

I also do have the external SolarEdge-supplied energy meter attached to this inverter. It measures the power/energy at my main grid connection. It's connected via Modbus, but those connections are internal to the meter. Perhaps that is also affecting things, but it seems that others have been able to meter data out using your library.

Thank you for the link but yes I've been though that document many times already! 

Here's the code I'm trying to use. It is quite straightforward but if you do see an issue please let me know.

`
>>> import solaredge_modbus

>>> inverter = solaredge_modbus.Inverter(host='192.168.1.50', port=502)

>>> inverter.connected()
True

>>> inverter.read('current')
{'current': False}

>>> inverter.read('c_model')
{'c_model': False}

>>> inverter.read('status')
{'status': False}

>>> inverter.read_all()
{}
`

I just ordered the correct format of microSD card so I can do the firmware upgrade. It'll be here on Thursday. I'll let you know how it goes.

-Allan

nmakel, 2020-09-15:
> The version of the SolarEdge SE7600H that I have includes an integrated EV charging accessory. I thought that was a pretty smart technology to integrate even though I don't yet have an electric [...]

You're probably correct. I haven't come across any EV specific functionality yet, but the most obvious way is to treat it as a consumption meter. Alternatively as a battery if the vehicle supports power delivery.

> I also do have the external SolarEdge-supplied energy meter attached to this inverter. It measures the power/energy at my main grid connection. It's connected via Modbus, but those connections are [...]

From what I've heard, yes. The tests I've run with a single revenue meter work fine. The inverter should support three.

> Here's the code I'm trying to use. It is quite straightforward but if you do see an issue please let me know.

What you could try, just to be sure, is to add the `unit` parameter, which you can look up on your inverter:

`inverter = solaredge_modbus.Inverter(host='192.168.1.50', port=502, unit=2)`

> I just ordered the correct format of microSD card so I can do the firmware upgrade. It'll be here on Thursday. I'll let you know how it goes.

I hope that solves it, but if not, would you mind adding the following logging functionality to your script? Add the following just below the `import solaredge_modbus` line:

```
import logging  
import sys

log_handler = logging.StreamHandler(sys.stdout)
log_handler.setFormatter(logging.Formatter("%(asctime)s %(levelname)s: %(message)s", datefmt="%Y-%m-%d %H:%M:%S"))
logger = logging.getLogger()
logger.setLevel(logging.DEBUG)
logger.addHandler(log_handler)
```

This will produce a fair bit of debug logging from the `pymodbus` module. Hopefully this reveals more of what's happening under the hood.

As I've mentioned I only have a screenless HD Wave to test with. This unit never exhibited the timeout problem that [others](https://github.com/jbuehl/solaredge/issues/80#issuecomment-399656964) have [mentioned](https://community.openhab.org/t/reading-data-from-solaredge-inverters-via-modbus-tcp/26290). Something to be aware of when you're testing the connection, as it might be an issue on the earlier firmware 3.x units.

allandaly, 2020-09-15:
Ha! Your idea to use `unit=2` worked!! What made you think of suggesting that?

>import solaredge_modbus
>
>inverter = solaredge_modbus.Inverter(host='192.168.1.50', port=502, unit=2)
>
>inverter.connected()
>True
>
>inverter.read('current')
>{'current': 150}
>
>inverter.read('c_model')
>{'c_model': 'SE7600'}
>
>inverter.read('status')
>{'status': 4}

Thanks so much for your help. I'll report back on what I find with the various meters in my configuration (EV model + external kWh meter).

-Allan

nmakel, 2020-09-15:
> Ha! Your idea to use `unit=2` worked!! What made you think of suggesting that?

Good to hear!

You should be able to confirm the modbus address of your inverter is set to 2 in the configuration menus. The module defaults to 1 if none is provided. Most likely your EV charger or revenue meter is set to address 1, because changing that of the inverter is usually less hassle.

closed 2020-09-29 by nmakel

geisserm, 2021-02-04:
Thanks for the hint ... my modbus device ID was set to 5 so adding unit=5 did the trick for me.