nmakel/solaredge_modbus #9: generalized sunspec

mihailescu2m, 2020-10-09:
Hi,

Is there anything that is actually SolarEdge specific in here?
SunSpec is not Solaredge specific, so as long as all is SunSpec, would you consider renaming your SolarEdge class to e.g. SunSpec and the project?

PS: can you also add a disconnect/close function, or would you prefer me to make a PR?

Thanks for your work!

nmakel, 2020-10-09:
> Is there anything that is actually SolarEdge specific in here?

Not sure. I haven't spent much time looking at other manufacturer implementations of SunSpec. If the library works for different manufacturers, great!

> PS: can you also add a disconnect/close function, or would you prefer me to make a PR?

I haven't needed this in a tcp setting, but should be trivial to rework the existing connected() function and add a connect() and disconnect() corresponding to the pymodbus calls. Thanks for the suggestion.

mihailescu2m, 2020-10-09:
> Not sure. I haven't spent much time looking at other manufacturer implementations of SunSpec. If the library works for different manufacturers, great!

I will try finding someone with a Fronius to check. But in theory it should work, and this is why it would be nice to have general names.

> I haven't needed this in a tcp setting, but should be trivial to rework the existing connected() function and add a connect() and disconnect() corresponding to the pymodbus calls. Thanks for the suggestion.

`inverter.client.close()` actually works... but it would nicer to have a `inverter.disconnect()`

mihailescu2m, 2020-10-10:
Can you also add this register to the inverter please:

```
        self.registers = {
            "c_manufacturer": (0x9c44, 16, registerType.HOLDING, registerDataType.STRING, str, "Manufacturer", "", 1),
            "c_model": (0x9c54, 16, registerType.HOLDING, registerDataType.STRING, str, "Model", "", 1),
```

nmakel, 2020-10-11:
> Can you also add this register to the inverter please:
> 
> ```
> ...
> self.registers = {
>     "c_manufacturer": (0x9c44, 16, registerType.HOLDING, registerDataType.STRING, str, "Manufacturer", "", 1),
>     "c_model": (0x9c54, 16, registerType.HOLDING, registerDataType.STRING, str, "Model", "", 1),
> ...
> ```

Good point, done.

closed 2020-12-24 by nmakel