nmakel/solaredge_modbus #10: Strange values for energy_total

addiejanssen, 2020-10-20:
Hi Niels,

I have created a plugin for Domoticz using your library (https://github.com/addiejanssen/domoticz-solaredge-modbustcp-plugin).
It's working great for most users, but there is one user that reports strange values for the `energy_total` values.
Initially, we thought it was a Domoticz topic or something related to the plugin, but I think there is an issue with the values returned by either the inverter itself or your library.
Here's a dump of what comes back from `Inverter.read_all()` (I removed the serial number of the inverter):

```json
{
    "c_model": "SE5K-RW0TEBNN4",
    "c_version": "0004.0010.0025",
    "c_deviceaddress": 1,
    "c_sunspec_did": 103,
    "current": 222,
    "p1_current": 67,
    "p2_current": 73,
    "p3_current": 81,
    "current_scale": -2,
    "p1_voltage": 3961,
    "p2_voltage": 3961,
    "p3_voltage": 4001,
    "p1n_voltage": 2319,
    "p2n_voltage": 2273,
    "p3n_voltage": 2291,
    "voltage_scale": -1,
    "power_ac": 18500,
    "power_ac_scale": -2,
    "frequency": 4999,
    "frequency_scale": -2,
    "power_apparent": 5078,
    "power_apparent_scale": -1,
    "power_reactive": -4730,
    "power_reactive_scale": -1,
    "power_factor": -3611,
    "power_factor_scale": -2,
    "energy_total": 1059323905,
    "energy_total_scale": 0,
    "current_dc": 25118,
    "current_dc_scale": -5,
    "voltage_dc": 7477,
    "voltage_dc_scale": -1,
    "power_dc": 18781,
    "power_dc_scale": -2,
    "temperature": 2735,
    "temperature_scale": -2,
    "status": 4,
    "vendor_status": 0
}
```

Looks ok, but the funny bit is the value of energy_total.
We have seen the following values comming back:

```
2020-10-11 15:00:00    57160500
2020-10-11 16:00:00    65182104
2020-10-11 17:00:00    20984628
2020-10-11 18:00:00     1350042

2020-10-12 08:00:00     1795686
2020-10-12 09:00:00    13434880
2020-10-12 10:00:00    27761050
2020-10-12 11:00:00    40317748
2020-10-12 12:00:00    65470464
2020-10-12 13:00:00   151807584
2020-10-12 14:00:00   127231592

2020-10-15 17:44:10  3128439603
2020-10-15 17:45:10  3128714854
2020-10-15 17:47:11  3129265356
2020-10-15 17:48:11  3129527500
2020-10-15 17:49:11  3129763430
2020-10-15 17:50:11  3129986252
2020-10-15 17:51:11  3130195968

2020-10-16 14:19:22  3658350592
2020-10-16 14:28:22  3691970560

2020-10-18 08:27:51   264896513

2020-10-20 12:21:02  1059323905
```

The inverter and panels are new and have been installed a few weeks ago.
According to the SolarEdge monitoring site and app, the installation produced just over 70 kWh in total so far.

All the other values are making perfect sense and seem to align with what SolarEdge is reporting as well.
It is just the energy_total that is all over the place and going up and down.

Can you help us out?

Thanks,

Addie

nmakel, 2020-10-20:
Hi Addie,

That's an endianness problem. It should be solved in 0.6.2.

Somewhere along the way batteries were added to the library, which for some reason require a different word order than inverters or kwh meters. Version 0.6.0 and 0.6.1 will display incorrect values for inverter.total_energy.

Please let me know if users are still experiencing this problem even if they have the latest git or pypi release of the module.

addiejanssen, 2020-10-21:
Thanks Niels,

I will try to find out which version he is currently using and then ask him to upgrade to 0.6.2.
Will come back with the results as soon as possible, but it may take a few days :-)

Kind regards,

Addie

addiejanssen, 2020-10-21:
Hi Niels,

Checked with the user; he was already using 0.6.2.
When I created the plugin, your library was at 0.5.0; we will try that version as well, just to make sure.

addiejanssen, 2020-10-21:
Same issues with 0.5.0.....

nmakel, 2020-10-21:
Thanks for all the detective work :)

Are you also experiencing this problem? I see a range of inverters and versions in your thread, each with different results:

SE3500H, mine, no problem, 4.9.30 firmware.
SE7K, yours, no problem, 3.x firmware.
SE5K, frank's, weird total_energy, 4.10.25 firmware.
SE3500, mcmellow's, weird total_energy, 3.x firmware.

Which makes... no sense.

What I also see is that both mcmellow and frank posted the debug output from your plugin. The 0.6.2 release includes c_manufacturer as the first value in the inverter's json object -- like in your debug output -- while theirs still show c_model as the first value. Is there any way they could still be using an older version? Caching by domoticz, or something similar?

addiejanssen, 2020-10-21:
Love the statement 'makes no sense'. I have had that feeling from the moment the issue was logged :-)

There are no issues with my setup; I have 2 servers running at this moment; one uses 0.5.0 and the other 0.6.2; both work fine.

Not sure what the status is for mcmellow; I have asked him for an update.
I did ask him at some point to downgrade to 0.5.0; in that case, it makes sense that the c_manufacturer is missing. Let's wait for his response and then ask him to upgrade.

With regards to Frank: I now realize that I was not completely clear in my instructions to him and he may now have 2 versions of your library on his computer. I have asked him to remove the library and install it again and then to restart his computer; just to make sure.

addiejanssen, 2020-10-22:
Hi Niels,

Quick update: all problems seem to be solved :-)

The slightly longer version:
Mcmellow responded that he has no issue any more.

Frank had 2 versions of your library on his computer; one of them being 0.6.1 and that one was active.
He deleted all of them and then installed the latest version, which reports the correct value.

So, for now, I think we are done.

Thanks,

Addie

closed 2020-10-22 by addiejanssen