nmakel/solaredge_modbus #11: NO ISSUE but QUESTION:   Writing to the inverters

Maikel-K, 2020-11-18:
Hi,

Would it be possible to write data to the inverter?
I got a smart meter which gives me data about my actual import/export. It would be nice if we can send that over modbus to the interver instead of buying a compatible meter. This would save me/us money.

nmakel, 2020-11-18:
Absolutely. Check out [solaredge_meterproxy](https://github.com/nmakel/solaredge_meterproxy).

I use this to allow my inverter to talk to my incompatible kwh meter, replying to queries from the inverter with values I get from my meter. In theory any data source can be used, be it a kwh meter that supports modbus, an sqlite database, or a time series database like influxdb.

Maikel-K, 2020-11-18:
Wow! That's was what I was looking for! I got the SDM630 so this is awesome! 

I search myself to death for this! Great news😁

nmakel, 2020-11-18:
Good to hear! Would be great to hear whether you get things working in your situation.

Maikel-K, 2020-11-18:
Yeah I wonder if I can get it to work. Do you want to continue conversation through GitHub or maybe Facebook chat? I think I have some questions for you now already.

closed 2020-12-24 by nmakel