nmakel/solaredge_modbus #12: Field type conflict Field Power_Apparent in Measurement Meter

barendberg, 2020-11-19:
Inverter gives back different datatypes for Power_Apparent. Sometimes INT sometimes Float.

Is it possible to convert it to a float always before sending JSON to Inlflux?

Error:
input field \"power_apparent\" on measurement \"meter\" is type integer, already exists as type float dropped=1"

nmakel, 2020-11-19:
Are you using the examply_influxdb.py script? Looking at the relevant code seems to show all inverter values are cast to float -- including `power_apparent`.

https://github.com/nmakel/solaredge_modbus/blob/5da97ceabc3bfe74318b100762aae4f09c5904d2/example_influxdb.py#L85

Do you have any logging that shows what value is actually being sent to influxdb?

barendberg, 2020-11-20:
I think the code is not in place for the meter_data only for the inverter_data. Did try myself to fix it to convert it with 
meter_data["fields"].update({k: float(v)}) on line 98, but getting error: ValueError: could not convert string to float: 'SolarEdge'

Off course it is imposible to convert Solaredge to float, but how can i only convert data that can be converted to float? 

I've not got any python knowledge.

influxdb_client.rest.ApiException: (500)
Reason: Internal Server Error

HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Platform-Error-Code': 'internal error', 'Date': 'Fri, 20 Nov 2020 06:26:43 GMT', 'Content-Length': '230'})
HTTP response body: {"code":"internal error","message":"unexpected error writing points to database: partial write: field type conflict: input field \"power_apparent\" on measurement \"meter\" is type integer, already exists as type float dropped=1"}

barendberg, 2020-11-20:
replaced meter_data["fields"].update({k: v}) with meter_data["fields"].update({k: float(v)})

Created a new bucket in Influx, because multiple values now changed from int to float.

nmakel, 2020-11-20:
You're absolutely right. I assumed you were talking about the inverter value. I have added the float cast to the meter int/float values.

closed 2020-11-24 by nmakel