nmakel/solaredge_modbus #13: example.py and serial connection to inverter

Maikel-K, 2020-11-21:
Hi Niels,

To test if the communication between SE3500H and RPI is working i want to test this script.

Im using a USB to RS485 dongle. its connected to /dev/ttyUSB1

Is example.py able to work with Modbus serial or only via TCP?
If working via serial, how should i run the script?

python3 example.py /dev/ttyUSB1/ 115200 doesnt work:

pi@raspberrypi4:~/solaredgemodbus $ python3 example.py /dev/ttyUSB1/ 115200
Traceback (most recent call last):
  File "example.py", line 26, in <module>
    values = inverter.read_all()
  File "/home/pi/solaredgemodbus/solaredge_modbus/__init__.py", line 343, in read_all
    results.update(self._read_all(register_batch, rtype))
  File "/home/pi/solaredgemodbus/solaredge_modbus/__init__.py", line 296, in _read_all
    data = self._read_holding_registers(offset, length)
  File "/home/pi/solaredgemodbus/solaredge_modbus/__init__.py", line 218, in _read_holding_registers
    result = self.client.read_holding_registers(address=address, count=length, unit=self.unit)
  File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/client/sync.py", line 107, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(/dev/ttyUSB1/:115200)]

i think because in example.py we dont have the argparser stuff for the serial connection?

nmakel, 2020-11-21:
Correct, `example.py` only supports a TCP connection. Replace line 9 to 23 with the following to change it to a serial connection:

```
if __name__ == "__main__":
    argparser = argparse.ArgumentParser()
    argparser.add_argument("device", type=str, help="Modbus device")
    argparser.add_argument("--stopbits", type=int, default=False, help="Stop bits")
    argparser.add_argument("--parity", type=str, default=False, choices=["N", "E", "O"], help="Parity")
    argparser.add_argument("--baud", type=int, default=False, help="Baud rate")
    argparser.add_argument("--timeout", type=int, default=1, help="Connection timeout")
    argparser.add_argument("--unit", type=int, default=1, help="Modbus unit")
    argparser.add_argument("--json", action="store_true", default=False, help="Output as JSON")
    args = argparser.parse_args()

    inverter = solaredge_modbus.Inverter(
        device=args.device,
        stopbits=args.stopbits,
        parity=args.parity,
        baud=args.baud,
        timeout=args.timeout,
        unit=args.unit
    )
```

Alternatively, start up a `python3` interpreter and follow the instructions in the readme, specifically [here](https://github.com/nmakel/solaredge_modbus#examples).

Maikel-K, 2020-11-23:
i got the following error.... i suck at python unfortunatly....

 python3 example.py /dev/ttyUSB0,8,N,9600,1,1
Traceback (most recent call last):
  File "example.py", line 30, in <module>
    values = inverter.read_all()
  File "/home/pi/solaredgemodbus/solaredge_modbus/__init__.py", line 343, in read_all
    results.update(self._read_all(register_batch, rtype))
  File "/home/pi/solaredgemodbus/solaredge_modbus/__init__.py", line 296, in _read_all
    data = self._read_holding_registers(offset, length)
  File "/home/pi/solaredgemodbus/solaredge_modbus/__init__.py", line 218, in _read_holding_registers
    result = self.client.read_holding_registers(address=address, count=length, unit=self.unit)
  File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/client/sync.py", line 107, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusSerialClient(rtu baud[115200])]

nmakel, 2020-11-23:
Try: python3 example.py /dev/ttyUSB0 --baud=9600

Most of the other options have defaults you don't need to change.

Maikel-K, 2020-11-23:
Tried that, it takes some time for something to happen and the following message appears.
I cant remember correctly what i set as the baudrate, but 115200 gives the same result.

Sorry to bother you

python3 example.py /dev/ttyUSB0 --baud=9600
Inverter(/dev/ttyUSB0, connectionType.RTU: stopbits=1, parity=N, baud=9600, timeout=1, retries=3, unit=0x1):

Registers:
Traceback (most recent call last):
  File "example.py", line 50, in <module>
    print(f"\tManufacturer: {values['c_manufacturer']}")
KeyError: 'c_manufacturer'

nmakel, 2020-11-23:
> Registers:
> Traceback (most recent call last):
> File "example.py", line 50, in
> print(f"\tManufacturer: {values['c_manufacturer']}")
> KeyError: 'c_manufacturer'

This error means you're getting an empty response from the inverter. Have you already set the RS485-1 to Multi-Device and configured a meter in the inverter's interface? If so, you need to set the RS485 port on the inverter to Sunspec (Non-SE Logger).

Maikel-K, 2020-11-23:
Hi,

After a reboot the ttyUSB changed.... the problem for now is that I have two of the same type usb-RS485 adapters so I cant(haven't figured out how) give them persistent names... I can't find a difference in the two dongle data for that...

Anyhow for setting up the inverter, I got one with a display so it's a bit different I think.
I tried following your description in the readme, i had first set it to Non-Solaredge logger with Sunspec. Then I had to change it to multi device and I could only add Meter 1/2/3 but the settings for non-Solaredge logger and sunspec etc disappeared...?

nmakel, 2020-11-23:
You will need to set it back to non-solaredge logger to be able to read its modbus values. If you have already defined a meter it won't respond to the example script.

Maikel-K, 2020-11-23:
Okay I will try that tomorrow... I forgot that this script was for reading the data only and that this wasn't the script for sending the data too....

My bad.

closed 2020-11-29 by nmakel