nmakel/solaredge_modbus #15: Second battery returning invalid unicode

jglazko, 2020-11-28:
Hello all,

having issues reading via Modbus/TCP on a SolarEdge SE7600A-US inverter with two LG Chem RESU10H batteries. Inverter and first battery returning queries via Modbus/TCP just fine. Second battery... not so much. Looks to me like registers on battery 2 are overlapping with battery 1.

I've tried changing the decode at line 244 from UTF-8 to ISO-8859-1 and latin-1 to no resolution.

Attaching screenshots with serial numbers redacted for privacy. Glad to open up the modbus connection to you nmakel if you can provide an IP address.

>>> inverter.read('c_model')
{'c_model': 'SE7600'}
>>> inverter.read('c_version')
{'c_version': '0003.2618'}

>>> batteries = inverter.batteries()
>>> print(inverter.batteries())
{'Battery1': Battery1(172.26.7.155:502, connectionType.TCP: timeout=1, retries=3, unit=0x1), 'Battery2': Battery2(172.26.7.155:502, connectionType.TCP: timeout=1, retries=3, unit=0x1)}
>>> battery1 = batteries['Battery1']
>>> battery2 = batteries['Battery2']

>>> values=batteries['Battery1'].read_all()
>>> print(values)
{'c_manufacturer': 'LG\x01\x18\x029999999999', 'c_model': 'R15563P3SSEG19999999999', 'c_version': 'DCDC 7.4.8 BMS 1.7.0.1', 'c_serialnumber': '9999999999', 'c_deviceaddress': 15, 'c_sunspec_did': 0, 'rated_energy': 9800.0, 'maximum_charge_continuous_power': 5000.0, 'maximum_discharge_continuous_power': 5000.0, 'maximum_charge_peak_power': 5000.0, 'maximum_discharge_peak_power': 5000.0, 'average_temperature': 23.399999618530273, 'maximum_temperature': 0.0, 'instantaneous_voltage': 408.29998779296875, 'instantaneous_current': 1.8466814756393433, 'instantaneous_power': 754.0, 'lifetime_export_energy_counter': 0, 'lifetime_import_energy_counter': 5768, 'maximum_energy': 9800.0, 'available_energy': 9310.0, 'soh': 100.0, 'soe': 84.48979949951172, 'status': 3, 'status_internal': 3, 'event_log': 0, 'event_log_internal': 0}

>>> values=batteries['Battery2'].read_all()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jack/.local/lib/python3.8/site-packages/solaredge_modbus/__init__.py", line 343, in read_all
    results.update(self._read_all(register_batch, rtype))
  File "/home/jack/.local/lib/python3.8/site-packages/solaredge_modbus/__init__.py", line 311, in _read_all
    results[k] = self._decode_value(data, length, dtype, vtype)
  File "/home/jack/.local/lib/python3.8/site-packages/solaredge_modbus/__init__.py", line 244, in _decode_value
    decoded = data.decode_string(length * 2).decode("utf-8").replace("\x00", "").rstrip()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xbb in position 19: invalid start byte

nmakel, 2020-11-28:
> I've tried changing the decode at line 244 from UTF-8 to ISO-8859-1 and latin-1 to no resolution.

The problem is likely what you describe: incorrect registers for battery 2. The UTF-8 decoding error is merely a symptom.

Would you mind trying the following modification on line 136?

```
BATTERY_REGISTER_OFFSETS = [
    0x0,
    0x64
]
```

Should be:

```
BATTERY_REGISTER_OFFSETS = [
    0x0,
    0x100
]
```

jglazko, 2020-11-28:
That seems to return valid values for battery 2 and certainly gets me to much futher than I was. :)
>>> values={}
>>> values=batteries['Battery1'].read_all()
>>> print(values)
{'c_manufacturer': 'LG\x01\x18\x029999999999', 'c_model': 'R15563P3SSEG19999999999', 'c_version': 'DCDC 7.4.8 BMS 1.7.0.1', 'c_serialnumber': '9999999999', 'c_deviceaddress': 15, 'c_sunspec_did': 0, 'rated_energy': 9800.0, 'maximum_charge_continuous_power': 5000.0, 'maximum_discharge_continuous_power': 5000.0, 'maximum_charge_peak_power': 5000.0, 'maximum_discharge_peak_power': 5000.0, 'average_temperature': 23.5, 'maximum_temperature': 0.0, 'instantaneous_voltage': 421.29998779296875, 'instantaneous_current': -0.4604794681072235, 'instantaneous_power': -194.0, 'lifetime_export_energy_counter': 347, 'lifetime_import_energy_counter': 5840, 'maximum_energy': 9800.0, 'available_energy': 9310.0, 'soh': 100.0, 'soe': 81.4930191040039, 'status': 4, 'status_internal': 3, 'event_log': 0, 'event_log_internal': 0}
>>> values={}
>>> values=batteries['Battery2'].read_all()
>>> print(values)
{'c_manufacturer': 'LG\x01\x18\x028888888888', 'c_model': 'R15563P3SSEG28888888888', 'c_version': 'DCDC 7.4.8 BMS 1.7.0.1', 'c_serialnumber': '8888888888', 'c_deviceaddress': 14, 'c_sunspec_did': 0, 'rated_energy': 9800.0, 'maximum_charge_continuous_power': 5000.0, 'maximum_discharge_continuous_power': 5000.0, 'maximum_charge_peak_power': 5000.0, 'maximum_discharge_peak_power': 5000.0, 'average_temperature': 19.5, 'maximum_temperature': 0.0, 'instantaneous_voltage': 419.0, 'instantaneous_current': 0.0, 'instantaneous_power': 0.0, 'lifetime_export_energy_counter': 0, 'lifetime_import_energy_counter': 0, 'maximum_energy': 9800.0, 'available_energy': 9356.0, 'soh': 100.5, 'soe': 24.497648239135742, 'status': 1, 'status_internal': 1, 'event_log': 0, 'event_log_internal': 0}
>>>

nmakel, 2020-11-28:
Excellent! I'll push the change asap.

closed 2020-11-29 by nmakel