nmakel/solaredge_modbus #16: Showstopper on importing solaredge_modbus __init__.py

jglazko, 2020-12-13:
Hi nmakel,

installed newest version using pip3 today, having issues in python 3.8. Looks like a syntax error on line 184 of __init__.py:

jack@acer:~/dev/solar$ pip3 install solaredge_modbus
Collecting solaredge_modbus
  Using cached solaredge_modbus-0.6.3-py3-none-any.whl (10 kB)
Requirement already satisfied: pymodbus>=2.3.0 in /usr/local/lib/python3.8/dist-packages (from solaredge_modbus) (2.4.0)
Requirement already satisfied: pyserial>=3.4 in /usr/lib/python3/dist-packages (from pymodbus>=2.3.0->solaredge_modbus) (3.4)
Requirement already satisfied: six>=1.15.0 in /home/jack/.local/lib/python3.8/site-packages (from pymodbus>=2.3.0->solaredge_modbus) (1.15.0)
Installing collected packages: solaredge-modbus
Successfully installed solaredge-modbus-0.6.3
jack@acer:~/dev/solar$ python3
Python 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import solaredge_modbus
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jack/.local/lib/python3.8/site-packages/solaredge_modbus/__init__.py", line 184
    else
       ^
SyntaxError: invalid syntax
>>> 

----- Lines 181-185, looks like a missing colon on "else". edited source, works ok now.

            if (parity
                    and parity.upper() in ["N", "E", "O"]):
                self.parity = parity.upper()
            else
                self.parity = False

jglazko, 2020-12-13:
Looks great, updated through pip3 and is working well. I'll be hitting the batteries on my system this week (LG Chem RESU 10H), will look for anything "interesting" in there. :)

closed 2020-12-13 by jglazko

nmakel, 2020-12-13:
Thanks for the heads up, let me know if you run into any other issues.