nmakel/solaredge_modbus #18: Solaredge - SunSpec registers

ianmidd, 2020-12-19:


ianmidd, 2020-12-19:
Hi, judging by this most impressive contribution by nmakel I was hoping to get some guidance on the register numbers used. The Solaredge / SunSpec document appears to have some errors / omissions. I am doing something similar in NodeRed and so far all the registers 40189 onwards appear to be empty and there is no mention of Battery Registers in the document.

nmakel, 2020-12-19:
The Solaredge document you linked to is, as you've discovered, far from complete. The beginning of the battery blocks can be found by simply scanning through the entire range of possible registers. After that, the length of a battery block can be added to the starting register to get battery number 2.

If you take a look in the library source you will find starting registers for meters and batteries in the inverter class:

https://github.com/nmakel/solaredge_modbus/blob/8ff60638865b28dc9d45435430d2fa4215227c8d/solaredge_modbus/__init__.py#L413-L422

ianmidd, 2020-12-19:
When you say “scan”, I’m guessing you mean keep reading up the registers until you don’t get an error. 

Presumably there’s another meter block higher up which is not “empty”.

When (if) I find other blocks, how can I tell what each field (register) is?

Appreciate you taking the time to respond but I don’t think I’m up to your level of programming skills. The function node in NodeRed is about my limit. 

Regards,

Ian
+44 7768 215798

nmakel, 2020-12-19:
If all you're looking for is the beginning of meter and battery register addresses:

Meter 1: 40188
Meter 2: 40362
Meter 3: 40537
Battery 1: 57664
Battery 2: 57920

ianmidd, 2020-12-19:
That’s a start. Thank you Niels. 

I guess I’ll figure out the fields by inspection. 

Regards,

Ian
+44 7768 215798

nmakel, 2020-12-19:
> I guess I’ll figure out the fields by inspection.

They're in the code, for example lines 553 til 557:

https://github.com/nmakel/solaredge_modbus/blob/8ff60638865b28dc9d45435430d2fa4215227c8d/solaredge_modbus/__init__.py#L553-L557

The first value, e.g. `0xe142` is the address of a battery's `rated_energy`. Convert that to decimal, and it becomes register `57666`, with a length of `2`.

ianmidd, 2020-12-19:
I’m slowly getting the hang of this and your code is beautifully laid out but I am not that familiar with that language. 

I have used the Batt1 register you suggested and these are the first 100 registers from 57664.

[24,0,0,17864,16384,17742,16384,17742,16384,17742,16384,17742,0,0,0,0,0,0,0,0,17748,19539,65,0,0,0,20,0,0,0,0,0,0,0,0,0,17748,19539,65,0,0,0,20,0,0,16720,0,0,52429,17351,0,0,0,0,49260,100,0,0,56358,105,0,0,0,17864,26624,17835,26624,17067,13083,16509,6,0,3,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,17748,19539]

It looks like “24” is the Battery ID as this appears on the inventor display as well. The SunSpec document contains no decode of the battery fields. I also tried to figure it out using your code as clues but, alas, I’m not getting anywhere.

This is what I got but I’m sure I’m not reading your code properly.

{"B_DID":24,"item2":0,"item3":0,"B_Rated_Energy":2.503279576669853e-41}

Is there another document showing the battery mappings?

Regards,

Ian

nmakel, 2020-12-20:
> It looks like “24” is the Battery ID as this appears on the inventor display as well. The SunSpec document contains no decode of the battery fields. I also tried to figure it out using your code as clues but, alas, I’m not getting anywhere.

I'm not familiar with NodeRed, so I can only help you with parsing the registers using this library.

> Is there another document showing the battery mappings?

You'll want to download the model reference document found [here](https://sunspec.org/sunspec-information-model-reference/).

ianmidd, 2020-12-20:
I’m ok with NodeRed, it’s based on JavaScript as far as I know. 

Parsing the registers is exactly what I need help with.

I’m in contact with SunSpec and Solaredge regarding the additional mappings. 

Regards,

Ian
+44 7768 215798

closed 2020-12-24 by nmakel