nmakel/solaredge_modbus #19: Reading different units through same (master) inverter

binsentsu, 2020-12-23:
Hi,
I'm the creator of a home-assistant integration for solaredge through modbus. Based on some issues created in my repo, I went to search a better native library than my own...  One of the issues talked about in my repo is about connecting multiple slave units through one master unit. The slaves are accessed through IP/Port of the master, and I strongly doubt if for this case multiple connections would be allowed as it's also not the case for eg. meters/batteries . At this time you are already supporting a parent in order to share the modbus client and use a single TCP connection. But at the moment the unit of the parent is always taken over by the child. Do you think it would be an option that another Inverter, which acts as a modbus slave, can be instantiated with a parent (Master) Inverter argument, but without defacto sharing the unitId? I add the setup picture of the solaredge manual to make the situation a bit more clear:
[image]

nmakel, 2020-12-23:
I'm assuming from your description that the slave inverters are accessible simply by providing a different modbus address. Because I only have a single inverter I've never been able to test this. However, I've modified the parent logic to allow the unit parameter to be set even if a parent connection is provided. When instantiating a new inverter you will need to do the following:

```
  >>> inverter_1 = solaredge_modbus.Inverter(host="10.0.0.123", port=1502, unit=1)
  >>> inverter_2 = solaredge_modbus.Inverter(parent=inverter_1, unit=2)
```

I'd be interested to hear whether this works.

binsentsu, 2020-12-23:
That was fast :) I also don't have this setup at my place, but your change is exactly what I meant. Don't know what you mean with 'by providing different modbus address' as the slave inverters don't have their own IP. I'm planning to change the home-assistant integration and use this lib as dependency iso using the raw modbusclient and own register parsing. Your sunspec implementation seems pretty complete 👍

nmakel, 2020-12-23:
> Don't know what you mean with 'by providing different modbus address' as the slave inverters don't have their own IP.

I mean I'm curious whether providing a different `unit` is enough to get the registers of a slave inverter when using Modbus TCP.

binsentsu, 2020-12-23:
Should work. Have an open PR by a user who tested it. (https://github.com/binsentsu/home-assistant-solaredge-modbus/pull/12)

closed 2020-12-25 by nmakel