nmakel/solaredge_modbus #20: Example script doesnt output the phase inverter values

vipertje, 2020-12-24:
The line below in example.py doesnt work. Maybe convert the list of enumerations to key value like with C_SUNSPEC_DID_MAP
`if values['c_sunspec_did'] is solaredge_modbus.sunspecDID.THREE_PHASE_INVERTER:`

nmakel, 2020-12-24:
Can you elaborate on what isn't working? Are you getting an exception? Do you have a three phase inverter but aren't seeing the additional phase values?

vipertje, 2020-12-24:
Yes, so I indeed have a three phase inverter and below part of the information from when I print all values.

`'c_manufacturer': 'SolarEdge',
    'c_model': 'SE7K-RW0TEBNN4',
    'c_version': '0004.0010.0025',
    'c_sunspec_did': 103,`

In that if statement values['c_sunspec_did'] equals 103 and solaredge_modbus.sunspecDID.THREE_PHASE_INVERTER equals sunspecDID.THREE_PHASE_INVERTER so the outcome will be false and you will not get the three phase information

nmakel, 2020-12-24:
I see what you mean. Should be fixed now.

vipertje, 2020-12-24:
Yep that fixed it thanks :) Great repo, will use it to log my data!

closed 2020-12-25 by nmakel