nmakel/solaredge_modbus #21: Unidecode error on Windows connecting to SE4000 inverter

GWS65, 2021-01-07:
Hello,

Thanks for all your good work on this module. Just installed it in Windows 10 Python 3.8.7 (64 bits) and connected to my SolarEdge inverter. 

At start an error was received: 
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 12: invalid start byte
Changed line 247 in __init__.py and added the "ignore" parameter base don this link
https://stackoverflow.com/questions/42339876/error-unicodedecodeerror-utf-8-codec-cant-decode-byte-0xff-in-position-0-in

decoded = data.decode_string(length * 2).decode("utf-8", "ignore").replace("\x00", "").rstrip()

This resolved my issue and data is received from the SE4000 inverter. Will continue with uploading to influxdb and create a panel.

Hope this helps improving a future release
GWS

nmakel, 2021-01-07:
Thanks for letting me know, GWS. The whole string stripping UTF-8 bit happening there feels terribly hacky. I will have a look at the whole bit when I have a chance. Perhaps there is a more elegant solution.

nmakel, 2021-01-08:
The string decoding has been modified. Would you mind trying it?

GWS65, 2021-01-12:
> 
> 
> The string decoding has been modified. Would you mind trying it?

Not at all but is it already in this library? Looks like this one is not changed much. Could you please explain what has changed?
Might be missing something here. Not that familiar with the collaboration options on Github (yet :))

GWS65, 2021-01-12:
Hello Niels

Are you a dutchman by any chance?  Ik ook...

Just send me the file and I will give it a go.

Thanks,

GWS65

nmakel, 2021-01-12:
> > The string decoding has been modified. Would you mind trying it?
> 
> Not at all but is it already in this library? Looks like this one is not changed much. Could you please explain what has changed?
> Might be missing something here. Not that familiar with the collaboration options on Github (yet :))

Oops! Looks like I pushed the changes to my local repository and not Github. It has been uploaded now. The change is on line 249:

https://github.com/nmakel/solaredge_modbus/blob/2ef9fa91552caa8650b769b81cc57264e365adb1/solaredge_modbus/__init__.py#L249

GWS65, 2021-01-12:
Niels,

Still the same issue:
d:\Dvlp\Python\SE4000_ModbusData>python example.py 192.168.19.41 502
Traceback (most recent call last):
  File "example.py", line 37, in <module>
    battery_values = params.read_all()
  File
"C:\Users\GWScheppink\AppData\Local\Programs\Python\Python38\lib\site-packages\solaredge_modbus\__init__.py",
line 348, in read_all
    results.update(self._read_all(register_batch, rtype))
  File
"C:\Users\GWScheppink\AppData\Local\Programs\Python\Python38\lib\site-packages\solaredge_modbus\__init__.py",
line 316, in _read_all
    results[k] = self._decode_value(data, length, dtype, vtype)
  File
"C:\Users\GWScheppink\AppData\Local\Programs\Python\Python38\lib\site-packages\solaredge_modbus\__init__.py",
line 249, in _decode_value
*    decoded = data.decode_string(length * 2).decode()*
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 12:
invalid start byte

Changing it to :
decoded = data.decode_string(length * 2).decode("utf-8", "ignore")

Did the trick. (For my SE4000 at least).

Thanks and good luck with recoding. If I need to test more just let me know.

GWS65

nmakel, 2021-01-12:
You're using Python 3, right?

Also, would you mind trying:

`decoded = data.decode_string(length * 2).decode("ascii")`

The encoding of strings is not defined in SolarEdge's documentation, nor is there a standard way to encode strings in Modbus. It's an -- incorrect -- assumption on my part that the strings were UTF-8. Maybe `ascii` or `latin1` is a better choice.

GWS65, 2021-01-12:
Hello Niels,

Yes I am using Python3 and your solution did not work. Thinking about the
message. 0xFF is not in the Ascii range so it could be expected that
decoding is not valid.

d:\Dvlp\Python\SE4000_ModbusData>python example.py 192.168.19.41 502
Traceback (most recent call last):
  File "example.py", line 37, in <module>
    battery_values = params.read_all()
  File
"C:\Users\GWScheppink\AppData\Local\Programs\Python\Python38\lib\site-packages\solaredge_modbus\__init__.py",
line 348, in read_all
    results.update(self._read_all(register_batch, rtype))
  File
"C:\Users\GWScheppink\AppData\Local\Programs\Python\Python38\lib\site-packages\solaredge_modbus\__init__.py",
line 316, in _read_all
    results[k] = self._decode_value(data, length, dtype, vtype)
  File
"C:\Users\GWScheppink\AppData\Local\Programs\Python\Python38\lib\site-packages\solaredge_modbus\__init__.py",
line 249, in _decode_value
    decoded = data.decode_string(length * 2).decode("ascii")
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 12:
ordinal not in range(128)

decoded = data.decode_string(length * 2).decode("ascii"*, "ignore"*)
does workand text is as expected
and
decoded = data.decode_string(length * 2).decode("ignore")
does NOT work needs encoding parameter
utf-16 does kind of work (with ignore param too) but text is garbled.

:\Dvlp\Python\SE4000_ModbusData>python example.py 192.168.19.41 502
Inverter(192.168.19.41:502, connectionType.TCP: timeout=1, retries=3,
unit=0x1):

Output with ascii ignore
Registers:
        Manufacturer: SolarEdge
        Model: SE4000
        Type: Single Phase Inverter
        Version: 0003.2186
        Serial: 7311FF55
        Status: Sleeping
        Temperature: 11.68°C
        Current: 0.00A
        Voltage: 232.40V
        Frequency: 49.97Hz
        Power: 0.00W
        Power (Apparent): 0.00VA
        Power (Reactive): 0.00VAr
        Power Factor: 0.00%
        Total Energy: 7397778Wh
        DC Current: 0.00A
        DC Voltage: 0.10V
        DC Power: 0.00W

Gert-Wim Scheppink
gw@scheppink.com

nmakel, 2021-01-12:
Beats me. I've re-added the string replace and stripping, as this was causing a bunch of other issues. Also added `errors="ignore"`. Hopefully this solves your problem, and maybe somebody will come along who knows what's happening here.

GWS65, 2021-01-13:
Hello Niels,

At least this works but I have no idea why ascii did not.  Used modpoll.exe
to read the registers in another way and this is what the SE4000 shows.

c:\Bin\ModPoll_v3.4\win32>modpoll -m tcp -a 1 -r 40001 -c 4 -t 4:hex -1
192.168.19.41
modpoll 3.4 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2013 proconX Pty Ltd
Visit http://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: MODBUS/TCP
Slave configuration...: address = 1, start reference = 40001, count = 4
Communication.........: 192.168.19.41, port 502, t/o 1.00 s, poll rate 1000
ms
Data type.............: 16-bit register (hex), output (holding) register
table

-- Polling slave...
[40001]: 0x5375
[40002]: 0x6E53
[40003]: 0x0001
[40004]: 0x0041

c:\Bin\ModPoll_v3.4\win32>modpoll -m tcp -a 1 -r 40005 -c 16 -t 4:hex -1
192.168.19.41
modpoll 3.4 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2013 proconX Pty Ltd
Visit http://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: MODBUS/TCP
Slave configuration...: address = 1, start reference = 40005, count = 16
Communication.........: 192.168.19.41, port 502, t/o 1.00 s, poll rate 1000
ms
Data type.............: 16-bit register (hex), output (holding) register
table

-- Polling slave...
[40005]: 0x536F
[40006]: 0x6C61
[40007]: 0x7245
[40008]: 0x6467
[40009]: 0x6520
[40010]: 0x0000
[40011]: 0x0000
[40012]: 0x0000
[40013]: 0x0000
[40014]: 0x0000
[40015]: 0x0000
[40016]: 0x0000
[40017]: 0x0000
[40018]: 0x0000
[40019]: 0x0000
[40020]: 0x0000

from address 40005 I read "SolarEdge" as expected (
https://www.rapidtables.com/convert/number/hex-to-ascii.html)
Is it possible that you are one byte "off" since many modbus implementation
have the option of choosing the register addressing mode starting with 0 or
1

GWS65

closed 2021-04-21 by nmakel