nmakel/solaredge_modbus #22: TCP Connection issues

Bonn93, 2021-01-21:
Hello,

I've packaged this into a docker container, and using the influx script which has been working great for testing. I've noticed some weird things happening when connecting to an inverter at the TCP layer. 

When I run the image on Ubuntu/RHEL, I get a TCP Socket error and the script crashes. But running on WSL ( still linux ) the script executes for a while, and then crashes. I think there's some optimizations needed at the TCP later. 

Using wireshark/pfsense, I've observed;

* WSL2/Windows10/Ubuntu/Docker - there's a lot of TCP retransmits, and TCP port number re-use
* Linux there's a lot of FIN/ACKS is a row

Is there a way to TCP keepalive the connection? It looks very chatty and perhaps the modbus servers/inverters can't handle this? Just a hunch. This is the error received when running on plain linux, some googling found there may be async methods?

File "/usr/local/lib/python3.9/site-packages/pymodbus-2.5.0rc2-py3.9.egg/pymodbus/client/sync.py", line 108, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(10.5.90.2:1502)]

nmakel, 2021-01-21:
I can't reproduce the socket error, TCP retransmits, nor the FIN/ACKs. I have the library running in multiple scripts for months on end now on a debian 10 system. I see you're using a release candidate version of pymodbus, have you tried using the 2.4.0 release?

Bonn93, 2021-01-21:
2.4.0 works! But I had to change the requirements.txt to

pymodbus==2.4.0, it was still pulling 2.5 RC versions... 

Dockerfile now
```
FROM python:3.7.9

RUN mkdir /modbus
WORKDIR /modbus
ADD . /modbus

RUN pip install influxdb
RUN pip install -r requirements.txt

WORKDIR /modbus
CMD ["/bin/bash", "-c", "python3 example_influxdb.py ${INVERTER_IP} ${INVERTER_PORT} --influx_host ${INFLUX_HOST} --influx_port ${INFLUX_PORT}  --influx_db ${INFLUX_DB} --influx_user ${INFLUX_USER} --influx_pass ${INFLUX_PASSWORD}"]
```

Requirements.txt
`pymodbus==2.4.0`

Might need to keep an eye on the 2.5~ release! 

Thanks for your help!

nmakel, 2021-01-21:
> 2.4.0 works! But I had to change the requirements.txt to

Thanks for letting me know. If I have a bit of time I'll have a look at what's happening with 2.5.

nmakel, 2021-02-06:
Had a quick look at reproducing the issue. Using pymodbus 2.5.0rc3 gives me similar results to yours. I've added a quick fix which makes things work ok with 2.5.0, but it isn't pretty. For some reason the connection is reset when issuing multiple read_holding_register commands in quick succession. Diffs between 2.4.0 and 2.5.0rc3 don't show anything obvious that could be causing this.

The 3.0.0 branch works fine, however. Will revisit this when 2.5.0 is released.

closed 2021-04-21 by nmakel