nmakel/solaredge_modbus #23: Cannot use module, example not working.

pczolee, 2021-01-25:
Hello.

The system is  Debian 9.13 x64, Python 3.5.3
I want to monitor my inverter with domoticz. The https://github.com/addiejanssen/domoticz-solaredge-modbustcp-plugin depend on your program. So I installed it with pip, but it is not working:

 2021-01-25 16:50:04.732 Error: (SolarEdge_ModbusTCP) failed to load 'plugin.py', Python Path used was '/opt/domoticz/plugins/domoticz-solaredge-modbustcp-plugin/:/usr/lib/python35.zip:/usr/lib/python3.5:/usr/lib/python3.5/plat-x86_64-linux-gnu:/usr/lib/python3.5/lib-dynload'.
2021-01-25 16:50:04.732 Error: (SolarEdge Inverter) Module Import failed, exception: 'ImportError'
2021-01-25 16:50:04.732 Error: (SolarEdge Inverter) Module Import failed: ' Name: solaredge_modbus'
2021-01-25 16:50:04.732 Error: (SolarEdge Inverter) Error Line details not available. 

To investigate what happening I followed instruction and installed it from your git repo. The problem is same, but now I have your example.py, so I tried to run that:

# ./example.py
  File "./example.py", line 43
    print(f"{inverter}:")
                       ^
SyntaxError: invalid syntax

If I remove the "f" from there, there coming another message:

Traceback (most recent call last):
  File "./example.py", line 6, in <module>
    import solaredge_modbus

So I think the problem is that solaredge_modbus not found.

# pip3 list
acme (0.28.0)
ardexaplugin (0.2.3)
asn1crypto (0.24.0)
bitstring (3.1.7)
certbot (0.28.0)
certifi (2016.2.28)
chardet (3.0.4)
click (7.1.2)
ConfigArgParse (0.11.0)
configobj (5.0.6)
cryptography (2.3)
ecdsa (0.16.1)
esptool (3.0)
future (0.15.2)
idna (2.6)
iotop (0.6)
josepy (1.1.0)
lxml (3.7.1)
mock (2.0.0)
natsort (4.0.3)
netifaces (0.10.4)
parsedatetime (2.4)
pbr (4.2.0)
pip (9.0.1)
pymodbus (2.4.0)
pyModbusTCP (0.1.8)
pyOpenSSL (19.0.0)
pyRFC3339 (1.0)
pyserial (3.5)
python-apt (1.4.3)
pythondialog (3.4.0)
pytz (2016.7)
pyudev (0.21.0)
reedsolo (1.5.4)
requests (2.21.0)
requests-toolbelt (0.7.0)
setuptools (33.1.1)
six (1.15.0)
solaredge-modbus (0.6.4)
urllib3 (1.24.1)
zope.component (4.3.0)
zope.event (4.2.0)
zope.hookable (4.0.4)
zope.interface (4.3.2)

What can I do to use this program?

Thanks

nmakel, 2021-01-25:
This library requires python 3.6 or later. You can try updating python using [debian backports](https://backports.debian.org/), upgrading to debian 10, or installing python manually.

pczolee, 2021-01-25:
Thanks. Are there any older release which working without python update?

nmakel, 2021-01-25:
I'm afraid not. Python 3.5 is end of life and will not be supported.

pczolee, 2021-01-25:
Ok, I try to update. Thanks

closed 2021-01-25 by nmakel