nmakel/solaredge_modbus #24: Connection showing true regardless of Unit #  and possible RTS issue?

pablo-smith, 2021-02-09:
Hey Niels,

First off great work on this library, this is saving me a ton of work and this is a somewhat new world for me (MODBUS). So this helps get me off on a great start. 

I modified your example script for RS485 RTU implementation. I am connecting to a SolarEdge SE 3800H. I am on the RS485-2 and using the app I have everything configured correctly as according to technical notes document you have referenced in prior post. I added the connect method and got it to reply true. The device ID is set to 1 (via the app), I have changed it to 2 and changed the command line argument to see if it ever returns false. Regardless, it always returns true (this might happen regardless if the unit is wrong or right). From what I can tell its similar to this gentlemen's [post](https://github.com/nmakel/solaredge_modbus/issues/4).

I am on beaglebone (dont think it matters) and am using a PMOD RS485 to UART adapter [link](https://store.digilentinc.com/pmod-rs485-high-speed-isolated-communication/). I think I might be having RTS/flow control issues since this chip requires me to toggle the flow control. And since I can only go low on RTS (via a GPIO toggle) after the inverter read call, I think I need to get a chip that controls this automatically. [like this](https://www.amazon.com/gp/product/B075V2NMV8/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1) Otherwise, I would have to hack the modbus library to handle this (to my knowledge). Since I can not toggle the pin low until after the read call returns, I think this is something period that needs to be changed. 

Here is the log output below, take note the first line (connection status) and last line (the false return on a single register read). 

**debian@beaglebone:~$ python3 example.py /dev/ttyO4 --baud 115200 --stopbits 1 --parity N --timeout 2 --unit 2
Is inverter connected: True
2021-02-09 15:27:13 DEBUG: Current transaction state - IDLE
2021-02-09 15:27:13 DEBUG: Running transaction 1
2021-02-09 15:27:13 DEBUG: SEND: 0x2 0x3 0x9c 0x54 0x0 0x10 0x2b 0xb5
2021-02-09 15:27:13 DEBUG: New Transaction state 'SENDING'
2021-02-09 15:27:13 DEBUG: Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2021-02-09 15:27:15 DEBUG: Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)) 
2021-02-09 15:27:15 DEBUG: Frame - [b''] not ready
2021-02-09 15:27:15 DEBUG: Getting transaction 2
2021-02-09 15:27:15 DEBUG: Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2021-02-09 15:27:15 DEBUG: Current transaction state - TRANSACTION_COMPLETE
2021-02-09 15:27:15 DEBUG: Running transaction 2
2021-02-09 15:27:15 DEBUG: SEND: 0x2 0x3 0x9c 0x54 0x0 0x10 0x2b 0xb5
2021-02-09 15:27:15 DEBUG: Changing state to IDLE - Last Frame End - None, Current Time stamp - 1612906035.144507
2021-02-09 15:27:15 DEBUG: New Transaction state 'SENDING'
2021-02-09 15:27:15 DEBUG: Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2021-02-09 15:27:17 DEBUG: Incomplete message received, Expected 37 bytes Recieved 0 bytes !!!!
2021-02-09 15:27:17 DEBUG: Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2021-02-09 15:27:17 DEBUG: RECV: 
2021-02-09 15:27:17 DEBUG: Frame - [b''] not ready
2021-02-09 15:27:17 DEBUG: Getting transaction 2
2021-02-09 15:27:17 DEBUG: Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2021-02-09 15:27:17 DEBUG: Current transaction state - TRANSACTION_COMPLETE
2021-02-09 15:27:17 DEBUG: Running transaction 3
2021-02-09 15:27:17 DEBUG: SEND: 0x2 0x3 0x9c 0x54 0x0 0x10 0x2b 0xb5
2021-02-09 15:27:17 DEBUG: Changing state to IDLE - Last Frame End - 1612906037.156145, Current Time stamp - 1612906037.177895
2021-02-09 15:27:17 DEBUG: New Transaction state 'SENDING'
2021-02-09 15:27:17 DEBUG: Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2021-02-09 15:27:19 DEBUG: Incomplete message received, Expected 37 bytes Recieved 0 bytes !!!!
2021-02-09 15:27:19 DEBUG: Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2021-02-09 15:27:19 DEBUG: RECV: 
2021-02-09 15:27:19 DEBUG: Frame - [b''] not ready
2021-02-09 15:27:19 DEBUG: Getting transaction 2
2021-02-09 15:27:19 DEBUG: Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
{'c_model': False}**

Any help or guidance be appreciated. 

Thank you in advance.

nmakel, 2021-02-10:
> I added the connect method and got it to reply true.

The pymodbus library will return the state of the socket, which in this case is always true irrespective of the presence or correct Modbus address of any device on the other end.

> Here is the log output below, take note the first line (connection status) and last line (the false return on a single register read).

Looks like no device is replying to the requests. Verify the Modbus address of your inverter, whether RS485-2 is configured as a Sunspec Logger device, and the polarity of the connected wires -- assuming the RS485 adapter is known to work.

pablo-smith, 2021-02-11:
Yes indeed. I went and ordered an RS485 to TTL converter 2 days ago and that takes care of the direction enable on its own. I can see on my scope it chatting a lot more than it did. So that is good news. It looks to be responding but on unit 1 it fails the CRC, and running as unit 2 (referencing that other gentleman's post) seemed to do more but still got nothing. For sure closer than I ever have been. If you have any other tidbits of advice to offer please do. 

Below are the logs from running as unit 1, and unit 2. Thank you in advance for helping me out. 

***debian@beaglebone:~$ python3 example.py /dev/ttyO4 --baud 115200 --stopbits 1 --parity N --timeout 2 --unit 1
Is inverter connected: True
2021-02-11 09:48:14 DEBUG: Current transaction state - IDLE
2021-02-11 09:48:14 DEBUG: Running transaction 1
2021-02-11 09:48:14 DEBUG: SEND: 0x1 0x3 0x9c 0x44 0x0 0x41 0xeb 0xbf
2021-02-11 09:48:14 DEBUG: New Transaction state 'SENDING'
2021-02-11 09:48:14 DEBUG: Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2021-02-11 09:48:14 DEBUG: Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2021-02-11 09:48:14 DEBUG: RECV: 0x1 0x3 0x82 0x42 0x6e 0x4c 0x41 0x70 0x45 0x64 0x66 0x60 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x33 0x38 0x30 0x30 0x48 0x2d 0x55 0x53 0x30 0x30 0x30 0x42 0x4e 0x43 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x34 0xe 0x30 0x30 0x31 0x31 0x2c 0x20 0x30 0x33 0x20 0x0 0x0 0x37 0x33 0x21 0x44 0x31 0x2 0x38 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x52 0x41
2021-02-11 09:48:14 DEBUG: CRC invalid, discarding header!!
2021-02-11 09:48:14 DEBUG: Resetting frame - Current Frame in buffer - 0x1 0x3 0x82 0x42 0x6e 0x4c 0x41 0x70 0x45 0x64 0x66 0x60 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x33 0x38 0x30 0x30 0x48 0x2d 0x55 0x53 0x30 0x30 0x30 0x42 0x4e 0x43 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x34 0xe 0x30 0x30 0x31 0x31 0x2c 0x20 0x30 0x33 0x20 0x0 0x0 0x37 0x33 0x21 0x44 0x31 0x2 0x38 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x52 0x41
2021-02-11 09:48:14 DEBUG: Frame check failed, ignoring!!
2021-02-11 09:48:14 DEBUG: Resetting frame - Current Frame in buffer - 
2021-02-11 09:48:14 DEBUG: Getting transaction 1
2021-02-11 09:48:14 DEBUG: Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2021-02-11 09:48:14 DEBUG: Current transaction state - TRANSACTION_COMPLETE
2021-02-11 09:48:14 DEBUG: Running transaction 2
2021-02-11 09:48:14 DEBUG: SEND: 0x1 0x3 0x9c 0x44 0x0 0x41 0xeb 0xbf
2021-02-11 09:48:14 DEBUG: Changing state to IDLE - Last Frame End - 1613058494.968618, Current Time stamp - 1613058494.996114
2021-02-11 09:48:14 DEBUG: New Transaction state 'SENDING'
2021-02-11 09:48:15 DEBUG: Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2021-02-11 09:48:15 DEBUG: Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2021-02-11 09:48:15 DEBUG: RECV: 0x1 0x3 0x82 0x53 0x6f 0x6c 0x61 0x72 0x5 0x60 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x52 0x45 0x33 0x38 0x30 0x30 0x8 0xc 0x55 0x53 0x30 0x30 0x30 0x42 0x4e 0x43 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x20 0x20 0x30 0x34 0x2c 0x20 0x30 0x31 0x31 0x2e 0x30 0x20 0x33 0x30 0x0 0x0 0x37 0x33 0x31 0x45 0x39 0x42 0x38 0x21 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x52 0x41
2021-02-11 09:48:15 DEBUG: CRC invalid, discarding header!!
2021-02-11 09:48:15 DEBUG: Resetting frame - Current Frame in buffer - 0x1 0x3 0x82 0x53 0x6f 0x6c 0x61 0x72 0x5 0x60 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x52 0x45 0x33 0x38 0x30 0x30 0x8 0xc 0x55 0x53 0x30 0x30 0x30 0x42 0x4e 0x43 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x20 0x20 0x30 0x34 0x2c 0x20 0x30 0x31 0x31 0x2e 0x30 0x20 0x33 0x30 0x0 0x0 0x37 0x33 0x31 0x45 0x39 0x42 0x38 0x21 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x52 0x41
2021-02-11 09:48:15 DEBUG: Frame check failed, ignoring!!
2021-02-11 09:48:15 DEBUG: Resetting frame - Current Frame in buffer - 
2021-02-11 09:48:15 DEBUG: Getting transaction 1
2021-02-11 09:48:15 DEBUG: Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2021-02-11 09:48:15 DEBUG: Current transaction state - TRANSACTION_COMPLETE
2021-02-11 09:48:15 DEBUG: Running transaction 3
2021-02-11 09:48:15 DEBUG: SEND: 0x1 0x3 0x9c 0x44 0x0 0x41 0xeb 0xbf
2021-02-11 09:48:15 DEBUG: Changing state to IDLE - Last Frame End - 1613058495.058434, Current Time stamp - 1613058495.085291
2021-02-11 09:48:15 DEBUG: New Transaction state 'SENDING'
2021-02-11 09:48:15 DEBUG: Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
[... cut ...]
2021-02-11 09:48:47 DEBUG: New Transaction state 'SENDING'
2021-02-11 09:48:47 DEBUG: Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2021-02-11 09:48:49 DEBUG: Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)) 
2021-02-11 09:48:49 DEBUG: Frame - [b''] not ready
2021-02-11 09:48:49 DEBUG: Getting transaction 2
2021-02-11 09:48:49 DEBUG: Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
{}***

nmakel, 2021-02-11:
Sounds like a framing issue. Try adding `framer=ModbusRtuFramer` to the client.
https://github.com/nmakel/solaredge_modbus/blob/d26891c02efaabd6408eb9663296e9d4a9e115e0/solaredge_modbus/__init__.py#L199-L201

Becomes:

```
                self.client = ModbusSerialClient(
                    method="rtu",
                    framer=ModbusRtuFramer
                    port=self.device,
```

pablo-smith, 2021-02-11:
Got it.

I applied your changes: Here is the output:

`Traceback (most recent call last):
  File "example.py", line 42, in <module>
    unit=args.unit
  File "/usr/local/lib/python3.7/dist-packages/solaredge_modbus/__init__.py", line 366, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/solaredge_modbus/__init__.py", line 206, in __init__
    timeout=self.timeout)
  File "/usr/local/lib/python3.7/dist-packages/pymodbus/client/sync.py", line 547, in __init__
    **kwargs)
TypeError: __init__() got multiple values for argument 'framer'`

If I leave off the comma I get a syntax error. 
  
`Traceback (most recent call last):
  File "example.py", line 6, in <module>
    import solaredge_modbus
  File "/usr/local/lib/python3.7/dist-packages/solaredge_modbus/__init__.py", line 202
    port=self.device,
       ^
SyntaxError: invalid syntax
`

I did notice the GitHub committed version and mine (obtained via pip) were not exactly the same. (no import time, at the top) I did copy paste that version and tried the change to make sure there was not something else missing (same outcome has above). Appreciate the help.

nmakel, 2021-02-11:
> `Traceback (most recent call last): File "example.py", line 42, in <module> unit=args.unit File "/usr/local/lib/python3.7/dist-packages/solaredge_modbus/__init__.py", line 366, in __init__ [...]

Ok, so it's not an incorrect framing issue. 'method=rtu' implies the framer argument. Never mind.

> If I leave off the comma I get a syntax error.

Yeah, my bad. Needs a comma at the end of the line.

> I did notice the GitHub committed version and mine (obtained via pip) were not exactly the same.

No real difference other than an extra check for an active connection.

Not quite sure what is causing your issues, though. With a unit of 1 the inverter looks to be responding correctly, but you should be able to verify the modbus address of the inverter in the setapp interface. You could try reducing the baud rate (remember to set both sides), perhaps the timing is off.

pablo-smith, 2021-02-23:
Good day,

Apologies for the delay in getting back back with you. We had some bad weather and were unable to get into the office.

The lower baud rate worked! It might be a hardware issue with our TTL to RS485 board. As I am writing this I iterated through all the baud rates and now 115200 works. Now I am slightly further perplexed by this. 

Your welcome to close the issue. If anything I be curious to hear any last thoughts on this final quirk, I will do a power cycle on everything to see it the problem resets itself. The default is 115200 and it be nice to not worry about that being set right (it seems to reset back on power cycles). Greatly appreciate your help on this.

Thanks!

nmakel, 2021-02-23:
> The lower baud rate worked! It might be a hardware issue with our TTL to RS485 board. As I am writing this I iterated through all the baud rates and now 115200 works. Now I am slightly further perplexed by this.

Great to hear it's working.
 
> Your welcome to close the issue. If anything I be curious to hear any last thoughts on this final quirk, I will do a power cycle on everything to see it the problem resets itself. The default is [...]

Two thoughts that might point you in the right direction. Are you sure the default power on behavior of your RS485 adapter is actually 115200? If you have success after setting the baud to 115200 _after_ trying a few other baud rates, it might just be that it powers on to another baud rate. If that doesn't help, the only other easy to verify part of the set-up is your cabling and environment. Proper twisted pair? Grounding connected? Length within specification? Are you operating in temperature extremes (outside in the sun, unventilated attic?) that may influence conductivity?

closed 2021-04-21 by nmakel