nmakel/solaredge_modbus #28: readout data not correct?

Deepdiigger, 2021-04-04:
hello, i am new to python and influx.
However i got everything running and setup a Chronograf chart to verify the readouts.

I wanted to check them against the API Data.

what i noticed is, that all data from the meter seems to be as expected and ok.
However i could not find the actual production of the solaredge PV in the meters section so i did look in the inverter section.
I think this should be power_ac. But i do not understand this. the values range between 3k and 30k. however i have a max 10,5 kW Peak Produktion and when the API says i have 7k Production power_ac shows. 3k.  Even more when the API Production says 3k (less than the 7k) power_ac says 28k so more than before by the factor 10. Any hints? i did not modify your script at all.

nmakel, 2021-04-04:
Have you tried using only `example.py`? Do those values match those in the SolarEdge mobile application?

If you're a factor 10 off, it sounds like you aren't applying the scaling factor to the `power_ac` value. `example_influxdb.py` should take care of this for you.

Deepdiigger, 2021-04-05:
i used the example_influxdb.py.
attached you can see a fotos. Today it seems as something happened at 16:00 the values before were reasonable and probably good also concerning the right scale.
i dont know what this could be. I did not modify the example_influxdb.py except hosts and ports.

[image]

nmakel, 2021-04-05:
Please try 'example.py' to verify the values being returned by the inverter.

Deepdiigger, 2021-04-05:
my first guess is, that the data directly read out with example.py is correct.

Seems like example_influxdb.py produces the error for me. maybe it has something to do with the calculations inside. 

Attached a screenshot from the influxdb data.
Can this be correct? meter.power = the same as import from grid(This seems not ok, shouldnt be Meterpower Total Power)?
 Self consumption is higher by the production(This seems ok).

[image]

nmakel, 2021-04-05:
Looks like scaling of meter values was not being done correctly. Please try the new [`example_influxdb.py`](https://github.com/nmakel/solaredge_modbus/blob/master/example_influxdb.py) I just uploaded. Let me know if the values make more sense now.

Deepdiigger, 2021-04-09:
after another day of read outs i think i found the problem.

_meter_values["power_selfconsumption"] = inverter_power_ac - meter_power_

"power_selfconsumption" from my understanding should be the Power the inverter produces and is directly used. Excess Power is going to the grid.

(inverter_power_ac - meter_power) however, to my understanding should be Total Consumption (Selfconsumption + import from grid)

So my fix would be 
1. relabeling power_selfconsumption to power_consumption  
2. power_selfconsumption = min(inverter_power_ac , power_consumption)

i just started programing with python only 2 weeks ago. So please excuse any rookie mistakes.

nmakel, 2021-04-09:
Thanks for letting me know the values look better now.

Regarding the self consumption and import/export calculations: I've removed them from the example script. They shouldn't have been in there to begin with as they're not obtained from the inverter, meter or battery. They should be computed based on the values that *are* returned. Properly computing the values in turn depends on the topology of the meter(s), and other complicating factors such as the presence of batteries. I will leave this up to the user to implement in either the Python glue, or presentation layer.

In the following example the kWh meter is installed in such a way that importing power is reported as *positive* meter power, while excess power is reported as *negative*. Self consumption is equal to your `inverter_power_ac` when `meter_power` is positive (and therefore importing from the grid) and `inverter_power_ac + meter_power` when `meter_power` is negative (and in that case exporting excess `inverter_power_ac` to the grid). Total consumption is always equal to `inverter_power_ac + meter_power`.

Translated roughly to python that will work in `example_influxdb.py`:

```
            inverter_power_ac = values["power_ac"] * (10 ** values["power_ac_scale"])
            meter_power = meter_values["power"] * (10 ** meter_values["power_scale"])
            meter_values["power_consumption"] = inverter_power_ac + meter_power

            # exporting to grid
            if meter_power < 0:
                meter_values["power_export_to_grid"] = float(abs(meter_power))
                meter_values["power_self_consumption"] = inverter_power_ac + meter_power
            else:
                meter_values["power_export_to_grid"] = float(0)

            # importing from grid
            if meter_power > 0:
                meter_values["power_import_from_grid"] = float(meter_power)
                meter_values["power_self_consumption"] = inverter_power_ac
            else:
                meter_values["power_import_from_grid"] = float(0)
```

closed 2021-04-21 by nmakel

fredlcore, 2021-07-22:
@nmakel and @Deepdiigger: I don't want to open a new issue for this because I basically have the same problem that @Deepdiigger had (at least in the beginning). I'm looking for the actual energy production in kWh of the SolarEdge PV modules. This is the output of `example.py` at half an hour after midnight:
```
 	Manufacturer: SolarEdge
	Model: SE5K-RWS48BNN4
	Type: Three Phase Inverter
	Version: 0004.0013.0038
	Serial: 7E047E6D
	Status: Producing
	Temperature: 43.97°C
	Current: 2.25A
	Phase 1 Current: 0.71A
	Phase 2 Current: 0.77A
	Phase 3 Current: 0.76A
	Phase 1 voltage: 410.70V
	Phase 2 voltage: 410.50V
	Phase 3 voltage: 411.20V
	Phase 1-N voltage: 237.20V
	Phase 2-N voltage: 236.70V
	Phase 3-N voltage: 237.70V
	Frequency: 50.02Hz
	Power: 213.00W
	Power (Apparent): 533.30VA
	Power (Reactive): -489.00VAr
	Power Factor: -39.90%
	Total Energy: 5710Wh
	DC Current: 0.26A
	DC Voltage: 816.30V
	DC Power: 216.24W
```
I couldn't figure out which of these power values refer to the current energy production, so I waited until it is pitch dark here (half past midnight) and the API shows 0 kWh in the monitoring web interface. However, none of the above values is zero (or at least near zero). During the day I also couldn't get closer than a few hundred Wh to what the API says. Also, total energy in the web interface is 4.91 kWh and here it is 5.71 kWh. As I can't say for sure that the API values are 100% correct, I wouldn't mind a bit of a difference, but a factor of 10-20% is a bit too much.

Do you have any idea what is wrong here?

fredlcore, 2021-07-23:
And for comparison, this is the output at 7:30am with sunshine on a clear-blue sky where the API currently reports 0.7kWh:
```
	Manufacturer: SolarEdge
	Model: SE5K-RWS48BNN4
	Type: Three Phase Inverter
	Version: 0004.0013.0038
	Serial: 7E047E6D
	Status: Producing
	Temperature: 43.84°C
	Current: 2.19A
	Phase 1 Current: 0.71A
	Phase 2 Current: 0.75A
	Phase 3 Current: 0.73A
	Phase 1 voltage: 406.40V
	Phase 2 voltage: 405.50V
	Phase 3 voltage: 406.10V
	Phase 1-N voltage: 234.50V
	Phase 2-N voltage: 234.20V
	Phase 3-N voltage: 234.70V
	Frequency: 49.96Hz
	Power: 174.00W
	Power (Apparent): 513.30VA
	Power (Reactive): -483.00VAr
	Power Factor: -33.79%
	Total Energy: 7045Wh
	DC Current: 0.22A
	DC Voltage: 815.90V
	DC Power: 176.64W
```

nmakel, 2021-07-23:
Would you mind posting the output of `example.py --json` _and_ `example.py` at about the same time? Those values you posted are odd to say the least.

fredlcore, 2021-07-23:
Sure, here you go (I named examply.py modbus.py, but it's one and the same script):
```
Air-17475:~ frederik$ ./modbus.py 192.168.1.132 502
Inverter(192.168.1.132:502, connectionType.TCP: timeout=1, retries=3, unit=0x1):

Registers:
	Manufacturer: SolarEdge
	Model: SE5K-RWS48BNN4
	Type: Three Phase Inverter
	Version: 0004.0013.0038
	Serial: 7E047E6D
	Status: Producing
	Temperature: 44.84°C
	Current: 10.14A
	Phase 1 Current: 3.36A
	Phase 2 Current: 3.39A
	Phase 3 Current: 3.38A
	Phase 1 voltage: 411.70V
	Phase 2 voltage: 411.80V
	Phase 3 voltage: 413.00V
	Phase 1-N voltage: 237.60V
	Phase 2-N voltage: 236.80V
	Phase 3-N voltage: 239.50V
	Frequency: 50.02Hz
	Power: 2412.00W
	Power (Apparent): 2421.60VA
	Power (Reactive): -216.00VAr
	Power Factor: -99.58%
	Total Energy: 8623Wh
	DC Current: 3.00A
	DC Voltage: 816.10V
	DC Power: 2448.70W
Air-17475:~ frederik$ ./modbus.py --json 192.168.1.132 502
{
    "c_manufacturer": "SolarEdge",
    "c_model": "SE5K-RWS48BNN4",
    "c_version": "0004.0013.0038",
    "c_serialnumber": "7E047E6D",
    "c_deviceaddress": 1,
    "c_sunspec_did": 103,
    "current": 957,
    "p1_current": 319,
    "p2_current": 319,
    "p3_current": 317,
    "current_scale": -2,
    "p1_voltage": 4109,
    "p2_voltage": 4112,
    "p3_voltage": 4123,
    "p1n_voltage": 2376,
    "p2n_voltage": 2360,
    "p3n_voltage": 2386,
    "voltage_scale": -1,
    "power_ac": 22730,
    "power_ac_scale": -1,
    "frequency": 5001,
    "frequency_scale": -2,
    "power_apparent": 22828,
    "power_apparent_scale": -1,
    "power_reactive": -21200,
    "power_reactive_scale": -2,
    "power_factor": -9955,
    "power_factor_scale": -2,
    "energy_total": 8629,
    "energy_total_scale": 0,
    "current_dc": 28279,
    "current_dc_scale": -4,
    "voltage_dc": 8160,
    "voltage_dc_scale": -1,
    "power_dc": 23076,
    "power_dc_scale": -1,
[... 93 lines cut ...]
            "c_sunspec_did": 0,
            "rated_energy": 4000.0,
            "maximum_charge_continuous_power": 3300.0,
            "maximum_discharge_continuous_power": 3300.0,
            "maximum_charge_peak_power": 1200.0,
            "maximum_discharge_peak_power": 3900.0,
            "average_temperature": 28.0,
            "maximum_temperature": 0.0,
            "instantaneous_voltage": 818.6364135742188,
            "instantaneous_current": 0.4083080589771271,
            "instantaneous_power": -286.0,
            "lifetime_export_energy_counter": 58,
            "lifetime_import_energy_counter": 3187,
            "maximum_energy": 4000.0,
            "available_energy": 4000.0,
            "soh": 100.0,
            "soe": 98.0,
            "status": 4,
            "status_internal": 5,
            "event_log": 0,
            "event_log_internal": 0
        }
    }
}
```

fredlcore, 2021-07-23:
Current PV production is around 1.9kW

nmakel, 2021-07-23:
> Current PV production is around 1.9kW and total power consumption in the house around 2.3kW. power_dc and/or power_ac seem to reflect that, also before when the washing machine was not running and consumption was around 0.2kW.

The values you just posted seem entirely reasonable, or am I missing something?

fredlcore, 2021-07-23:
Thanks for your quick reply, yes, the power consumption values are reasonable, but there is no parameter which reflects the actual PV production (1.9 kW at this point in time). power_ac and power_dc are the power *consumption*, that is why it is around 0.2kW at half past midnight. Or have I overseen some kind of conversion that will get me to 1.9kW somehow?

nmakel, 2021-07-23:
In your latest post the PV installation is producing at 2.27kW AC, 2.3kW DC. This matches the current and voltage values. Your kWh meter is registering import of 682W and 732W on phases 1 and 3, while phase 2 is exporting at 1732W. Are you sure the directionality of the kWh meter is correct?

fredlcore, 2021-07-23:
Sorry if I'm a bit slow, but here's another output:
```
Air-17475:~ frederik$ ./modbus.py --json 192.168.1.132 502
{
    "c_manufacturer": "SolarEdge",
    "c_model": "SE5K-RWS48BNN4",
    "c_version": "0004.0013.0038",
    "c_serialnumber": "7E047E6D",
    "c_deviceaddress": 1,
    "c_sunspec_did": 103,
    "current": 1723,
    "p1_current": 575,
    "p2_current": 576,
    "p3_current": 571,
    "current_scale": -2,
    "p1_voltage": 4144,
    "p2_voltage": 4132,
    "p3_voltage": 4135,
    "p1n_voltage": 2381,
    "p2n_voltage": 2398,
    "p3n_voltage": 2388,
    "voltage_scale": -1,
    "power_ac": 4125,
    "power_ac_scale": 0,
    "frequency": 4996,
    "frequency_scale": -2,
    "power_apparent": 4126,
    "power_apparent_scale": 0,
    "power_reactive": -12600,
    "power_reactive_scale": -2,
    "power_factor": -9992,
    "power_factor_scale": -2,
    "energy_total": 10123,
    "energy_total_scale": 0,
    "current_dc": 5133,
    "current_dc_scale": -3,
    "voltage_dc": 8157,
    "voltage_dc_scale": -1,
    "power_dc": 4187,
    "power_dc_scale": 0,
    "temperature": 4638,
    "temperature_scale": -2,
    "status": 4,
    "vendor_status": 0,
    "meters": {
        "Meter1": {
            "c_manufacturer": "SolarEdge",
            "c_model": "SE-MTR-3Y-400V-A",
            "c_option": "Export+Import",
            "c_version": "74",
            "c_serialnumber": "606641685",
            "c_deviceaddress": 1,
            "c_sunspec_did": 203,
            "current": 162,
            "p1_current": 55,
            "p2_current": 51,
            "p3_current": 55,
            "current_scale": -1,
            "voltage_ln": 23849,
            "p1n_voltage": 23849,
            "p2n_voltage": 24003,
            "p3n_voltage": 23870,
            "voltage_ll": 0,
            "p12_voltage": 0,
            "p23_voltage": 0,
            "p31_voltage": 0,
            "voltage_scale": -2,
            "frequency": 4997,
            "frequency_scale": -2,
            "power": 3824,
            "p1_power": 1324,
            "p2_power": 1178,
            "p3_power": 1322,
            "power_scale": 0,
            "power_apparent": 3836,
            "p1_power_apparent": 1324,
            "p2_power_apparent": 1205,
            "p3_power_apparent": 1322,
            "power_apparent_scale": 0,
            "power_reactive": 304,
            "p1_power_reactive": 21,
            "p2_power_reactive": 255,
            "p3_power_reactive": 27,
            "power_reactive_scale": 0,
            "power_factor": -9860,
            "p1_power_factor": -9969,
            "p2_power_factor": -9639,
            "p3_power_factor": -9969,
            "power_factor_scale": -2,
            "export_energy_active": 13571,
            "p1_export_energy_active": 6778,
            "p2_export_energy_active": 4194,
            "p3_export_energy_active": 5802,
            "import_energy_active": 42181,
            "p1_import_energy_active": 13911,
            "p2_import_energy_active": 14527,
            "p3_import_energy_active": 16946,
            "energy_active_scale": 0,
            "export_energy_apparent": 0,
            "p1_export_energy_apparent": 0,
            "p2_export_energy_apparent": 0,
            "p3_export_energy_apparent": 0,
            "import_energy_apparent": 0,
            "p1_import_energy_apparent": 0,
            "p2_import_energy_apparent": 0,
            "p3_import_energy_apparent": 0,
            "energy_apparent_scale": -32768,
            "import_energy_reactive_q1": 0,
            "p1_import_energy_reactive_q1": 0,
            "p2_import_energy_reactive_q1": 0,
            "p3_import_energy_reactive_q1": 0,
            "import_energy_reactive_q2": 0,
            "p1_import_energy_reactive_q2": 0,
            "p2_import_energy_reactive_q2": 0,
            "p3_import_energy_reactive_q2": 0,
            "export_energy_reactive_q3": 0,
            "p1_export_energy_reactive_q3": 0,
            "p2_export_energy_reactive_q3": 0,
            "p3_export_energy_reactive_q3": 0,
            "export_energy_reactive_q4": 0,
            "p1_export_energy_reactive_q4": 0,
            "p2_export_energy_reactive_q4": 0,
            "p3_export_energy_reactive_q4": 0,
            "energy_reactive_scale": -32768
        }
    },
    "batteries": {
        "Battery1": {
            "c_manufacturer": "48V_BYD",
            "c_model": "BYD Premium LVS 4.0",
            "c_version": "48V DCDC 2.2.91 BMS 0.0.0",
            "c_serialnumber": "7E047E6D",
            "c_deviceaddress": 112,
            "c_sunspec_did": 0,
            "rated_energy": 4000.0,
            "maximum_charge_continuous_power": 3300.0,
            "maximum_discharge_continuous_power": 3300.0,
            "maximum_charge_peak_power": 50.0,
            "maximum_discharge_peak_power": 3950.0,
            "average_temperature": 27.0,
            "maximum_temperature": 0.0,
            "instantaneous_voltage": 818.2760009765625,
            "instantaneous_current": 0.013198137283325195,
            "instantaneous_power": 0.0,
            "lifetime_export_energy_counter": 173,
            "lifetime_import_energy_counter": 3337,
            "maximum_energy": 4000.0,
            "available_energy": 4000.0,
            "soh": 100.0,
            "soe": 100.0,
            "status": 6,
            "status_internal": 5,
            "event_log": 0,
            "event_log_internal": 0
        }
    }
}
```

power_ac and power_dc here seem to match the current PV production. But how comes that both values are around 0.2kW in the middle of the night? Or is this the battery counted as PV production as well?

nmakel, 2021-07-23:
> power_ac and power_dc here seem to match the current PV production. But how comes that both values are around 0.2kW in the middle of the night? Or is this the battery counted as PV production as well?

Your current consumption should be about 274W. The 213W production at midnight, together with the inverter reporting as producing is what seemed odd to me in your first post. A `--json` output at that time would be useful. The power_ac and power_dc values in the inverter's data should not be influenced by a kWh meter or a battery, at all. They may be taken into account by the SolarEdge API and phone app, however.

Deepdiigger, 2021-07-23:
im not sure if this helps. But i recall that i read some posts about the solaredge battery not beeing correctly documented. It seems that it needs substantial watts even in idle mode. Examples in the mid of the night were given were it should be zero but it was not. Maybe a point to consider too. I dont have a battery, but this script works now perfectly for me.

nmakel, 2021-07-23:
> im not sure if this helps. But i recall that i read some posts about the solaredge battery not beeing correctly documented. It seems that it needs substantial watts even in idle mode. Examples in [...]

Interesting. @fredlcore , is the battery connected using a storedge module?

fredlcore, 2021-07-23:
I only have this setup for a few days, how could I find out? What I found out so far is that the battery is connected via a ModBus connection to the SolarEdge inverter.

nmakel, 2021-07-23:
> I only have this setup for a few days, how could I find out? What I found out so far is that the battery is connected via a ModBus connection to the SolarEdge inverter.

If the battery is connected _via_ the inverter to your electrical installation, the explanation provided by @Deepdiigger makes sense. The most obvious way of finding out is following the cabling from the battery. If there is DC cabling from the battery to a box with SolarEdge on it the inverter is acting as inverter from battery DC to AC.

The consumption is sum(inverter.power_ac - meter.p1_power - meter.p2_power - meter.p3_power).

fredlcore, 2021-07-23:
Yes, the battery seems to be connected via the inverter, all cables seem to go through it. So that means there is no way to figure out the actual PV production? Or I guess I could calculate it with the help of the values from the battery1 JSON output?

nmakel, 2021-07-23:
I'm not familiar with all of the settings available for batteries. However, the actual consumption can be calculated using the formula I posted above. The sum of all generating components minus the power that your kWh meter registers as export is the consumption of your entire system. The power required by the battery when charging and when operating normally are (delayed) consumption, but consumption nonetheless -- power provided to the battery will never be exported. If you're interested in tracking the actual consumption of your system at all times you will need to also subtract the battery.instantaneous_power value from the power being exported. At night there will be no export, which means the p1/2/3_power values from your combined with the instantaneous_power value will equal your actual consumption.

fredlcore, 2021-07-23:
Thanks a lot for this comprehensive response, I really appreciate it and get a better understanding of things now. Thanks also for this really cool program! With this, I may be able to completely stop using the Web API which seems to continuously send data to SolarEdge.

fredlcore, 2021-07-29:
Just an additional info: Power provided to the battery can actually be exported - my SolarEdge inverter has a mode where I can tell it to export both PV and battery power to the grid. I plan to do this in the morning when PV power is higher than consumption in order to empty the battery before I reach the 70% threshold that usually starts around 11am in summer. This way the battery is empty when I switch to the mode where the battery is only charged with excess PV power.
As for the calculation, "instantaneous_power" is the power drawn/provided from the battery.
So (at least for my system), the calculation needs to be done like this:

PV production: inverter.power_ac * (10 ** inverter.power_ac_scale) + batteries.instantaneous_power
Grid export/import: meters.power * (10 ** meters.power_scale) (positive values mean export, negative ones mean import)
Battery export/import: batteries.instantaneous_power (positive values mean charging, negative values mean discharging)
Household energy consumption: inverter.power_ac * (10 ** inverter.power_ac_scale) - meters.power * (10 ** meters.power_scale)

Together with batteries.soe (which gives me the percentage of charge of the battery), I can now dynamically change the strategy how the produced energy should be used. For example, if PV production is higher than consumption in the morning, I can discharge the rest of the battery (except for a safety buffer of 10-20%) to the grid so that I have enough capcity before peak production during noon where I switch to charging the battery only with PV excess production. In the afternoon, I can switch back to maximize for self consumption where consumption has priority over charging which again has priority over grid export.

nmakel, 2021-07-29:
Interesting situation. I lack battery storage, so I don't face these issues. Do you have restrictions on how much you are allowed to export to the grid, or otherwise have an incentive to export at certain times? Also, have you figured out whether the charging and discharging of the battery is reflected in the power_ac values of your inverter?

fredlcore, 2021-07-29:
Any kind of export (be it 100% PV or a mix of PV and previously stored charge in the battery or even 100% from battery) is limited to the 70% cap that is configured in the inverter. I have a 5.92 kW system installed, so export is capped at 4.11 kW. In the morning, when production is around 1-2kW and I only consume 0.2-0.3kW, I can export to the grid whatever charge is left in the battery after the night. With a 4kW battery, I still have 25-35% capacity left. When I discharge the battery down to 10% (to have some charge left for the occasional use of the water boiler or so), I have enough capacity to store the excess capacity around noon when production is often times more than 5kW, but I can only export 4.11kW. The rest goes into consumption and the battery. During summer months, I can easily charge the battery to almost 100% just with this excess production which would otherwise go to waste.

As for inverter.power_ac, it is a bit strange: If the battery is discharging, the power that comes from the battery is included in inverter.power_ac. If the battery is charging, inverter.power_ac does not include the power that goes into the battery. 
So in order to know what the panels on my roof are producing, I have to *add* batteries.instantaneous_power to inverter.power_ac. If batteries.instantaneous_power is negative, it means that power is drawn from the battery, so adding this negative value removes the battery's contribution from inverter.power_ac. If it is charging, a positive value is added, so it gives me the total PV production.

nmakel, 2021-07-29:
> Any kind of export (be it 100% PV or a mix of PV and previously stored charge in the battery or even 100% from battery) is limited to the 70% cap that is configured in the inverter.

Is that a wiring limitation?

> As for inverter.power_ac, it is a bit strange: If the battery is discharging, the power that comes from the battery is included in inverter.power_ac. If the battery is charging, inverter.power_ac does not include the power that goes into the battery.

Does that explain the 200W reported by your inverter at midnight?

fredlcore, 2021-07-29:
No, it is a legal limitation (in Germany). Since 2012, every PV installation with less than 30kW is allowed to export only 70% of it's maximum production power:
https://www.photovoltaik-web.de/photovoltaik/wechselrichter/vergleich-70-regelung-und-einspeisemanagement

And yes, that explains the 200W reported at midnight, because the 200W came from the battery and power drawn from the battery is added to inverter.power_ac.

fredlcore, 2021-07-29:
One more interesting fact:
The term `inverter.power_ac + batteries.instantaneous_power` can result in an (invalid) negative value:
```
{
    "c_manufacturer": "SolarEdge",
    "c_model": "SE5K-RWS48BNN4",
    "c_version": "0004.0013.0038",
    "c_serialnumber": "7E047E6D",
    "c_deviceaddress": 1,
    "c_sunspec_did": 103,
    "current": 217,
    "p1_current": 69,
    "p2_current": 74,
    "p3_current": 74,
    "current_scale": -2,
    "p1_voltage": 4057,
    "p2_voltage": 4059,
    "p3_voltage": 4075,
    "p1n_voltage": 2346,
    "p2n_voltage": 2328,
    "p3n_voltage": 2364,
    "voltage_scale": -1,
    "power_ac": 15800,
    "power_ac_scale": -2,
    "frequency": 4993,
    "frequency_scale": -2,
    "power_apparent": 5100,
    "power_apparent_scale": -1,
    "power_reactive": -4850,
    "power_reactive_scale": -1,
    "power_factor": -3092,
    "power_factor_scale": -2,
    "energy_total": 192776,
    "energy_total_scale": 0,
    "current_dc": 19661,
    "current_dc_scale": -5,
    "voltage_dc": 8158,
    "voltage_dc_scale": -1,
    "power_dc": 16040,
    "power_dc_scale": -2,
    "temperature": 4436,
    "temperature_scale": -2,
    "status": 4,
    "vendor_status": 0,
    "storage_control_mode": 2,
    "storage_ac_charge_policy": 1,
    "storage_ac_charge_limit": 0.0,
    "storage_backup_reserved_setting": 0.0,
    "storage_default_mode": 0,
    "rc_cmd_timeout": 235929600,
    "rc_cmd_mode": 0,
    "rc_charge_limit": 2.00424861907959,
    "rc_discharge_limit": 2.00424861907959,
    "meters": {
        "Meter1": {
            "c_manufacturer": "SolarEdge",
            "c_model": "SE-RGMTR-1D-240C-A",
            "c_option": "Export+Import",
            "c_version": "0",
            "c_serialnumber": "0",
            "c_deviceaddress": 1,
            "c_sunspec_did": 203,
            "current": 29,
            "p1_current": 9,
            "p2_current": 10,
            "p3_current": 9,
            "current_scale": -1,
            "voltage_ln": 23451,
            "p1n_voltage": 23451,
            "p2n_voltage": 23324,
[... 78 lines cut ...]
            "average_temperature": 25.0,
            "maximum_temperature": 0.0,
            "instantaneous_voltage": 818.9421997070312,
            "instantaneous_current": 0.39138081669807434,
            "instantaneous_power": -249.0,
            "lifetime_export_energy_counter": 51,
            "lifetime_import_energy_counter": 0,
            "maximum_energy": 4000.0,
            "available_energy": 4000.0,
            "soh": 100.0,
            "soe": 93.0,
            "status": 4,
            "status_internal": 5,
            "event_log": 0,
            "event_log_internal": 0
        }
    }
}
```
inverter.power_ac is 158,00W, meter1.power is -8W (i.e. getting 8W from the grid) and batteries.instantaneous_power is -249.0W, so the total sum in that case is -99W. Some part of that deviation might be stemming from the fact that the battery is running on DC, so maybe inverter.power_dc should be taken into account rather than inverter.power_ac, but the difference is minimal (160.4 compared to 158.0).
I was then thinking that the battery is too "slow" to reduce it's power, but when I log these values (currently it's 9.30pm), the values stay the same when polling them in 10 seconds intervals for a while.
When I compare this to the values in the web interface, the battery power seems to be correct, it's the household consumption that is off.
As mentioned before, I calculate household consumption as the sum of inverter.power_ac + batteries.instantaneous_power. During daytime, this seems to be fine, but not during night time. Consumption in the web interface is about the same as batteries.instantaneous_power, i.e. 336W at the point in time of the above JSON dump. The closest value to this would be meters.p2_power, but that would be off by factor 10.

Do you have any ideas how this is possible?

fredlcore, 2021-08-10:
I found out that this problem only occurs when the battery is discharging. Once the battery is empty (late into the night for example) and all electricity comes from the grid, this behaviour does not exist. I have therefore taken the approach that once PV production is negative and the battery is discharging, the total household consumption is the amount of energy flowing from the battery plus the amount of energy coming from the grid.

By the way, I have kind of finished my predictive battery charging for SolarEdge inverters, just in case anyone is interested:
https://github.com/fredlcore/SolarEdge_Predictive_Charging/
Thanks again for your module, @nmakel - without it, this would not have been possible!