nmakel/solaredge_modbus #30: Export limit available?

drejc12, 2021-04-16:
Hello, can you tell me if export-limit is available to view using registers on the inverter? If yes, can it be changes on-the-fly? I would like to change export limit based on some other circumstances in the grid.

nmakel, 2021-04-18:
No, the list of exposed values is found on page 16 of [this document](https://github.com/nmakel/solaredge_modbus/blob/master/documents/sunspec-implementation-technical-note-v2.0.pdf). The value you are looking for is available through the SetApp API, though. Have a look at [solaredge-local](https://github.com/drobtravels/solaredge-local) or [solaredge_setapp](https://github.com/nmakel/solaredge_setapp).

drejc12, 2021-04-19:
Thanks I will have a look and try to work with these repos. So you think export limit can't be changed with code, only read?

nmakel, 2021-04-19:
> Thanks I will have a look and try to work with these repos. So you think export limit can't be changed with code, only read?

SetApp is a standard rest API over http. If you can access SetApp, you can change the export limit.

drejc12, 2021-04-19:
Hey, thanks for your fast feedback and your help. Hope I will figure out.

closed 2021-04-21 by nmakel