nmakel/solaredge_modbus #33: Monitoring individual PV modules?

JackKelly, 2021-06-15:
This looks like an amazing Python package - thanks loads for sharing!

One quick question:  Can this Python package monitor data for individual PV modules (if the modules are ['smart modules'](https://www.solaredge.com/sites/default/files/se-smart-modules-all-black-datasheet.pdf) or the PV modules have [power optimizers](https://www.solaredge.com/uk/products/power-optimizer#/) installed)?  Apparently the Solar Edge monitoring platform captures module-level data, but I don't know if module-level data is available over MODbus.

nmakel, 2021-06-15:
> One quick question: Can this Python package monitor data for individual PV modules (if the modules are ['smart [...]

Not as far as I'm aware. While there is more data exposed on higher register addresses not mentioned in the documentation, I have yet to encounter anything that resembles module data.

JackKelly, 2021-06-16:
Cool, thank you for the quick reply!

Whilst it's not documented, apparently it is possible to get module-level data from Solar Edge's API: https://twitter.com/AyrtonBourn/status/1405071870026731523

nmakel, 2021-06-16:
> Whilst it's not documented, apparently it is possible to get module-level data from Solar Edge's API: https://twitter.com/AyrtonBourn/status/1405071870026731523

Luckily there are still ways. You may also want to try polling the SetApp API on the inverter itself. This is accessible via the Wifi hotspot that is enabled when you toggle the switch on newer model inverters. Have a look at [solaredge_setapp](https://github.com/nmakel/solaredge_setapp), or this [discussion at solaredge_local](https://github.com/drobtravels/solaredge-local/issues/24).

cooldil, 2021-06-16:
> Cool, thank you for the quick reply!
> 
> Whilst it's not documented, apparently it is possible to get module-level data from Solar Edge's API: https://twitter.com/AyrtonBourn/status/1405071870026731523

I wasn't able to find a way to get the module-level data from the inverter itself using modbus.  However, as you have found out, you can get the (15 min resolution) panel data using the solaredge website API.  I have a python script running that queries the API every 15 minutes and puts the data into an InfluxDB....  Full disclosure that I am not the author of the script in question but I did modify it a bit to "customize" the module names so that they made more sense than the "numbers" that are returned by the API....   Take a look at https://gist.github.com/cooldil/0b2c5ee22befbbfcdefd06c9cf2b7a98 to see if that would be helpful to you...

closed 2021-07-24 by nmakel

thibaultmol, 2023-04-24:
This still hasn't become available trough the actual modbus ,I'm guessing?