nmakel/solaredge_modbus #34: Not working setup

Keesiej, 2021-06-18:
I tried the modbus library for the first time. The example generates data but when i try to use the read_all i get the following.

`>>> inverter.connect()
True
>>> inverter.connected()
True
>>> inverter.read_all()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/solaredge_modbus/__init__.py", li                                                                                                       ne 343, in read_all
    results.update(self._read_all(register_batch, rtype))
  File "/usr/local/lib/python3.7/dist-packages/solaredge_modbus/__init__.py", li                                                                                                       ne 296, in _read_all
    data = self._read_holding_registers(offset, length)
  File "/usr/local/lib/python3.7/dist-packages/solaredge_modbus/__init__.py", li                                                                                                       ne 218, in _read_holding_registers
    result = self.client.read_holding_registers(address=address, count=length, u                                                                                                       nit=self.unit)
  File "/usr/local/lib/python3.7/dist-packages/pymodbus/client/common.py", line                                                                                                        114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.7/dist-packages/pymodbus/client/sync.py", line 10                                                                                                       8, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to co                                                                                                       nnect[ModbusTcpClient(192.168.2.8:1502)]
>>> inverter.read("current")
{'current': 95}
`

Does anyone have a idea how to fix?

nmakel, 2021-06-18:
Odd. I just checked both 0.6.5 and the github master, combined with pymodbus 2.4.0 and 2.5.2 and cannot replicate.

Could you provide me with a few details: your `solaredge_modbus` version, your `pymodbus` version (run `pip3 list` to find the versions), the inverter type, and whether you have any other devices communicating with the inverter's modbus TCP service.

Keesiej, 2021-06-23:
Thanks for contacting. I found out that i had the pymodbus installed for python 2 instead of python 3. After fixing this all worked like a charm!!

closed 2021-07-24 by nmakel