nmakel/solaredge_modbus #35: Works perfectly!! I Sugget to add it to the PP-MANAGER

CreasolTech, 2021-07-24:
Hi Niels, 
you plugins is really very good and well documented.
I just suggest you to add it to the PP-MANAGER and Domoticz Plugins Manager:
Check https://www.domoticz.com/wiki/Plugins .
These two plugins are commonly used to install other plugins automatically, using git.
Adding your plugin to the list of plugins supported by PP-MANAGER and Domoticz Plugins Manager requires just few minutes.
Thanks a lot again for your good plugin.
Paolo

CreasolTech, 2021-07-24:
I have to add that I'm using the domoticz plugin by Addie Janssen
https://github.com/addiejanssen/domoticz-solaredge-modbustcp-plugin
Bye.
Paolo

nmakel, 2021-07-24:
Thanks for the kind words, Paolo. I'm not a domoticz user, so I'm happy to have @addiejanssen provide his plugin :)

closed 2021-07-24 by nmakel