nmakel/solaredge_modbus #37: Error: ValueError: Not a valid parity: False

ctrl-alt-d, 2021-07-26:
I just run a fresh copy following instructions and the error `ValueError: Not a valid parity: False` is raised. What can be the issue? Many thanks. Sorry if this is not the forum to post this issue.

```
Python 3.8.11 (default, Jul 22 2021, 15:32:17) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import solaredge_modbus
>>> inverter = solaredge_modbus.Inverter("192.168.86.156", 1502, 10, 1)

>>> inverter.read_all()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/site-packages/solaredge_modbus/__init__.py", line 354, in read_all
    results.update(self._read_all(register_batch, rtype))
  File "/usr/local/lib/python3.8/site-packages/solaredge_modbus/__init__.py", line 307, in _read_all
    data = self._read_holding_registers(offset, length)
  File "/usr/local/lib/python3.8/site-packages/solaredge_modbus/__init__.py", line 225, in _read_holding_registers
    self.connect()
  File "/usr/local/lib/python3.8/site-packages/solaredge_modbus/__init__.py", line 330, in connect
    return self.client.connect()
  File "/usr/local/lib/python3.8/site-packages/pymodbus/client/sync.py", line 652, in connect
    self.socket = serial.serial_for_url(self.port,
  File "/usr/local/lib/python3.8/site-packages/serial/__init__.py", line 87, in serial_for_url
    instance = klass(None, *args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/serial/serialutil.py", line 225, in __init__
    self.parity = parity
  File "/usr/local/lib/python3.8/site-packages/serial/serialutil.py", line 336, in parity
    raise ValueError("Not a valid parity: {!r}".format(parity))
ValueError: Not a valid parity: False

>>> inverter.read("c_manufacturer")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/site-packages/solaredge_modbus/__init__.py", line 342, in read
    return {key: self._read(self.registers[key])}
  File "/usr/local/lib/python3.8/site-packages/solaredge_modbus/__init__.py", line 273, in _read
    return self._decode_value(self._read_holding_registers(address, length), length, dtype, vtype)
  File "/usr/local/lib/python3.8/site-packages/solaredge_modbus/__init__.py", line 225, in _read_holding_registers
    self.connect()
  File "/usr/local/lib/python3.8/site-packages/solaredge_modbus/__init__.py", line 330, in connect
    return self.client.connect()
  File "/usr/local/lib/python3.8/site-packages/pymodbus/client/sync.py", line 652, in connect
    self.socket = serial.serial_for_url(self.port,
  File "/usr/local/lib/python3.8/site-packages/serial/__init__.py", line 87, in serial_for_url
    instance = klass(None, *args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/serial/serialutil.py", line 225, in __init__
    self.parity = parity
  File "/usr/local/lib/python3.8/site-packages/serial/serialutil.py", line 336, in parity
    raise ValueError("Not a valid parity: {!r}".format(parity))
ValueError: Not a valid parity: False

```

Tested with python 3.8, 3.7 and 3.9

ctrl-alt-d, 2021-07-26:
Update, If I just set ip and port the error is not raised but not info is provided from inverter:

```
>>> inverter = solaredge_modbus.Inverter("192.168.86.156", port=1502)
>>> inverter.read_all()
{}
```

Any help is appreciated. Thanks!

nmakel, 2021-07-26:
> inverter = solaredge_modbus.Inverter("192.168.86.156", 1502, 10, 1)

The first two parameters, host and port, will work positionally, but the other values you're providing (timeout and unit?) map to device and stopbits. You'll want to use named parameters:

`>>> inverter = solaredge_modbus.Inverter(host="192.168.86.156", port=1502, timeout=10, unit=1)`

> Update, If I just set ip and port the error is not raised but not info is provided from inverter:

Have you enabled Modbus TCP and specified the correct port number on the inverter? Are there any other devices using the Modbus TCP service simultaneously?

ctrl-alt-d, 2021-07-26:
hi @nmakel , thanks about your help. Let's try to isolate problem :raised_hands:  ! Maybe the problem is with the inverter.

All worked fine until last Friday, when a technician came home to install the meter. Also, congrats about this library, it is awesome.

Since Friday, I have not been able to connect anymore. Port is set to 1502 (with technician, we change to 502 for testing and back to 1502) Also TCP is enable (we switched enable - disable and back to make tests). Now, enable at 1502.

I don't know how to isolate the problem. I installed "simply modbus tcp" to test the connection, but I don't know what parameters I must set to check if I can read data from inverter. Maybe exists an easiest way to isolate issue.

Best regards.

[image]

nmakel, 2021-07-26:
> All worked fine until last Friday (congrats about this library, it is awesome), when a technician came home to install the meter.

You were using solaredge_modbus before Friday, and then a meter was added to the setup, after which solaredge_modbus no longer worked, correct?

I suspect your inverter's ID has changed. Did the technician perhaps modify the modbus device ID in order to not have it conflict with the ID set on the meter? You can try changing the `unit` parameter to something other than `1`, which is the default. In Simply Modbus the value you want to try changing is `Slave ID`.

What inverter model do you have? Do you have access to the SetApp commissioning interface?

nmakel, 2021-07-26:
> Wow. After changing Id to 2 now I get data!!! +1

Excellent!

> Because I have a metter, why meters json section is empty?

I suspect there's a bug. Could you try the following in a Python interpreter?

```
import solaredge_modbus
inverter = solaredge_modbus.Inverter(host="192.168.86.156", port=502, unit=2)
inverter
meter1 = inverter.meters()["Meter1"]
meter1
```

ctrl-alt-d, 2021-07-26:
here it is:

```
(venv) dani@localhost solarx $ python
Python 3.9.6 (default, Jun 29 2021, 00:00:00) 
[GCC 10.3.1 20210422 (Red Hat 10.3.1-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import solaredge_modbus
s.Inverter(host="192.168.86.156", port=502, unit=2)
inverter
meter1 = inverter.meters()["Meter1"]
meter1>>> inverter = solaredge_modbus.Inverter(host="192.168.86.156", port=502, unit=2)
>>> inverter
Inverter(192.168.86.156:502, connectionType.TCP: timeout=1, retries=3, unit=0x2)
>>> meter1 = inverter.meters()["Meter1"]
>>> meter1
Meter1(192.168.86.156:502, connectionType.TCP: timeout=1, retries=3, unit=0x1)
>>> 
```

nmakel, 2021-07-26:
Thanks. The `unit` value is not being passed from parent to client. Could you try:

```
import solaredge_modbus
meter1 = solaredge_modbus.Meter(host="192.168.86.156", port=502, unit=2)
meter1.read_all()
```

ctrl-alt-d, 2021-07-26:
>The unit value is not being passed from parent to client.

Yes, this was the reason, :clap: 

```
>>> import solaredge_modbus
>>> meter1 = solaredge_modbus.Meter(host="192.168.86.156", port=502, unit=2)
>>> meter1.read_all()
{'c_manufacturer': 'WattNode', 'c_model': 'WND-3Y-400-MB', 'c_option': 'Export+Import' ...
```

nmakel, 2021-07-26:
Great! I just uploaded version 0.6.6, which should solve this bug. Would you mind trying it out and letting me know whether it fixes your problem? Just running `./example.py 192.168.86.156 502 --json --unit 2` should work.

ctrl-alt-d, 2021-07-26:
Like a charm!

```
(venv) dani@localhost solarx $ ./e.py 192.168.86.156 502 --json --unit 2
{
    "c_manufacturer": "SolarEdge",
    "c_model": "SE3680H-RW000BNV4",
    "c_version": "0004.0012.0035",
    "c_serialnumber": "XXXXXXX",
    "c_deviceaddress": 2,
    "c_sunspec_did": 101,
    "current": 204,
    "p1_current": 204,
    "p2_current": 0,
    "p3_current": 0,
    "current_scale": -2,
    "p1_voltage": 2320,
    "p2_voltage": 0,
    "p3_voltage": 0,
    "p1n_voltage": 0,
    "p2n_voltage": 0,
    "p3n_voltage": 0,
    "voltage_scale": -1,
    "power_ac": 4564,
    "power_ac_scale": -1,
    "frequency": 49991,
    "frequency_scale": -3,
    "power_apparent": 4750,
    "power_apparent_scale": -1,
    "power_reactive": 13166,
    "power_reactive_scale": -2,
    "power_factor": 9608,
    "power_factor_scale": -2,
    "energy_total": 331188,
    "energy_total_scale": 0,
    "current_dc": 11539,
    "current_dc_scale": -4,
    "voltage_dc": 4015,
    "voltage_dc_scale": -1,
    "power_dc": 4633,
    "power_dc_scale": -1,
    "temperature": 4952,
    "temperature_scale": -2,
    "status": 4,
    "vendor_status": 0,
    "rrcr_state": 0,
    "active_power_limit": 100,
    "cosphi": 0,
    "meters": {
        "Meter1": {
            "c_manufacturer": "WattNode",
            "c_model": "WND-3Y-400-MB",
            "c_option": "Export+Import",
            "c_version": "31",
            "c_serialnumber": "5070223",
            "c_deviceaddress": 2,
            "c_sunspec_did": 203,
            "current": -17,
            "p1_current": -17,
            "p2_current": 0,
            "p3_current": 0,
            "current_scale": -1,
            "voltage_ln": 23252,
            "p1n_voltage": 23252,
            "p2n_voltage": 0,
            "p3n_voltage": 0,
            "voltage_ll": 23125,
            "p12_voltage": 23112,
            "p23_voltage": 0,
            "p31_voltage": 23138,
            "voltage_scale": -2,
            "frequency": 4997,
            "frequency_scale": -2,
            "power": 249,
            "p1_power": 249,
            "p2_power": 0,
            "p3_power": 0,
            "power_scale": 0,
            "power_apparent": 319,
            "p1_power_apparent": 319,
            "p2_power_apparent": 0,
            "p3_power_apparent": 0,
            "power_apparent_scale": 0,
            "power_reactive": -199,
            "p1_power_reactive": -199,
            "p2_power_reactive": 0,
            "p3_power_reactive": 0,
            "power_reactive_scale": 0,
            "power_factor": 4598,
            "p1_power_factor": -6203,
            "p2_power_factor": 10000,
            "p3_power_factor": 10000,
            "power_factor_scale": -2,
            "export_energy_active": 93402,
            "p1_export_energy_active": 93402,
            "p2_export_energy_active": 0,
            "p3_export_energy_active": 0,
            "import_energy_active": 19049,
            "p1_import_energy_active": 19049,
            "p2_import_energy_active": 0,
            "p3_import_energy_active": 0,
            "energy_active_scale": 0,
            "export_energy_apparent": 0,
            "p1_export_energy_apparent": 0,
            "p2_export_energy_apparent": 0,
            "p3_export_energy_apparent": 0,
            "import_energy_apparent": 0,
            "p1_import_energy_apparent": 0,
            "p2_import_energy_apparent": 0,
            "p3_import_energy_apparent": 0,
            "energy_apparent_scale": -32768,
            "import_energy_reactive_q1": 0,
            "p1_import_energy_reactive_q1": 0,
            "p2_import_energy_reactive_q1": 0,
            "p3_import_energy_reactive_q1": 0,
            "import_energy_reactive_q2": 0,
            "p1_import_energy_reactive_q2": 0,
            "p2_import_energy_reactive_q2": 0,
            "p3_import_energy_reactive_q2": 0,
            "export_energy_reactive_q3": 0,
            "p1_export_energy_reactive_q3": 0,
            "p2_export_energy_reactive_q3": 0,
            "p3_export_energy_reactive_q3": 0,
            "export_energy_reactive_q4": 0,
            "p1_export_energy_reactive_q4": 0,
            "p2_export_energy_reactive_q4": 0,
            "p3_export_energy_reactive_q4": 0,
            "energy_reactive_scale": -32768
        }
    },
    "batteries": {}
}
```

closed 2021-07-26 by nmakel