nmakel/solaredge_modbus #38: incomplete install output

SvenDowideit, 2021-07-29:
I'm assuming the result I have is because things are not yet fully configured - but I get

```
./example.py --json 10.10.10.146 1502
{
    "meters": {},
    "batteries": {
        "Battery1": {},
        "Battery2": {}
    }
}
```

and 

```
./example.py 10.10.10.146 1502
Inverter(10.10.10.146:1502, connectionType.TCP: timeout=1, retries=3, unit=0x1):

Registers:
Traceback (most recent call last):
  File "/home/sven/src/github/solaredge_modbus/./example.py", line 46, in <module>
    print(f"\tManufacturer: {values['c_manufacturer']}")
KeyError: 'c_manufacturer'
```

If I'm right, it might be useful to add this info to the docs and for the non-json output to tell the user what's probably going on :)

SvenDowideit, 2021-07-29:
ooo, then again, I'm not sure that its successful!

nmakel, 2021-07-29:
> ooo, then again, I'm not sure that its successful!

It certainly isn't :)

Are you sure Modbus TCP is enabled, the port is correct, and that the modbus address of the inverter is 1?

fredlcore, 2021-07-29:
I had the same problem once I had started to access the ModBus port from my FHEM installation as well. If another program is keeping the connection on that port open, this (empty result) is what you get. Once I had closed that open connection, things went well.

SvenDowideit, 2021-07-29:
I suspect not.
The installers just left, and it's a new model, so 💯 unknowns

ctrl-alt-d, 2021-07-29:
1.- Be sure the modbus tcp port is open and listening. You can use `netcat` to check open ports:

```bash
netcat -p- 10.10.10.146  # the port 1502 should be listed
```

2.- If do you have a meter, maybe, the ID changes, try with other Id:

```bash
./example.py --json 10.10.10.146 1502 --unit 2
```

SvenDowideit, 2021-07-29:
yup, sees to only be listening on 80 and 8080 - and those are a sink.

IDK enough about it to know if i can configure it to listen, or if this being a newer model, everything is locked down

so mostly - this issue is about not-so-clean output on failure to communicate :)

nmakel, 2021-08-11:
> yup, sees to only be listening on 80 and 8080 - and those are a sink.

Unless you have access to the setapp interface on port 80 you'll need to either register an installer account with solaredge (free, no checks whatsoever) and turn on Modbus TCP yourself using the SetApp application for IOS or Android, or ask your installer to do so.

> so mostly - this issue is about not-so-clean output on failure to communicate :)

Good point. When using the library you will get an empty dict, which you can then do with what you need. The example files assume a working setup, and would benefit from a bit more verbose output checks.