nmakel/solaredge_modbus #39: Sample dashboards / grafana

Bonn93, 2021-08-09:
Hi, thanks a tonne for this project, it's quite awesome! 

I've used your sample influx script to dump the data to influxdb, and have Grafana plugged in over the top and been able to create a few graphs such as power usage, production, temp monitors, and wondering if you have anything built or what you use to visualize the data. 

The few metrics I have are great, but its a bit funky trying to match some of the data to mirror the solaredge cloud, ie total kWh per day, or what sort of voltages I should be looking at. 

[image]

iNaiks, 2021-08-10:
hi @Bonn93, 
Seeing that you are running, I've a question for you xD.
I've an error running the example_influxdb.py, maybe I'm running wrong.
I use the terminal with command: pythoon3 example_influxdb.py [and the other commands with database, user, password, host...] 
Is correct? or I need to run with a python program or something? (I use a raspberry pi)

Can you explain me how do you run? (if you want)

Thanks 
Nico

Bonn93, 2021-08-11:
@iNaiks so I originally tested with a python/terminal and wrapped into a docker image that a built, so I could set a few things to work for me, I had an earlier version where it tried a new modbus library which didn't work well, and had to tweak the requirements. 

But, sharing is caring! Here's my docker run script ( it was in k8s, but had some issues with influx storage and had to move to plain old docker whilst rebuilt things ). 

I also use the latest 1.8.X release on InfluxDB, not 2.X as it had a bunch of weird issues. 

#!/bin/bash

docker stop modbus
docker rm mobus

```
docker run -d \
--name=modbus \
--restart=always \
-e INVERTER_IP=10.5.90.2 \
-e INVERTER_PORT=1502 \
-e INFLUX_HOST=10.5.70.3 \
-e INFLUX_PORT=8086 \
-e INFLUX_DB=solaredge \
-e INFLUX_USER=solaredge\
-e INFLUX_PASSWORD=solaredge\
mbern/modbus-solaredge:latest
```

The dockerfile is more interesting and shows how I run it. 

```
FROM python:3.7.9

RUN mkdir /modbus
WORKDIR /modbus
ADD ./python/solaredge_modbus /modbus
ADD ./requirements.txt /modbus
ADD ./example_influxdb.py /modbus

RUN pip install influxdb
RUN pip install -r requirements.txt
#RUN python setup.py install

ENV PYTHONUNBUFFERED=1

WORKDIR /modbus
CMD ["/bin/bash", "-c", "python3 example_influxdb.py ${INVERTER_IP} ${INVERTER_PORT} --influx_host ${INFLUX_HOST} --influx_port ${INFLUX_PORT}  --influx_db ${INFLUX_DB} --influx_user ${INFLUX_USER} --influx_pass ${INFLUX_PASSWORD}"]

```

It's not perfect, but does the job. Ideally, I need to raise a PR and would like to use python to read the OS.envs to build the config, and it does need more logging that works differently when run direct via python in your own shell.

nmakel, 2021-08-11:
> Hi, thanks a tonne for this project, it's quite awesome!

Thanks!

> I've used your sample influx script to dump the data to influxdb, and have Grafana plugged in over the top and been able to create a few graphs such as power usage, production, temp monitors, and wondering if you have anything built or what you use to visualize the data.

My setup is similar, `solaredge_modbus`, combined with `solaredge_setapp` (for module level reporting) and `sdm_modbus` (which is my kWh meter measuring import+export at the grid), all reporting to influxdb, with grafana on top.

On a PV panel I plot voltage_dc, power_ac, and temperature to get a general picture of the state of things. I then have individual graphs of module voltages and current grouped by orientation (east facing modules, and west facing modules). In dashboard style I show the latest power_ac, ac voltage, ac frequency, dc voltage, production today, this month, and all time (from `solaredge_setapp`).

On a more general panel which also show temperature readings from around the house, along with central heating stats, I show a large graph of the inverter's power_ac, voltage dc, and the consumption as calculated from the actual import or export power measurement by my kWh meter. This graph shows whether the system is producing anything, when the production and consumption spikes were, and gives a general picture of production and consumption throughout the day. I find this graph alone enough for a quick glance at how things are going. Add a couple dashboard style gauges for current power_ac, voltages and temperature, and you should be fine.

iNaiks, 2021-08-11:
@Bonn93 @nmakel Thanks for the info,

But my first problem is build the python into a docker... (I know networks things, not programming, sorry)
I've a nas (Synology) with docker (for read and calculate the information) and a raspberry with a screen to see Grafana graphics.

What docker image I need to install or build with python?

Thanks 
Nico

nmakel, 2021-08-11:
> What docker image I need to install or build with python?

You don't *need* to run the script in a docker container. What error are you getting when you run the script currently?

Bonn93, 2021-08-12:
> @Bonn93 @nmakel Thanks for the info,
>
> But my first problem is build the python into a docker... (I know networks things, not programming, sorry)
> I've a nas (Synology) with docker (for read and [...]

So the dockerfile i linked above shows how you can build it, mines a little different because I use this repo as a submodule, that way I get all the version control and updates from this repo within my own repo. 

But, if you read the dockerfile carefully, you'll understand how it works. 

`FROM python:3.7.9` is using the python docker image as the base, and i use python 3.7.9 specifically. 

```
RUN mkdir /modbus
WORKDIR /modbus
```
These lines I create a directory in the container image, and set my working directory to this one. 

```
ADD ./python/solaredge_modbus /modbus
ADD ./requirements.txt /modbus
ADD ./example_influxdb.py /modbus
```
These lines I add basically the entire repo to my workdir, but then I also add the requirements.txt file ( which is consumed by pip ) I then add the exact script into the container being example.influx.py to my workdir. 

```
RUN pip install influxdb
RUN pip install -r requirements.txt
```
I then run pip ( which is like a package manger for python, included in the docker image you may need to install this locally to play/test ( apt install python3-pip ) for example.  

Pip then installs all the libraries/packages defined in requirements.txt 

```
CMD ["/bin/bash", "-c", "python3 example_influxdb.py ${INVERTER_IP} ${INVERTER_PORT} --influx_host ${INFLUX_HOST} --influx_port ${INFLUX_PORT}  --influx_db ${INFLUX_DB} --influx_user ${INFLUX_USER} --influx_pass ${INFLUX_PASSWORD}"]
```
Finally, I run the script via bash, it executes python, in a normal shell, this would be something like this: 

```
python3 example_influx.py 10.0.0.50 1502 --influx-host=10.0.0.51
```

Ultimately, ignore all the docker stuff, get it working with python locally first. Install python 3.7.9, install pip, run the pip install, run the script at let us know how you go with the basics, in a shell, it'll spit out stuff, or get the example.py working first.

Bonn93, 2021-08-12:
> > Hi, thanks a tonne for this project, it's quite awesome!
> 
> Thanks!
> 
> > I've used your sample influx script to dump the data to influxdb, and have Grafana plugged in over the top and been able to create a few graphs such as power usage, production, temp monitors, and wondering if you have anything built or what you use to visualize the data.
> 
> My setup is similar, `solaredge_modbus`, combined with `solaredge_setapp` (for module level reporting) and `sdm_modbus` (which is my kWh meter measuring import+export at the grid), all reporting to influxdb, with grafana on top.
> 
> On a PV panel I plot voltage_dc, power_ac, and temperature to get a general picture of the state of things. I then have individual graphs of module voltages and current grouped by orientation (east facing modules, and west facing modules). In dashboard style I show the latest power_ac, ac voltage, ac frequency, dc voltage, production today, this month, and all time (from `solaredge_setapp`).
> 
> On a more general panel which also show temperature readings from around the house, along with central heating stats, I show a large graph of the inverter's power_ac, voltage dc, and the consumption as calculated from the actual import or export power measurement by my kWh meter. This graph shows whether the system is producing anything, when the production and consumption spikes were, and gives a general picture of production and consumption throughout the day. I find this graph alone enough for a quick glance at how things are going. Add a couple dashboard style gauges for current power_ac, voltages and temperature, and you should be fine.
> 
> Edit: I should add that I calculate daily usage, import and export, by looking at values reported by my kWh meter _only_. It is possible to do basic difference calculations on values such as `energy_total` with queries such as:
> 
> `SELECT difference(last("energy_total")) *1000 FROM "pv" WHERE $timeFilter GROUP BY time(1d)`

I didn't even see your panel level stuff, I wonder if my SE10000H is supported, I did play with setapp stuff to get modbus going... 

Do you think you could share the grafana dashboards/template you have? It would be pretty cool if that could be a grafana template, I think many others would like it, I know a few folks with solaredge inverters all trying for the same type of monitoring.

iNaiks, 2021-08-13:
> > What docker image I need to install or build with python?
> 
> You don't _need_ to run the script in a docker container. What error are you getting when you run the script currently?

Ok, but I prefer to run all the things on the nas with docker. And use the raspberry only for watch the graphics in Grafana.
How can I create a docker to run the python scrip. I've the influxDB server and grafana server, installed on other dockers.

Very Thanks
Nico

Bonn93, 2021-08-16:
@iNaiks not trying to be rude here, but everything is in the above, you might need to go and understand the basics on containers and building images before you look at this.

iNaiks, 2021-08-17:
> @iNaiks not trying to be rude here, but everything is in the above, you might need to go and understand the basics on containers and building images before you look at this.

Yes, I have deleted everything and I have started again from the beginning and now everything works XD.
The only problem is that the NAS just broke ... But before that, it was perfect. Thanks a lot

iNaiks, 2021-08-18:
The only think is sometimes the scrip stops and I need to open the terminal in docker container, and run again.
I've a docker container in a Synology. I access via web to the UI of docker and I have a terminal (bash) open in a specified directory (/home/solaredge_modbus-master), and then I run the command (python3 example_influxdb.py 10.1.0.254 1502 --influx_host XXXX --influx_port 8086 --influx_db XXXX --influx_user XXXX --influx_pass XXXX --timeout 2)

A few hours later, the command stops (I can't see the error because when I go in the terminal via web on Synology, there isn't any error and is like a terminal console waiting for an input command)

There is anything that I can do for run again the scrips when stops or stop the scrip every hour and restart it again automatically? 

Thanks Nico

TuxFan777, 2021-10-20:
Thanks @Bonn93 - Appreciate the feedback!

Here is my modified version of Niels' "example_influxdb.py" utility which includes the additional "graph" MEASUREMENT that I've used in Grafana. I wasn't sure whether it was appropriate to upload this directly to the project's Code repository without Niels' blessing, so leaving it here for anyone who wants it : 

[example_influxdb_with_graph_measurement_TuxFan777.zip](https://github.com/nmakel/solaredge_modbus/files/7384948/example_influxdb_with_graph_measurement_TuxFan777.zip)

I'll also spend some time cleaning up a json export of the Grafana dashboard for you (minus db credentials, etc.).
Will post that as soon as it's ready.