nmakel/solaredge_modbus #40: How to get "current solar energy" value

stscde, 2021-09-28:
Hi Niels!

Thank you for creating and maintaining this library. I used it for some weeks now to collect the data of my SE7K to an influx DB. Which was very easy thanks to your documentation.

What I try to figure out for some time now is how to get the "current solar power" value from the inverter. See value "A" in the following screenshot from the SolarEdge app.

[image]

The value "power" in the following script output shows 259W which matches "B" in the screenshot. This is "I_AC_Power" register 40083. I searched the registers in the SolarEdige register documentation but found no other one that could contain "A".

Value "C" I can get from the meter. Value "D" probably from the battery. Did not yet investigate that. With B, C and D I could calculate A, but is this really the way?

```
Inverter(localhost:1502, connectionType.TCP: timeout=1, retries=3, unit=0x1):

Registers:
        Manufacturer: SolarEdge
        Model: SE7K-RWS48BEB4
        Type: Three Phase Inverter
        Version: 0004.0013.0040
        Serial: nnnnnnnnnn
        Status: Producing
        Temperature: 37.24°C
        Current: 2.63A
        Phase 1 Current: 0.85A
        Phase 2 Current: 0.89A
        Phase 3 Current: 0.88A
        Phase 1 voltage: 409.30V
        Phase 2 voltage: 409.50V
        Phase 3 voltage: 407.80V
        Phase 1-N voltage: 235.90V
        Phase 2-N voltage: 237.20V
        Phase 3-N voltage: 235.10V
        Frequency: 49.99Hz
        Power: 259.00W
        Power (Apparent): 621.50VA
        Power (Reactive): -565.00VAr
        Power Factor: 41.56%
        Total Energy: 454451Wh
        DC Current: 0.32A
        DC Voltage: 815.20V
        DC Power: 262.94W
Batteries found:
        Model: BYD Premium LVS 8.0
        SOE: 11.0
Meters found:
        Model: SE-MTR-3Y-400V-A
        Export Energy Active: 267017Wh
        Import Energy Active: 147045Wh
```

Note: Screenshot and script output were take at the same time (as fast as I could press the keys on my phone and on my keyboard).

Best wishes
Stefan

nmakel, 2021-09-28:
In my experience the Power value you have showing 259.00W is the actual AC output of the inverter, and is therefore the "current solar energy" in most normal setups.

The presence of a battery may be complicating matters, however. While the SolarEdge app is not real time, and a poor indicator of actual current production, the values B + D = A in this case. If your battery is connected via DC to your inverter the power used to charge the battery will not be exported to the grid or used as self consumption. The inverter will therefore only report the AC power it actually outputs. In this case you _will_ need to do some addition and subtraction.

See [this previous question](https://github.com/nmakel/solaredge_modbus/issues/28) for a similar situation.

stscde, 2021-09-29:
Thank you for pointing to the other question. The participants there made similar or the same experiences as I did. And yes my DC battery is directly connected to the SE7K.

When the sun in shining the SE7K routes energy to the inverter and to the battery or grid. But only the energy going to the inverter is shown in I_AC_Power. When its dark the SE7K routes energy from the battery to the inverter. This is then also shown in I_AC_Power.

Therefore the "production" value in the SolarEgde app is reduced by the amount of energy routed to the battery but increased by the amount taken from the battery. Of course this is not documented anywhere - or at least I did not found it.

Conclusion: I have to do the math myself.

frezeen, 2021-10-09:
i dont think B is from inverter. its the house load and u can read it from meter only. A from inverter is "power".

stscde, 2021-10-09:
@frezeen Thanks for your feedback. 

The way B is calculated depends on the setup. In my case I have an SE7K with a DC battery directly connected to the inverter.

In this setup B can come from three sources:
* solar power
* battery power
* grid (meter power)

As the solar power and the battery power is always routed through the inverter "Power" is already the sum of both. In cases both power sources do not provide enough energy for the house there is also import from the grid. Then I have to add "Power" + "Meter Power" to get B.

In situations where we have enough power to export energy to the grid, we have to subtract "Meter Power" from "Power" to get B.

I have implemented this in a Modbus project [https://github.com/stscde/modbus-esp8266-solaredge/blob/master/src/ModbusSolarEdge.cpp](https://github.com/stscde/modbus-esp8266-solaredge/blob/master/src/ModbusSolarEdge.cpp) I am currently working on.

B is there calculated with the following method. From my comparisons in the last 7 days it gives the same values as the SolarEdge app. Therefore I am quite confident it is ok.

```C++
/**
 * Calculate power used by house (B)
 * @param i_ac_power_norm Inverter output normalized in W
 * @param m1_m_ac_power Power going through the meter in W
 */
int calculate_house_usage(int16_t i_ac_power_norm, int16_t m1_m_ac_power) {
    // NOTE: if we import power from the grid the meter value is negative

    // when exporting: remove the exported power from the inverters output to get the house usage
    // when importing: we have to add the imported power to the inverters output
    return i_ac_power_norm + (m1_m_ac_power * -1);
}
```

But like I said before. In other setups a different calculation may be required.

Best wishes
Stefan

dmaunder, 2021-11-30:
Can you temporarily turn off or disconnect the battery to simplify the diagnosis and understanding? I dont have a battery, and Power setting will always match A in your diagram.

stscde, 2021-12-02:
@dmaunder Thanks for your input. 

When turning off the battery the system works like yours. Inverter output always matches A.

In case battery is turned on I use the following code from my project linked above to calculate the sun power. I'm using this for some month now and it works as expected.

```c
/**
 * Calculate available sun power
 * @param i_ac_power_norm Inverter output normalized in W
 * @param b1_b_instantaneous_power Power from the battery in W
 */
int ModbusSolarEdge::calculate_sun_power(int16_t i_ac_power_norm, float b1_b_instantaneous_power) {
    // if we have no inverter output sun is not shining bright enough
    if (i_ac_power_norm <= 0) {
        return 0;
    }

    // NOTE: when the battery is loading "b1_b_instantaneous_power" is positive when unloading it is negative
    // NOTE 2: "i_ac_power_morm" contains the power used by the house + the power send to the grid but NOT the power sent to the battery BUT the power coming from the battery when unloading

    // when exporting: as the inverter output "i_ac_power_morm" already contains the power send to the grid we do not have to deal with it here
    // when importing: import from battery has to be removed from inverter output "i_ac_power_morm" as we want to have the sun power not the battery power
    i_ac_power_norm += b1_b_instantaneous_power;

    // import from battery may result in values lower 0 because of conversion losses DC > AC
    return i_ac_power_norm > 0 ? i_ac_power_norm : 0;
}
```

Best wishes
Stefan

closed 2021-12-02 by stscde