nmakel/solaredge_modbus #42: support for influxdb 2.0

frezeen, 2021-10-09:
hi,please can u add support for influxdb api v2?

nmakel, 2021-10-09:
Can you share how far you've gotten modifying `example_influxdb.py` to get it to work?

Jusufs, 2022-10-15:
I would prefer to storage not the whole transfer but some parsed important values. I will try to do it. This approach is also valid for openhab persistence.

Stoischo, 2023-02-24:
I have modified the example script for influxdb api v2 (see attachment) according to this documentation: https://www.influxdata.com/blog/getting-started-with-python-and-influxdb-v2-0/ 

Only thing you have do to is to import this library:
`pip install influxdb-client`

Example call:
`./example_influxdbV2.py --influx_token <your_token> influx_org <your_org> --influx_bucket <your_bucket> --influx_url http://<your_ip>:8086 <your_inverter_ip> <your_inverter_port>`

 
[example_influxdbV2.py.zip](https://github.com/nmakel/solaredge_modbus/files/10824696/example_influxdbV2.py.zip)