nmakel/solaredge_modbus #44: Unable to connect to TCP Modbus after upgrade to 4.14.107

lnowakowski, 2021-11-04:
Hi, for starters great project, I'm using this Python module for primary monitoring of my PV installation. This is maybe not an issue for module itself, but more like a question. whether anyone else experience same issue as I did just now, when I upgraded my SolarEdge SE7K-RW0TEBNN4 inverter via SetApp from version for CPU 4.13.x to 4.14.107. After that I get connection refused errors when trying to connect to 1502 port, tried to disable TCP Modbus or change the port number, but nothing helps. Anyone else with same issue, or better - any solution?

lnowakowski, 2021-11-15:
Solved by technical support and some remote adjustments on inverter.

closed 2021-11-15 by lnowakowski

KM192, 2021-11-27:
Have the same issue. Can you described what had be done?

lnowakowski, 2021-11-27:
> Have the same issue. Can you described what had be done?

I had to raise support request (https://www.solaredge.com/service/support/installer) and explain over a couple of days that my wifi is working fine (yes, using wifi instead ethernet, for which support was also complaining, but it worked before and it's not forbidden in manual from SolarEdge), explained how and why I used software based solution, instead of some paid external recorder over RS485, proven that monitoring portal is seeing fine my installation etc. After few days it just started to work again and they stated that engineer did something remotely on my inverter. That's it. But I will not risk installing an upgrade for a while for sure.