nmakel/solaredge_modbus #45: How tp change configuration?

muenchris, 2021-11-04:
I am fairly new to HASS and learn about all the "quirks".
One thing that is very frustrating is the missing (easy) ability to change settings on "integrations".
If I want to add a meter to the SolarEdge Modbus configuration after the initial configuration, how can I do this? I would expect somewhere to get to the dialog that was showing during the first installation of the integration. Do I miss something?

Also, I do have two inverters setup as Leader and Follower via RS485. I do not see an option to configure this anywhere. I can set my second inverter to use Modbus TCP as well but there is no way to install "two" integrations of the same type (IP Address of the second inverter is different).

Again, might be me as a HASS newbee ;)

purcell-lab, 2021-11-05:
>
> If I want to add a meter to the SolarEdge Modbus configuration after the
> initial configuration, how can I do this? I would expect somewhere to get
> to the dialog that was showing during the first installation of the
> integration. Do I miss something?
>

You need to delete your integration and then reinstall with the new
settings for your meter.

> Also, I do have two inverters setup as Leader and Follower via RS485. I do
> not see an option to configure this anywhere. I can set my second inverter
> to use Modbus TCP as well but there is no way to install "two" integrations
> of the same type (IP Address of the second inverter is different).
>

I have my two inverters setup with different IP addresses. I just installed
another instance with the new IP address.

Again, might be me as a HASS newbee ;)
>

We have all been there. Happy to help.

>

muenchris, 2021-11-05:
Thanks, I reloaded my integration and added two meters - works great....but

The SolarEdge API shows that there are 5 meters on my primary inverter but I can only see 2. 
I guess the SolarEdge cloud is splitting the "Import/Export" meter into two (Separate Import and Export meter) and two of the 5 are "virtual" meters: one labeled "Consumption" and the other "Self Consumption". 
Can I only see/access the first two meters using this integration because there might only be two real meters attached to the inverter?

I also have a battery (LG Resu Prime) on each inverter but I cannot access them via Modbus. 

The battery access is actually quite important. My plan is to monitor the battery level during a power outage and shut off some smart (Tuya) switches when the battery gets low. That way the batteries will last longer during an outage for critical devices.

Do you have any idea why the battery access  via Modbus TCP will not work? Their DeviceID is set to #15 (read for the SolarEdge Energy Hub inverter commissioning screen)

muenchris, 2021-11-09:
One more interesting thing: The Modbus TCP server in the SolarEdge Energy Hub seems to be disabled during a "Power Outage - Battery Backup" situation. That prevents the solaredge-modbus integration from reading anything from the system.
I hope SolarEdge is improving on their firmware for the Energy Hub soon

muenchris, 2021-11-09:
Log shows this:

2021-11-09 10:06:11 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading modbus data
Traceback (most recent call last):
File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in async_refresh_modbus_data
update_result = self.read_modbus_data()
File "/config/custom_components/solaredge_modbus/__init__.py", line 230, in read_modbus_data
and self.read_modbus_data_meter2()
File "/config/custom_components/solaredge_modbus/__init__.py", line 247, in read_modbus_data_meter2
return self.read_modbus_data_meter("m2_", 40364)
File "/config/custom_components/solaredge_modbus/__init__.py", line 405, in read_modbus_data_meter
exported = validate(self.calculate_value(exported, energywsf), ">", 0)
File "/config/custom_components/solaredge_modbus/__init__.py", line 124, in validate
raise ValueError(f"Value {value} failed validation ({comparison}{against})")
ValueError: Value 0 failed validation (>0)