nmakel/solaredge_modbus #47: Timeout when trying to connect...

mollele27, 2021-11-27:
Hi!

I have tried to establish contact with my SolarEdge inverter but sofar I always get a timeout when trying:

Inverter(192.168.10.243:502, connectionType.TCP: timeout=1, retries=3, unit=0x1):

I can see at the inverter display that Modbus over TCP is active and ready.
 The inverter SW is 3.2537.

I have tried increasing the timeout but that does not help.

Anybody else with this problem?

Best regards / Leif, Sweden

purcell-lab, 2021-11-27:
> Inverter(192.168.10.243:502, connectionType.TCP: timeout=1, retries=3,
> unit=0x1):
>
From the command line:

Can you 'ping' that address?
Can you 'telnet' to that address and port?

Sometime SolarEdge MODBUS is setup on port 1502 rather than port 502, so
you can try that as well

czartm, 2022-01-07:
Communication stopped working after update SE12.5K to new version (4.14.107). ModBus TCP  in the device settings is set to ON, port 1502.

PING works fine but telnet gives me "connection refused". 

Any ideas?

binsentsu, 2022-01-10:
@czartm Other people have reported that starting from 4.14.107 Solaredge disabled support for modbus TCP through wifi interface. Are you using wifi?

jprinsen, 2022-05-18:
> @czartm Other people have reported that starting from 4.14.107 Solaredge disabled support for modbus TCP through wifi interface. Are you using wifi?

Yes, same here. New CPU version being pushed (4.14.2XX) and suddenly ModbusTCP via WiFi stopped working. 
My device only seems to have a wifi antenna - how difficult would it be to add a LAN port for this purpose?