nmakel/solaredge_modbus #48: active_power_limit

dmaunder, 2021-11-30:
Does anybody understand how to use/interpret the value  "active_power_limit" field that is returned in the json data? It is an integer between 1 and 100. It seems to be related to setting an Export Limit. eg I have a 10kw inverter, and the export limit is set to 5kw. If it is (probably) generating 10kw and exporting 5kw, it will have a value of 50, if it is generating under 5kw, it will have a value of 100. If it is between 5 and 10kw, but it is all being used, ie none is being curtailed, then it will also show 100. I would have thought dividing the AC Power value by this number would show the uncurtailed value, but it doesnt seem to quite work that way. It isnt in the Sunspec documentation either.

Jusufs, 2022-10-15:
If you set for example this F001 parameter to 1%, actually power drops to 1% of nominal photovoltaic power value.
If you can not supply back to grid, according value to zero grid feeding you can dynamically reduce power.
Similar on Fronius inverter with native smart meter.

herbi3, 2023-12-04:
I wouldn't recommend to adjust the power setting, but to instead adjust the export limit via modbus instead.

Adjusting the export control limit option is more accurate in terms of wattage and the inverters do their magic.

Changing the active power percentage, will adjust the output of the inverter to whatever you set it to, which can cause your household to begin pulling from the grid if you set it too low.

PeterVermaas, 2024-04-17:
Agree with herbi3 that it's better to let the inverter do this work but then the inverter should have some metering. 
Is there an example available about how to implement writing this F001 or similar registers like export limit? I don't see a possibility to write registers in this solaredge_modbus-package at this moment.

herbi3, 2024-05-21:
> Agree with herbi3 that it's better to let the inverter do this work but then the inverter should have some metering. Is there an example available about how to implement writing this F001 or similar [...]

@PeterVermaas apologies, I didn't see your reply.  once the inverter is set, you can use `inverter.write("export_control_site_limit", 0)` this will set 0W as the site limit, the default is "11400" for maximum output. ensure `inverter.write("export_control_mode", 1)` is enabled. ("1" for export limit, "4" for production limit)

I recommend keeping the export_control_mode set to 1 or 4. and when an export limit is needed, then to lower the value from 11400 to your choosing, then after reset it back to 11400.

currently, with inverter.write for these specific settings only, there is an endian issue for these registers, try modifying the `__init__.py` ive mentioned [elsewhere](https://github.com/nmakel/solaredge_modbus/issues/36#issuecomment-1412904827). Once you've done that, then set `StorageInverter` as the same you would for `inverter` now, `storage = solaredge_modbus.StorageInverter(parent=inverter, unit=type)`  this does has all the settings for that require little endian.

then use `storage.write("export_control_mode", 1)` and `storage.write("export_control_site_limit", 0)`

LotadaC, 2025-03-09:
> currently, with inverter.write for these specific settings only, there is an endian issue for these registers

Looks like this issue is already solved and merged (#100) 
and maybe, the (now) obsolete StoreEdge-Class will be removed soon (#101)

@herbi3 thank you again, and @nmakel too of course