nmakel/solaredge_modbus #49: Incomplete information example.py

artur5236, 2022-02-02:
Hello,
I using example.py script and get only below information:
`
 python3 example.py 192.168.1.80 1502
Inverter(192.168.1.80:1502, connectionType.TCP: timeout=1, retries=3, unit=0x1):

Registers:
        Manufacturer: SolarEdge
        Model: SE6K-RW0TEBNN4
        Type: Three Phase Inverter
        Version: 0004.0014.0107
        Serial: 7E0E5A7E
        Status: Producing
        Temperature: 19.79°C
        Current: 2.71A
Traceback (most recent call last):
  File "/home/pi/solaredge_modbus/example.py", line 57, in <module>
    print(f"\tPhase 1 Current: {(values['l1_current'] * (10 ** values['current_scale'])):.2f}{inverter.registers['l1_current'][6]}")
KeyError: 'l1_current'
`

artur5236, 2022-02-02:
I tried downgrade to version 0.6.6 and works fine

artur5236, 2022-02-02:
Issue resolveed

nmakel, 2022-02-02:
Hey Artur, was there an issue that needs to be fixed?

artur5236, 2022-02-03:
It was my mistake, I used old version solaredge_modbus package and newest example.py file.
When I upgraded solardge_modbus to new version, issue was reselved.