nmakel/solaredge_modbus #52: Unable to communicate with the inververter

Antoamal, 2022-05-06:
Thank you, your examples and codes helped me a lot. 
I have 3 units in my system, and I tried to test communication with each of them using different unit values 1, 2 and 3. But only got respond for two inverters, I couldn't receive response from second inverter. usage: example.py [-h] [--timeout TIMEOUT] [--unit UNIT] [--json] host port 
Also I couldn't grasp how and when to use multiple inverters part. Shall I all these commands as they are
# Master inverter over Modbus TCP >>> master = solaredge_modbusinverter(host="10.0.0.123", port=1502, unit=1) 
# Second inverter using master's connection >>> second = solaredge_modbusinverter(parent=master, unit=2) 
# Third inverter >>> third = solaredge_modbus.lnverter(parent=master, unit=3) 
Or shall I modify other commands in the example.py to see all there inverter data in one query.

nmakel, 2022-05-06:
> Thank you, your examples and codes helped me a lot. I have 3 units in my system, and I tried to test communication with each of them using different unit values 1, 2 and 3. 

Thanks! Good to hear.

Are you certain the unit IDs are correct? You should be able to determine each inverter's modbus ID in SetApp or using the display menu.

Check out [this](https://www.solaredge.com/sites/default/files/se_configuiring_multiple_slave_inverters_via_a_master.pdf#page2) document from SolarEdge for inverters with an LCD display, specifically the `Device IDs` menu option.

Your code looks fine.

Antoamal, 2022-05-07:
Unfortunately, the inverters that are not responding have no displays. Instead, they have an LED and it's solid blue and green.
Even on the inverters with the display, I am unable to find the unit. There is a serial number in the place of an ID, not unit no.
Is there some other way to get the unit no. or establish comm.?

nmakel, 2022-05-08:
Do you have SetApp access to the two non-display inverters?

In the commisioning menu you can view `Site Communication` -> `RS485-1` and `RS485-2` -> `Device ID`. Your inverters should be configured as one `SolarEdge Leader` and two `SolarEdge Follower`. The `Device ID` is equivalent to the Modbus unit you should provide to `solaredge_modbus`.