nmakel/solaredge_modbus #54: KeyError on example.py

icepick3000, 2022-06-09:
When I run the example.py I get this error;

```
Registers:
Traceback (most recent call last):
  File "./example.py", line 46, in <module>
    print(f"\tManufacturer: {values['c_manufacturer']}")
KeyError: 'c_manufacturer'
```

Any ideas? 

Alex

nmakel, 2022-06-09:
This means you are not connected to the inverter.

icepick3000, 2022-06-09:
I found out from the solaredge manual the modbus can only be reached over ethernet.. so i plugged in ethernet and the script now works 50% of the time. 
Do you know if there is a limit of the number of calls you are allowed to make to the modbus?I was making one every 15 secs but it seems to lock me out after a while.

addiejanssen, 2022-07-09:
I had the same issue with the example. The issue was caused by timeouts.
Try adding --timeout 30 to the command line and see if the issue remains.