nmakel/solaredge_modbus #55: Power of Meter and Inverter not matching

simoon-de, 2022-07-26:
Hi,

i try to use the module on an SE82.8K, (on an SE10 it works perfect) and i get this data:
{'c_manufacturer': 'SolarEdge',
 'c_model': 'SE82.8K-RW0P0BNA4', 
 'c_version': '0004.0015.0119', 
 'c_serialnumber': '7E08B1DA', 
 'c_deviceaddress': 1, 
 'c_sunspec_did': 103, 
 'current': 11017, 
 'l1_current': 3665, 
 'l2_current': 3678, 
 'l3_current': 3672, 
 'current_scale': -2, 
 'l1_voltage': 4065, 
 'l2_voltage': 4062, 
 'l3_voltage': 4056, 
 'l1n_voltage': 2341, 
 'l2n_voltage': 2339, 
 'l3n_voltage': 2351, 
 'voltage_scale': -1, 
 'power_ac': 25859, 
 'power_ac_scale': 0, 
 'frequency': 5000, 
 'frequency_scale': -2, 
 'power_apparent': 25892, 
 'power_apparent_scale': 0, 
 'power_reactive': -13130, 
 'power_reactive_scale': -1, 
 'power_factor': -9986, 
 'power_factor_scale': -2, 
 'energy_total': 172573312, 
 'energy_total_scale': 0, 
 'current_dc': 35090, 
 'current_dc_scale': -3, 
 'voltage_dc': 0, 
 'voltage_dc_scale': -32768, 
 'power_dc': 26252, 
 'power_dc_scale': 0, 
 'temperature': 5405, 
 'temperature_scale': -2, 
 'status': 4, 
 'vendor_status': 0, 
 'rrcr_state': 0,
  'active_power_limit': 100, 
  'cosphi': 0}

{'c_manufacturer': '&P\x02\x7f\x02%[l\x02P\x03P:\x16\x01R0 \x05',

 'c_model': 'False',
  'c_option': 'False', 
  'c_version': 'False', 
  'c_serialnumber': 'False',
   'c_deviceaddress': 0, 
   'c_sunspec_did': 32768, 
   'current': 0, 
   'l1_current': 701, 
   'l2_current': 153, 
   'l3_current': 2, 
   'current_scale': 1, 
   'voltage_ln': -1, 
   'l1n_voltage': 1, 
   'l2n_voltage': 0, 
   'l3n_voltage': 0, 
   'voltage_ll': -1, 
   'l12_voltage': -1, 
   'l23_voltage': 2823, 
   'l31_voltage': 2826, 
   'voltage_scale': -126, 
   'frequency': 99, 
   'frequency_scale': -32768, 
   'power': 4069, 
   'l1_power': 2349, 
   'l2_power': 0, 
   'l3_power': -15562, 
   'power_scale': -1, 
   'power_apparent': -1, 
   'l1_power_apparent': -1, 
   'l2_power_apparent': -1, 
   'l3_power_apparent': -1, 
   'power_apparent_scale': -1, 
   'power_reactive': -1, 
   'l1_power_reactive': -1, 
   'l2_power_reactive': -1, 
   'l3_power_reactive': -1, 
   'power_reactive_scale': -1, 
   'power_factor': -1, 
   'l1_power_factor': -1, 
   'l2_power_factor': -1, 
   'l3_power_factor': -1, 
   'power_factor_scale': -1, 
   'export_energy_active': 2147516416, 
   'l1_export_energy_active': 2147516416, 
   'l2_export_energy_active': 2147516416, 
   'l3_export_energy_active': 2147516416, 
   'import_energy_active': 2147549183, 
   'l1_import_energy_active': 0, 
   'l2_import_energy_active': 154009599, 
   'l3_import_energy_active': 0, 
   'energy_active_scale': -1, 
   'export_energy_apparent': 0, 
   'l1_export_energy_apparent': 0, 
   'l2_export_energy_apparent': 0, 
   'l3_export_energy_apparent': 0, 
   'import_energy_apparent': 0, 
   'l1_import_energy_apparent': 0, 
   'l2_import_energy_apparent': 2147516416, 
   'l3_import_energy_apparent': 2147549183, 
   'energy_apparent_scale': -1, 
   'import_energy_reactive_q1': 4294904108, 
   'l1_import_energy_reactive_q1': 0, 
   'l2_import_energy_reactive_q1': 0, 
   'l3_import_energy_reactive_q1': 0, 
   'import_energy_reactive_q2': 0, 
   'l1_import_energy_reactive_q2': 0, 
   'l2_import_energy_reactive_q2': 0, 
   'l3_import_energy_reactive_q2': 0, 
   'export_energy_reactive_q3': 0, 
   'l1_export_energy_reactive_q3': 2147516416, 
   'l2_export_energy_reactive_q3': 2147549183, 
   'l3_export_energy_reactive_q3': 0, 
   'export_energy_reactive_q4': 154337279, 
   'l1_export_energy_reactive_q4': 0, 
   'l2_export_energy_reactive_q4': 0, 
   'l3_export_energy_reactive_q4': 0, 
   'energy_reactive_scale': -1}

There is an error in the power of the meter, the one from the inverter has 'power_ac': 25859, and the Meter has    'power': 4069, which is not true. it must be also around 24000 at this moment. i think there is something wrong.
How can this be fixed.

Is there a way to get import power, export power and used in home power?

Thanks
Alex

nmakel, 2022-07-26:
And with regard to your other question, self consumption is a function of import, export and production, all of which are present in the data. Your import/export values look really weird, though. Some more info on the meter and how it is configured would be helpful.

simoon-de, 2022-07-26:
It is the Solaredge SE-WND-300-4Y-MB. It is connected to the RS485-1 port of the inverter and show as id #2 
here a screenshot of the setapp status

[image]

simoon-de, 2022-07-26:
Maybe there is a Problem with the primary, left, right parts of the WR. The value power_ac: 25859  the inverter shows in the data above is right and matches the solaredge app. Allso the app is working fine and showing the right data.

nmakel, 2022-07-26:
Looking at the meter data more closely practically all values are wrong or outright gibberish. You said the meter values are correct when you used a SE10 inverter -- is that unit using a much older firmware than 4.15.x that your SE82 is using? It could be a number of things right now, perhaps the register offsets for meters changed, or the endianness...

Are you comfortable modifying the underlying python code?

simoon-de, 2022-07-27:
i checked the inverters
on the inverters both have the 4.15.119 on cpu and on the others its the same, but the meter is different, i did not realize this. 
i think i am able to modify the code.
is it possible to dump all data coming from the inverters and meta just to see what happens?

[image]

[image]

simoon-de, 2022-07-27:
ok the installed counter is an SE-MTR-3Y-400V-A which has another pupose than the SE-WND-300-4Y-MB, i think i have to add an SE-WND-300-4Y-MB and it will work, right?

nmakel, 2022-07-27:
Thanks for the extra details. Can you also screenshot the meter rs485 configuration page in setapp?

simoon-de, 2022-07-28:
sure :)
[image]
[image]

nmakel, 2022-07-31:
So the hunch I have is that this is related to the synergy tech in your larger inverters, which causes meter registers to be shifted by 50 in the register mapping. Could you modify the following block of code:

https://github.com/nmakel/solaredge_modbus/blob/add6a3d96f9412c2d1caddb098a37f5cc50beaeb/src/solaredge_modbus/__init__.py#L130-L134

And change `0x0` to `0x32`. Leave the others untouched, unless you have a second or third meter, in which case they should read `0xE0` and `0x18E`.

This should cause your meter values to align correctly.

simoon-de, 2022-08-01:
i changed this in the file 
/usr/local/lib/python3.9/dist-packages/solaredge_modbus/__init__.py
but there is no difference in the output ot the power values.

nmakel, 2022-08-01:
Bah! You're sure the meter (and the script) does work when connected to the SE10k inverter, right?

nmakel, 2022-08-02:
Ok, tut mir leid, I didn't completely read the documentation. You have three units coupled together, which according to the sunspec docs require an offset of 70, not 50. Could you please try again, now with the offsets `0x46`, `0xF4` and `0x1A2` on lines 131 through 133.

simoon-de, 2022-08-03:
thanks for you helping, i tryed but it is not working. 
I did not test the same meter at the SE10k its a different  SE-MTR-3Y-400V-A  vs. SE-WND-300-4Y-MB.
i Think i get another SE-WND-300-4Y-MB because we will expand the solar power anyway.

here is the output of the meter actual meter with the modification:

{'c_manufacturer': '\x02\x01\x01\x14\x169\\\x10,\tV', 'c_model': 'N', 'c_option': 'False', 'c_version': '\tV', 'c_serialnumber': 'False', 'c_deviceaddress': 32768, 'c_sunspec_did': 0, 'current': -1, 'l1_current': 2377, 'l2_current': -1, 'l3_current': -1, 'current_scale': -1, 'voltage_ln': -1, 'l1n_voltage': -1, 'l2n_voltage': -1, 'l3n_voltage': -1, 'voltage_ll': -1, 'l12_voltage': -1, 'l23_voltage': -1, 'l31_voltage': -1, 'voltage_scale': -1, 'frequency': -1, 'frequency_scale': -1, 'power': -1, 'l1_power': -1, 'l2_power': -32768, 'l3_power': -32768, 'power_scale': -32768, 'power_apparent': -1, 'l1_power_apparent': -1, 'l2_power_apparent': -1, 'l3_power_apparent': 2400, 'power_apparent_scale': -1, 'power_reactive': -1, 'l1_power_reactive': -1, 'l2_power_reactive': -1, 'l3_power_reactive': -1, 'power_reactive_scale': -1, 'power_factor': -1, 'l1_power_factor': -1, 'l2_power_factor': -1, 'l3_power_factor': -1, 'power_factor_scale': -1, 'export_energy_active': 0, 'l1_export_energy_active': 0, 'l2_export_energy_active': 0, 'l3_export_energy_active': 0, 'import_energy_active': 65535, 'l1_import_energy_active': 4294770689, 'l2_import_energy_active': 4294836225, 'l3_import_energy_active': 65536, 'energy_active_scale': 0, 'export_energy_apparent': 0, 'l1_export_energy_apparent': 0, 'l2_export_energy_apparent': 0, 'l3_export_energy_apparent': 0, 'import_energy_apparent': 0, 'l1_import_energy_apparent': 0, 'l2_import_energy_apparent': 0, 'l3_import_energy_apparent': 0, 'energy_apparent_scale': 0, 'import_energy_reactive_q1': 0, 'l1_import_energy_reactive_q1': 0, 'l2_import_energy_reactive_q1': 0, 'l3_import_energy_reactive_q1': 0, 'import_energy_reactive_q2': 0, 'l1_import_energy_reactive_q2': 0, 'l2_import_energy_reactive_q2': 0, 'l3_import_energy_reactive_q2': 0, 'export_energy_reactive_q3': 46006322, 'l1_export_energy_reactive_q3': 1131414208, 'l2_export_energy_reactive_q3': 5243584, 'l3_export_energy_reactive_q3': 5260144, 'export_energy_reactive_q4': 3255878160, 'l1_export_energy_reactive_q4': 0, 'l2_export_energy_reactive_q4': 0, 'l3_export_energy_reactive_q4': 150735330, 'energy_reactive_scale': 2023}