nmakel/solaredge_modbus #57: Dont' see battery or meters information

rploeg, 2022-09-30:
Hi,

I have three inverts (and 1 meter) and 1 battery. But I don't see that back in the JSON export. I see this data in the solaredge app, but not in this script. Any idea what I need to do? Below the output and the last two arrays are empty.

[image]

nmakel, 2022-09-30:
Hi Remco, could you provide the inverter model/types and firmware versions? There was another issue a while back similar to yours which I think was caused by alternate register locations for a specific type of inverter.

rploeg, 2022-09-30:
"c_manufacturer": "SolarEdge",
    "c_model": "SE3500H-RW000BNN4",
    "c_version": "0004.0012.0035",

nmakel, 2022-09-30:
Ok, that's odd. The other issue was with synergy inverters.

I wrote a quick and dirty script to map all relevant register blocks. Would you mind running it, and posting the results here?

[mapse.txt](https://github.com/nmakel/solaredge_modbus/files/9683233/mapse.txt)

rploeg, 2022-09-30:
Here it is:

[output.txt](https://github.com/nmakel/solaredge_modbus/files/9683777/output.txt)

nmakel, 2022-09-30:
Thanks! The output looks like what you'd expect given the output of example.py.

You say you have three inverters, can you confirm the following:

- They are three SE3500H units, each single phase, each on a different phase.
- They are connected together via the RS485 bus, and not each inverter individually to your network via ethernet or WiFi.
- They are configured to operate in a master-slave-slave mode.
- The kwh meter is connected to the primary inverter, which has IP address 192.168.1.246.
- The battery (or its storedge module) is also connected to the primary inverter.

Most of these questions can be answered by checking the SetApp interface on your primary inverter. There, check the "site communication" settings for RS485-1 and RS485-2. I would expect to see at least one of the interfaces set to SolarEdge Leader, and another to Modbus Multi-device.

rploeg, 2022-09-30:
Hi,

Yes there are al connect to a single phase (and are single phase inverter).
1xse6000H (primary)
1x se3500H without display
1x se1500h
They are connected to RS485 and only the main one (192.168.1.246) is connected to the internet
Yes, master and slave
KWH meter is connected to the primary inverter
Yes, battery is directly connected to the primary inverter

So I found a strange thing on my side. the primary (6000) has the IP address of the SE3500H (that one is also connected + and via modbus connected). It makes it even more strange because the 6000 is connected to a managed port where I can see the IP address of (that is 192.168.1.246), but when I connect now with your script to that IP address I am connected to the 3500. So I have to figure this first out with my installer what they have done...

nmakel, 2022-09-30:
Have you tried using a different `--unit`?

rploeg, 2022-09-30:
Perfect, thank you! Unit 2 has the batteries. Interesting to see why 2 converts has network connectivity.

nmakel, 2022-09-30:
Excellent, graag gedaan.

rploeg, 2022-09-30:
Jij voor je werk in deze library!

closed 2022-09-30 by rploeg