nmakel/solaredge_modbus #64: Parent Inverter Modbus TCP not working

rmnsrrs, 2022-12-09:
I am trying to query a system with 2 inverters. 
If I run the example script with each unit (1 & 2) then it is working. 
However the following code does not work and query twice the inverter 1.
I notice that if I change the order in the list I then get twice the inverter 2.
In other word I get the 1st inverter in the list twice and never the 2nd one.
If that normal behavior ? How should I use parent parameter ?
Thanks

`#!/usr/bin/env python3

import argparse
import json

import solaredge_modbus

class solarEdge_extractor(object):
    def __init__(self, host, port):
        self.inverters = [solaredge_modbus.Inverter(
            host=host,
            port=port,
            unit=1
        )]
        self.inverters.append(solaredge_modbus.Inverter(
            parent=self.inverters[0],
            unit=2
        ))

    def get_data(self):
        values = {}
        values["inverters"] = {}
        values["meters"] = {}
        values["batteries"] = {}
        for i,c_i in enumerate(self.inverters):
            print(c_i.read("current"))
            values["inverters"][i] = c_i.read_all()

            meters = c_i.meters()
            batteries = c_i.batteries()
            values["meters"][i] = {}
            for meter, params in meters.items():
                meter_values = params.read_all()
                values["meters"][i][meter] = meter_values

            values["batteries"][i] = {}
            for battery, params in batteries.items():
                battery_values = params.read_all()
                values["batteries"][i][battery] = battery_values

        return json.dumps(values, indent=4)

if __name__ == "__main__":
    argparser = argparse.ArgumentParser()
    argparser.add_argument("host", type=str, help="Modbus TCP address")
    argparser.add_argument("port", type=int, help="Modbus TCP port")
    args = argparser.parse_args()
    se = solarEdge_extractor(args.host, args.port)
    print(se.get_data())`

rmnsrrs, 2022-12-09:
You were correct, It did fix the issue. However now the meter is now empty. While it worked perfectly before. It detects "Meter1" but return an empty dict. Any chance you would know why ?

rmnsrrs, 2022-12-09:
Yes sorry, it was not clear. But I am always testing with the example given on the git.
When calling unit 2, I can see the meter is "Meter1" but  meter_values is an empty dict {} (cf below)
Also I reverted back to the pipy version and the meter is working fine again.
It seems that there is bug introduced in between the pipy version and the latest github

        for meter, params in meters.items():
            meter_values = params.read_all()
            values["meters"][meter] = meter_values
            print(meter_values)

rmnsrrs, 2022-12-09:
I can confirm after several test that this is only due do these lines:

             if unit:
                self.unit = unit
            else:
                self.unit = parent.unit

I will try to investigate why this is happening

jgyates, 2023-09-10:
I am having a similar issue. the problem is that in the above lines:

         if unit:
            self.unit = unit
        else:
            self.unit = parent.unit

If unit is anything other than 1 then it is false. Here is the output form the python console:

        >>> import solaredge_modbus
        >>> inv1 = solaredge_modbus.Inverter(host = "192.168.1.51", port=1502, unit =1, timeout=3)
        >>> print(inv1)
        Inverter(192.168.1.51:1502, connectionType.TCP: timeout=3, retries=3, unit=0x1)
        >>> inv2 = solaredge_modbus.Inverter(parent=inv1, unit = 2)
        >>> print(inv2)
        Inverter(192.168.1.51:1502, connectionType.TCP: timeout=3, retries=3, unit=0x1)
        >>> inv2.unit == True
        True
        >>> 2 == True
        False
        >>> 1 == True
        True

This makes the 2nd inverter inaccessible if trying to access two inverters (via a parent) in the same program. It works with example.py if you pass a unit of 2 as there is no parent in that case.

jgyates, 2023-09-10:
If you change the lines:

 if unit:
        self.unit = unit
    else:
        self.unit = parent.unit

to 

 if unit != parent.unit:
        self.unit = unit
    else:
        self.unit = parent.unit

it corrects the problem.