nmakel/solaredge_modbus #65: Connection and polling interval best practices ?

jburdy, 2022-12-17:
Hi,
Thanks for the job, it's been running in my house for a few months now and it's allowed me to automate my W consumers very well! Mainly from Home-Assistant.

What is the best practice for fetching the data? **Leave the connection open between pollings or close it each time?**

And **how often can we pool the data?** Today I do it every 20 seconds... I don't know if I'm abusing or if it's slow!

Thanks for your advice,

jburdy, 2022-12-18:
Yes today I have a variant of your example running in a systemd timer. I'm thinking of putting it into service. So I'm going to try to leave the connection open, but I'm not a fan of thinking that it's blocking access to other connection. I hope it doesn't block anything on the SolarEdge side either. Although my access should be exclusive (after I relay to MQTT for all my other consumers. I took the liberty of making a push request, it might be interesting for others).

closed 2022-12-18 by jburdy

Jusufs, 2023-01-02:
> Hi, Thanks for the job, it's been running in my house for a few months now and it's allowed me to automate my W consumers very well! Mainly from Home-Assistant.
>
> What is the best practice for [...]

I can poll via python script each 5 seconds. My python poll part sends data to Raspi Openhab2, Influx, Grafana.

Roemke, 2024-12-01:
I tried to poll every 5 seconds but the real time is around 10 seconds (see on timestamp)
If I continously read (1 second delay) I get the Data every 5 seconds. 
And I get
Connection to (192.168.0.205, 1502) failed: [Errno 111] Connection refused
Traceback (most recent call last):
  File "/home/pi/solaredge_modbus/./influx2WebServ.py", line 96, in <module>
    "c_model": meter_values["c_model"],
KeyError: 'c_model'
after some minutes :-(