nmakel/solaredge_modbus #71: Support for Backup Interface

herbi3, 2023-01-29:
With solaredge BUI, it shows if the system is running on or off grid. It would be great to somehow grab that information.

These are connected via rs485, but I don't believe there's anything modbus related. Do you or anyone have an idea of how to determine if the system is running in on or off grid mode?

This is their install guide, nothing mentioned about modbus etc. nothing shows under modbus section within the inverter either (I was hoping to grab a modbus ID but no luck)
https://964176.app.netsuite.com/core/media/media.nl?id=16354282&c=964176&h=1ff37e5f176af711cf09&_xt=.pdf#page19

herbi3, 2023-01-29:
thanks @nmakel !

I had to change `from pymodbus.client import ModbusTcpClient` to `from pymodbus.client.sync import ModbusTcpClient`

I have no other connections active to the inverter, but it keeps failing. are there any other recent changes from pymodbus I need to include?

```
Traceback (most recent call last):
  File "./mapse.py", line 33, in <module>
    result = client.read_holding_registers(i, 1, slave=args.unit)
  File "/usr/local/lib/python3.8/dist-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.8/dist-packages/pymodbus/client/sync.py", line 108, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(10.0.2.3:1502)]
```

herbi3, 2023-01-31:
thanks @nmakel - here is the output_with_backup

```
40000 21365
40001 28243
40002 1
40003 65
40004 21359
40005 27745
40006 29253
40007 25703
40008 25888
40009 0
40010 0
40011 0
40012 0
40013 0
40014 0
40015 0
40016 0
40017 0
40018 0
40019 0
40020 21317
40021 12592
40022 12336
40023 12360
40024 11585
40025 21843
40026 20034
40027 16984
40028 12596
40029 0
40030 0
40031 0
40032 0
40033 0
40034 0
40035 0
40036 0
40037 0
40055 12850
40056 0
40057 0
40058 0
40059 0
40060 0
40062 0
40063 0
40064 0
40065 0
40066 0
40068 1
40069 101
40070 50
40072 653
40073 65535
40074 65535
40075 65534
40077 65535
40078 65535
40079 2416
40080 65535
40081 65535
40082 65535
40083 15494
40084 65535
40085 49971
40086 65533
40087 15530
[... 1963 lines cut ...]
59696 0
59712 255
59786 0
59787 0
59788 0
59789 0
59790 0
59791 0
59792 0
59793 0
59794 0
59795 0
59796 0
59797 0
59798 0
59799 0
59800 0
59801 0
59802 0
59904 0
59920 0
59936 0
59952 0
59968 255
```

Ruebenschus, 2023-11-08:
Hi there,
I have a new hybrid inverter with backup box, too. I think my backup box is newer than the one mentioned in the thread. It has a little different form factor. It's called SolarEdge Home Backup-Interface. I also have two batteries installed, plus one meter at the main network inlet and one within the Backup-Interface. 

solaredge_modbus does only see the smart meter and the batteries, but not the Backup-Interface with builtin meter. 
Is there anything I can provide to help you skilled guys to include those devices?

Cheers,
 Joachim

The mapse.py sais "Failed to connect" whereas example.py shows (in the case of not-backup operation):

`{
    "c_id": "SunS",
    "c_did": 1,
    "c_length": 65,
    "c_manufacturer": "SolarEdge",
    "c_model": "SE10K-RWB48BFN4",
    "c_version": "0004.0017.0136",
    "c_serialnumber": "..........",
    "c_deviceaddress": 1,
    "c_sunspec_did": 103,
    "c_sunspec_length": 50,
    "current": 178,
    "l1_current": 54,
    "l2_current": 65,
    "l3_current": 57,
    "current_scale": -2,
    "l1_voltage": 4019,
    "l2_voltage": 4008,
    "l3_voltage": 4028,
    "l1n_voltage": 2323,
    "l2n_voltage": 2317,
    "l3n_voltage": 2320,
    "voltage_scale": -1,
    "power_ac": 3833,
    "power_ac_scale": -1,
    "frequency": 4998,
    "frequency_scale": -2,
    "power_apparent": 4129,
    "power_apparent_scale": -1,
    "power_reactive": -15348,
    "power_reactive_scale": -2,
    "power_factor": -9271,
    "power_factor_scale": -2,
    "energy_total": 10944,
    "energy_total_scale": 0,
    "current_dc": 4742,
    "current_dc_scale": -4,
    "voltage_dc": 8204,
    "voltage_dc_scale": -1,
    "power_dc": 3891,
    "power_dc_scale": -1,
    "temperature": 3794,
    "temperature_scale": -2,
    "status": 4,
    "vendor_status": 0,
    "rrcr_state": 0,
    "active_power_limit": 100,
    "cosphi": 0,
    "commit_power_control_settings": 0,
    "restore_power_control_default_settings": 0,
    "reactive_power_config": 0,
    "reactive_power_response_time": 200,
    "advanced_power_control_enable": 1,
    "export_control_mode": 0,
    "export_control_limit_mode": 0,
    "export_control_site_limit": 0,
    "meters": {
        "Meter1": {
            "c_manufacturer": "SolarEdge",
[... 87 lines cut ...]
            "c_sunspec_did": 0,
            "rated_energy": 9200.0,
            "maximum_charge_continuous_power": 5650.0,
            "maximum_discharge_continuous_power": 8192.0,
            "maximum_charge_peak_power": 5150.0,
            "maximum_discharge_peak_power": 8600.0,
            "average_temperature": 18.30000114440918,
            "maximum_temperature": 0.0,
            "instantaneous_voltage": 819.8668823242188,
            "instantaneous_current": -1.5764518976211548,
            "instantaneous_power": 1290.0,
            "lifetime_export_energy_counter": 82,
            "lifetime_import_energy_counter": 3081,
            "maximum_energy": 9200.0,
            "available_energy": 8280.0,
            "soh": 100.0,
            "soe": 42.22222137451172,
            "status": 3,
            "status_internal": 3,
            "event_log": 0,
            "event_log_internal": 0
        }
    }
}
`

closed 2023-12-09 by herbi3