nmakel/solaredge_modbus #75: How to interpret the data returned by example.py

pictureaday, 2023-03-15:
Below is the return I get. It looks fairly different than what I see other folks posting I am able to easily understand what is being produced but I cannot for the life of me figure out how to figure out how much is exported/consumed and was hoping someone here could help. Mysolaredge accurately has the values I'm looking for but that's a website guarded by a captcha, so that won't do for my purposes.

Thanks!

Command run :
`./example.py REDACTED_IP  1502 --json`

{
    "c_id": "SunS",
    "c_did": 1,
    "c_length": 65,
    "c_manufacturer": "SolarEdge",
    "c_model": "SE11400H-US000BEI4",
    "c_version": "0004.0017.0046",
    "c_serialnumber": "REDACTED",
    "c_deviceaddress": 1,
    "c_sunspec_did": 102,
    "c_sunspec_length": 50,
    "current": 1133,
    "l1_current": 1133,
    "l2_current": 0,
    "l3_current": 0,
    "current_scale": -2,
    "l1_voltage": 2423,
    "l2_voltage": 0,
    "l3_voltage": 0,
    "l1n_voltage": 1213,
    "l2n_voltage": 1210,
    "l3n_voltage": 0,
    "voltage_scale": -1,
    "power_ac": 27418,
    "power_ac_scale": -1,
    "frequency": 60003,
    "frequency_scale": -3,
    "power_apparent": 27460,
    "power_apparent_scale": -1,
    "power_reactive": 15074,
    "power_reactive_scale": -2,
    "power_factor": 9984,
    "power_factor_scale": -2,
    "energy_total": 1465843,
    "energy_total_scale": 0,
    "current_dc": 7241,
    "current_dc_scale": -3,
    "voltage_dc": 3844,
    "voltage_dc_scale": -1,
    "power_dc": 27836,
    "power_dc_scale": -1,
    "temperature": 3604,
    "temperature_scale": -2,
    "status": 4,
    "vendor_status": 0,
    "meters": {
        "Meter1": {
            "c_manufacturer": "SolarEdge",
            "c_model": "SE-RGMTR-1D-240C-B",
            "c_option": "Production",
            "c_version": "78",
            "c_serialnumber": "REDACTED",
            "c_deviceaddress": 1,
            "c_sunspec_did": 203,
            "c_sunspec_length": 105,
            "current": 120,
            "l1_current": 0,
            "l2_current": 0,
            "l3_current": 120,
            "current_scale": -1,
            "voltage_ln": 0,
            "l1n_voltage": 0,
            "l2n_voltage": 0,
            "l3n_voltage": 24252,
            "voltage_ll": 0,
            "l12_voltage": 0,
            "l23_voltage": 0,
            "l31_voltage": 0,
            "voltage_scale": -2,
            "frequency": 6000,
            "frequency_scale": -2,
            "power": 2643,
            "l1_power": 0,
            "l2_power": 0,
            "l3_power": 2643,
            "power_scale": 0,
            "power_apparent": 2643,
            "l1_power_apparent": 0,
            "l2_power_apparent": 0,
            "l3_power_apparent": 2643,
            "power_apparent_scale": 0,
            "power_reactive": 33,
            "l1_power_reactive": 0,
            "l2_power_reactive": 0,
            "l3_power_reactive": 33,
            "power_reactive_scale": 0,
            "power_factor": 9990,
            "l1_power_factor": 0,
            "l2_power_factor": 0,
            "l3_power_factor": 9990,
            "power_factor_scale": -2,
            "export_energy_active": 1461528,
            "l1_export_energy_active": 0,
            "l2_export_energy_active": 0,
            "l3_export_energy_active": 1461528,
            "import_energy_active": 3797,
            "l1_import_energy_active": 0,
            "l2_import_energy_active": 0,
            "l3_import_energy_active": 3797,
            "energy_active_scale": 0,
            "export_energy_apparent": 0,
            "l1_export_energy_apparent": 0,
            "l2_export_energy_apparent": 0,
            "l3_export_energy_apparent": 0,
            "import_energy_apparent": 0,
            "l1_import_energy_apparent": 0,
            "l2_import_energy_apparent": 0,
            "l3_import_energy_apparent": 0,
            "energy_apparent_scale": -32768,
            "import_energy_reactive_q1": 0,
            "l1_import_energy_reactive_q1": 0,
            "l2_import_energy_reactive_q1": 0,
            "l3_import_energy_reactive_q1": 0,
            "import_energy_reactive_q2": 0,
            "l1_import_energy_reactive_q2": 0,
            "l2_import_energy_reactive_q2": 0,
            "l3_import_energy_reactive_q2": 0,
            "export_energy_reactive_q3": 0,
            "l1_export_energy_reactive_q3": 0,
            "l2_export_energy_reactive_q3": 0,
            "l3_export_energy_reactive_q3": 0,
            "export_energy_reactive_q4": 0,
            "l1_export_energy_reactive_q4": 0,
            "l2_export_energy_reactive_q4": 0,
            "l3_export_energy_reactive_q4": 0,
            "energy_reactive_scale": -32768
        }
    },
    "batteries": {
        "Battery1": {},
        "Battery2": {}
    }
}

herbi3, 2023-03-15:
The example shows the raw values from the inverter only, the solar production values need to be multiplied by the scale as a fraction ( 10**scale ) 

There is an influxdb script from the great  @nmakel that does this, and posts the data to an influxdb instance.

I use Emoncms, where my data from my inverters, batteries and meters are first posted to Emoncms, from there I post process certain data (I.e consumption = solar production + grid import/export) 

From there, it goes to Emoncms mqtt, then telegraf > influxdb then the data is visualised with grafana. My public dashboard link is below, if you want to do something similar to me, I can help replicate it for you. This will take a few weeks if you want to do it yourself with my scripts etc. (a few weeks because I can't really help during the day)

HTTPS://public.winterfell.tv/

pictureaday, 2023-03-15:
I appreciate your response. Your site looks great, I'm not a python guy but I'll def look at that more.

I do want to clarify what I'm trying to do here (because either I don't understand your response or I didn't ask my question correctly):  with the data in my first post I expect that I'd be able to figure out how much power my home is using at any given moment, but I don't know what fields/values to use to make that happen.

pictureaday, 2023-03-15:
Unintentionally closed it

herbi3, 2023-03-15:
thanks @pictureaday - you are reading only the data from the inverter itself, you also need to read the data from the solaredge meter. the meter usually has a different modbus ID, and different values.

Once you are reading both devices (inverter and meter) then you can calculate the consumption.

if you only read from the inverter, then you will only be able to get the solar production metric.

which in python, that will be:

power_ac * (10** power_ac_scale) = solar production in watt-hours

in your data above, that value from the calculation is **2,741.8 watts**

27418 * (10**-1)

(10**-1) = 0.1
27418 * 0.1 = 2,741.8

this is important, because the scale will change throughout the day is the solar production goes above a certain value. the raw value will become something completely different, and that's where the scale calculation is needed.

pictureaday, 2023-03-15:
`"Meter1": {
"c_manufacturer": "SolarEdge",
"c_model": "SE-RGMTR-1D-240C-B",
"c_option": "Production",
"c_version": "78",
"c_serialnumber": "REDACTED",
"c_deviceaddress": 1,
"c_sunspec_did": 203,`

Is this not my meter data?

herbi3, 2023-03-15:
ah @pictureaday I didn't see that!

then your consumption based on that is:

GRID = "l3_power" * (10 ** "power_apparent_scale")
SCALE = (10** 0) = 0
GRID = (2643 * 0) * **-1**

the negative x1 is because solaredge data seems reversed (when exporting, by default is shows a positive number. when importing it shows a negative number. for simplicity of other calculations, I always multiply the GRID value by -1. then the data when exporting is a negative number etc. easier for future calculations)

GRID = -2,643 watt-hours

CONSUMPTION = SOLAR+GRID
CONSUMPTION = 2,741.8 + -2643 = 98 watt-hours, 0.098kwh

pictureaday, 2023-03-15:
Any chance it's off by a factor of 10 here? Our home is usually around 0.93-1.05 kw unless something big is on.

Also what is 10**?

herbi3, 2023-03-15:
l3_power is used, because it looks like the Split core transformer was connected to Line #3 in the meter, instead of Line #1.

you can install and configure your own influx database, and post the data into influxdb (the influxdb script would need to be modified to change the l3_power to always include the *-1 calculation before posting into influxdb.

Then install and configure Grafana, connect grafana to influxdb. Create a dashboard, and have grafana do the calculation for consumption.

pictureaday, 2023-03-15:
That's a great idea! I'll do that, thank you!

closed 2023-03-15 by pictureaday

herbi3, 2023-03-15:
> Any chance it's off by a factor of 10 here? Our home is usually around 0.93-1.05 kw unless something big is on.
> 
> Also what is 10**?

the 10** means raising 10 to the power of power_scale. without this, you'll eventually see some crazy high and irrelevant  solar production value as the l3_power value is dependant on the scale for us humans to understand.