nmakel/solaredge_modbus #77: Data returned by example.py

Topsham, 2023-04-12:
For my own smart home needs I am trying to extract a home consumption figure from the modbus inverter and meter readouts. It appears this should be (power_ac x power_ac_scale) - (power x power_scale). 

At the time the example.py data below was returned, the SolarEdge app was indicating Solar Production of  0.98kW and home consumption of 2.65kW. But using the returned data with the above equation would get nowhere close to the correct consumption figure. Am I missing something?

PS - The battery 'soe's were correct however.
```
{
    "c_manufacturer": "SolarEdge",
    "c_model": "SE5000H-RWSACBNN4",
    "c_version": "0004.0017.0218",
    "c_serialnumber": "741570BD",
    "c_deviceaddress": 2,
    "c_sunspec_did": 101,
    "current": 647,
    "l1_current": 647,
    "l2_current": 0,
    "l3_current": 0,
    "current_scale": -2,
    "l1_voltage": 2419,
    "l2_voltage": 0,
    "l3_voltage": 0,
    "l1n_voltage": 2419,
    "l2n_voltage": 0,
    "l3n_voltage": 0,
    "voltage_scale": -1,
    "power_ac": 15577,
    "power_ac_scale": -1,
    "frequency": 50036,
    "frequency_scale": -3,
    "power_apparent": 15651,
    "power_apparent_scale": -1,
    "power_reactive": 15282,
    "power_reactive_scale": -2,
    "power_factor": 9952,
    "power_factor_scale": -2,
    "energy_total": 148733,
    "energy_total_scale": 0,
    "current_dc": 4100,
    "current_dc_scale": -3,
    "voltage_dc": 3857,
    "voltage_dc_scale": -1,
    "power_dc": 15814,
    "power_dc_scale": -1,
    "temperature": 2900,
    "temperature_scale": -2,
    "status": 4,
    "vendor_status": 0,
    "rrcr_state": 0,
    "active_power_limit": 100,
    "cosphi": 0,
    "meters": {
        "Meter1": {
            "c_manufacturer": "WattNode",
            "c_model": "WND-3Y-400-MB",
            "c_option": "Production ext.",
            "c_version": "31",
            "c_serialnumber": "5090883",
            "c_deviceaddress": 1,
            "c_sunspec_did": 203,
            "current": 43,
            "l1_current": 43,
            "l2_current": 0,
            "l3_current": 0,
            "current_scale": -1,
            "voltage_ln": 24177,
            "l1n_voltage": 24177,
            "l2n_voltage": 0,
            "l3n_voltage": 0,
            "voltage_ll": 23945,
            "l12_voltage": 23930,
[... 175 lines cut ...]
            "c_serialnumber": "6307B382",
            "c_deviceaddress": 45954,
            "c_sunspec_did": 0,
            "rated_energy": 9700.0,
            "maximum_charge_continuous_power": 5000.0,
            "maximum_discharge_continuous_power": 5000.0,
            "maximum_charge_peak_power": 5000.0,
            "maximum_discharge_peak_power": 5000.0,
            "average_temperature": 12.441507339477539,
            "maximum_temperature": 0.0,
            "instantaneous_voltage": 383.43597412109375,
            "instantaneous_current": 2.1353626251220703,
            "instantaneous_power": -820.0,
            "lifetime_export_energy_counter": 84061,
            "lifetime_import_energy_counter": 93966,
            "maximum_energy": 9700.0,
            "available_energy": 9112.5,
            "soh": 99.0,
            "soe": 88.29080963134766,
            "status": 4,
            "status_internal": 2,
            "event_log": 0,
            "event_log_internal": 0
        }
```

herbi3, 2023-04-12:
When multiplying by the scale, did you use 10 raised to the power of the scale?

Scale being -1, then it would be

Power_ac * (10** scale)
Power_ac * (10** -1)

Topsham, 2023-04-12:
Apologies - for clarity I should have included the calculation initially. Consumption = (power_ac x power_ac_scale) - (power x power_scale)

Using the example.py figures this gives the result below of 331.7w compared to the actual consumption at the time example.py was run of 2.65kW so it's not even close. Any thoughts? 
```
>>> result = (15577*(10**-1)) - (1226*(10**0))
>>> print(result)
331.70000000000005
```

herbi3, 2023-04-13:
> Apologies - for clarity I should have included the calculation initially. Consumption = (power_ac x power_ac_scale) - (power x power_scale)
>
>
>
> Using the example.py figures this gives the result [...]

That's fairly odd. What was the last reported time within the solaredge app?

The script also isn't asynchronous, so it will sequentially get the data, one after the other. The meter data and batteries will be a few seconds behind each other, the consumption amount will vary significantly because of that too. If you run the same script with an interval of a few seconds for about an hour or so, then visualise the data, you may see periodic spikes from "dirty" data

Topsham, 2023-04-13:
It does indeed seem to be a timing issue. I have fitted my own current clamp to the house consumer unit and compared it to the home consumption figures on the app. They are never the same although they 'trend' correctly over time. As the house consumption is the most dynamic reading, I have decided to use the readings from my own clamp for that and use the modbus readings for solar production and battery % for my smart home needs. So many thanks for your excellent work on this project and I will close this issue.