nmakel/solaredge_modbus #81: Which Parameter is which

PanterSoft, 2023-07-02:
Hello,

First of all great project. it works like it should. but I got one small Issue.

How can I differentiate between home power usage, Net power and Producting Power. I want to separate the power used for home and the power I am sending to the grid but I can't figure out how to get these values.

is there a good documentation which parameter is what physical value ?

I figured out that the Net power is described by the Meter Param [power]. but where do I get the parameter for Home usage ? or The current production Power.

tornadomaus, 2023-07-03:
He PanterSoft,

Same problem. I'm not sure, i will test it the next days.

power_ac = producting power
meter.power = When minus consume power, when plus export power to grid.

So you can calculate it. Take care, there some scale. power_ac_scale and power_scale

` #Inverter power ac in watt
inverter_power = power_ac * (10 ** power_ac_scale)
# consume or export power ac in watt
meter_power = power * (10 ** power_scale)`

Now,

When meter_power is plus, you export power.
meter_power = export power to grid.

Self consumption and home power usage is
`inverter_power_ac - meter_power`

When meter_power is minus.
meter_power = import from grid

home power usage is (minus and minus are plus)
`inverter_power_ac - meter_power`

Self consumption is
`inverter_power_ac`

I hope this helps and that I am correct.

tornadomaus, 2023-07-03:
Sorry, I'm not a pro.
this is my first version.

`inverter_power = power_ac * (10 ** power_ac_scale)
meter_power = power * (10 ** power_scale)

# export
if meter_power > 0:
      import_power = 0
      export_power= meter_power
      self_use_power = inverter_power - meter_power
      home_use_power = self_use_power
elif meter_power < 0:
      import_power = meter_power
      export_power = 0
      self_use_power = inverter_power
      home_use_power = inverter_power - meter_power
elif meter_power == 0:
      print("Waring: Blackout")
`

PanterSoft, 2023-07-04:
Hi @tornadomaus,

I found out that 

Power to Net = meter_values["power"]

corresponds directly to the values in the Solaredge app which means Net power is working fine.

Power to Home = power_dc

Battery Charging Power = 

Power_ac is 0W for me but I am producing over 4Kw but I am using the full power by charging the battery and directly consuming it. so I cannot use this as production power.

I still have to tinker around a bit to really interpret the values correctly.

here is a Screenshot how my current dashboard looks. Ignore the Net,Home,Production Power they are only working if the battery is full. that's the current problem xD

[image]

tornadomaus, 2023-07-05:
Hi @PanterSoft 

I have no battery, so i can't test that.

There are different systems, but can it be that the battery is charged direct with DC power?
So when the battery isn't full, all DC power go directly to the battery and from the battery to home use,
and power_ac will >0 when the battery is full?

Cause power_dc is not my "power to home". On my side, power_dc is the DC power direct from the photovoltaic.

PanterSoft, 2023-07-05:
Hi,

Yes that could be. Unfortunately I have no time the next days but I will eventually find out how it plays together and then I am posting my findings here.

AnneWielis, 2023-11-07:
Does anyone have a clue how to get the actual solar production?

LotadaC, 2023-11-07:
May you explain?
How is your hardware?

Mine should be:

```
stringed module optimizers
   ↓
   ↓ ◄--► BAT-05K48
   ↓
SE8/10K
   ↓
home consumption
   ↑
   ↓
MTR-240-3PC1-D-A-MW
   ↑
   ↓
grid
```
I think (I guess!!) that "I_DC_Power" is the raw output of the optimizer string, pure actual photovoltaic production.
Charging the battery consumes some of this power...
The sum of "I_DC_Power" and "Battery_1_Instantaneous_Power" (positive means charging, negative means discharging) should be the input of the inverter.
"I_AC_Power" should be the output of the inverter
"M_AC_Power" ist the power through the meter.

AnneWielis, 2023-11-08:
I have no idea how the hardware is connected physically, tbh. But yes, looks similar on my end.

So, there are only three meters:
ac_power, meter_power, battery_charge

The rest has to be calculated, right?

LotadaC, 2023-11-08:
Only three meters?

the official description of the SunSpec implementation is [here](https://knowledge-center.solaredge.com/sites/kc/files/sunspec-implementation-technical-note.pdf).
The Registers are listed from page 15ff.
Common Registers -> 40000..40068
Inverter Registers -> 40069..40108
Meter 1 -> 40121..40294
Meter 2 -> 40295..40468
Meter 3 -> 40469..40644
 
extendet Inverter Registers  (Multi-MPPT) -> 40121..40190
(value of 40121=160 -> extendet Inverter, value of 40121=0x0001 -> Meter 1)

 The description of the battery registers is [here](https://www.photovoltaikforum.com/core/attachment/88445-power-control-open-protocol-for-solaredge-inverters-pdf/) (Page 19ff)
Battery 1 -> 57600..57785 (0xE100..0xE1B9)
Battery 2 -> 57856..58041 (0xE200..0xE2B9)

It looks like there are some "undocumented" registers concerning the Home Backup Interface (#71 )

The Registers supported by nmakel's library are defined here:
https://github.com/nmakel/solaredge_modbus/blob/003f29785387424c71067e1291d0d7be5a3ca630/src/solaredge_modbus/__init__.py#L424-L675

40100 = 0x9CA4 is I_DC_Power (DC Power value)
40101 = 0x9CA5 is I_DC_Power_SF (Scale factor)
is here:
https://github.com/nmakel/solaredge_modbus/blob/003f29785387424c71067e1291d0d7be5a3ca630/src/solaredge_modbus/__init__.py#L481-L482