nmakel/solaredge_modbus #89: How is the SolarEdge Wallbox integrated?

LotadaC, 2023-10-22:
Does anybody know, how the SolarEdge EV-Charger (SE-EVK22xxx-01) will be integrated in the SolarEdge infrastructure?

The HomeBatteries are connected via CAN, the meter via RS458 Modbus (or a wireless interface?).
The HomeHubInverter (SE5/8/10K) gathers data and mirrors it in the register-table.

The EV-Charger supports excess charging -> there has to be any communication link between Inverter and EV-Charger.
The EV-Charger has an (W)LAN interface and may be acessed (both directions?) via Modbus TCP

Do the inverter as master connect the charger (as slave) via modbus?
Or the charger (as master) reads the inverter register this way?

The inverter does support only one modbus connection (except the meter-modbus) - if the Charger is integrated,I'm not able to use this project/library??

herbi3, 2023-12-04:
I have tested, and you can still read the data for the inverters, meters and batteries with the EV charger, however I have not yet been successful in reading the data for the EV charger only, and I doubt we ever will to be honest, unless solaredge add in a local api for the devices so we can at least read the data.

The master inverter sees the charger as another follower device. Would be nice to be able to control the charger via modbus

LotadaC, 2023-12-09:
Hi herbi3 (and thank you for replying).
> I have tested, and you can still read the data for the inverters, meters and batteries with the EV charger[...]

I have no hardware yet to test anything.

> however I have not yet been successful in reading the data for the EV charger only, and I doubt we ever will to be honest, unless solaredge add in a local api for the devices so we can at least read [...]

What did you try?

Inverter and wallbox are connected to your LAN, each.
As far as I know, wallboxes are not supported by SunSpec, therefore they will not be readable through the inverter.
The SE-EVK22xxx-01 is a branded/modified/special version of the Keba Contact P30 (C?), did you try to contact it using the [Keba-App](https://play.google.com/store/apps/details?id=com.keba.emobility.app&hl=de&gl=US)?

Documentation for the KEBA wallbox is [here](https://www.keba.com/de/emobility/service-support/downloads/downloads).
The [installation manual (page 41)](https://www.keba.com/download/x/5f05ed5aca/kecontactp30_ihen_web.pdf) shows, that dip switch 1.3 controls the used protocol (UDP or Modbus TCP).

How ist this switch set in your wallbox?

Following this setting, we have to use the programmers guide for [UDP](https://www.keba.com/download/x/4a925c4c61/kecontactp30udp_pgen.pdf) or [Modbus TCP](https://www.keba.com/download/x/dea7ae6b84/kecontactp30modbustcp_pgen.pdf).

If it is Modbus, the "solaredge_modbus"-library could be an... inspiration ;) .

herbi3, 2023-12-09:
Thanks @LotadaC 

I have three 400v solaredge batteries and three inverters, blackout protection for 2/3 phases (we are on three phase supply (240x3, 415v line to line) and have tried testing with the solaredge branded EV charger

Solaredge master/leader support many rs485 connections, and each slave/follower needs a seperate ID for it to work correctly.

However, the master/leader inverter can only handle one TCPmodbus connection at a time, this is only
For reading and controlling data via the master/leader. So I suggest using mproxy (modbus proxy) if you wish to have multiple devices in your LAN that need to read and write data to the inverter. I noticed modbus proxy also helps speed up connections for reading and writing.

Solaredge state in their manuals that, if the EV charger is connected to the same LAN network / same wifi etc, then it will auto detect the inverter and connect to it, I haven't tried that method and only have them connected to the inverter via rs485 instead.

The batteries are connected to the inverter for communication, via the eNet card, where the battery just wirelessly connects to the inverter via radio instead of rs485.