nmakel/solaredge_modbus #95: Beginner question: where can I find a detailed description of modbus fields (e.g. different power values)?

WoslSE, 2024-02-03:
I had installed the Modbus TCP setup and can run the Python programs (e.g. example.py) successfully. This is all good so far. As a beginner, now I need some assistance to find the appropriate power values. Where can I find a more detailed description of the listed fields?

Especially, I´m looking for the following power values:
- produced power by PV in [W]
- consumed power in [W]
- delivered power (to the grid) in [W]
- supplied power (from the grid) in [W]
- charged power (battery) in [W]
- discharged power (battery) in [W]

I tried to match the Modbus data (from example.py) with those delivered by the Solaredge portal but they don't fit as expected.

Sorry for that simple question.

Regards

nmakel, 2024-02-12:
Some of the values you want can be found in the data (use `--json` when running `example.py`). For example, _produced power by PV_ is `energy_total` in Wh. The battery also reports total energy consumed (by charging) and delivered (by discharging) as `lifetime_export_energy_counter` and `lifetime_import_energy_counter`.

The other values need to be computed by yourself, as you'll need a kWh meter at the grid connection point.

WoslSE, 2024-02-22:
Thank you for your response. 

I tested the JSON output format and filtered out fields from Modbus request. With the fetched fields you can calculate the other values later.

I hope I'm doing it in a correct way and other readers can learn from it. In case I'm wrong please let me know.
 
Regards

        c_time = json_body[0]['time']
        c_inverter_power_ac = json_body[0]['fields']['power_ac']
        c_meter_power = json_body[1]['fields']['power']
        if lBattery == True:
            b_available_energy = json_body[2]['fields']['available_energy']
            b_instantaneous_power = json_body[2]['fields']['instantaneous_power']
            Battery_Power_W = float(b_instantaneous_power)
            Work_Storage_kWh = float(b_available_energy)

closed 2024-02-22 by WoslSE