nmakel/solaredge_modbus #96: KeyError "Status"

infoedkok, 2024-02-19:
Hi, After upgrading Domoticz to 2024-1 (docker container) and installing the plugin from version 1.1.1 and solaredge_modbus-0.8.0 I have these errors:
2024-02-13 15:27:22.544 Error: SolarEdgeMODBUS: Call to function 'onHeartbeat' failed, exception details:
2024-02-13 15:27:22.544 Error: SolarEdgeMODBUS: Traceback (most recent call last):
2024-02-13 15:27:22.544 Error: SolarEdgeMODBUS: File "/opt/domoticz/userdata/plugins/domoticz-solaredge-modbustcp-plugin/plugin.py", line 544, in onHeartbeat
2024-02-13 15:27:22.544 Error: SolarEdgeMODBUS: _plugin.onHeartbeat()
2024-02-13 15:27:22.544 Error: SolarEdgeMODBUS: File "/opt/domoticz/userdata/plugins/domoticz-solaredge-modbustcp-plugin/plugin.py", line 352, in onHeartbeat
2024-02-13 15:27:22.544 Error: SolarEdgeMODBUS: to_lookup = int(inverter_values[unit[Column.MODBUSNAME]])
2024-02-13 15:27:22.544 Error: SolarEdgeMODBUS: KeyError: 'status'

'Status' looks like one of the 22 items that are passed. 
This status reports on what the inverter is doing.

"KeyError" message is generated by in ---init---.py"!

Here in ---init---.py there's a list of keys for Status which refers to INVERTER_STATUS_MAP, line 110:
INVERTER_STATUS_MAP = [
"Undefined",
"Off",
"Sleeping",
"Grid Monitoring",
"Producing",
"Producing (Throttled)",
"Shutting Down",
"Fault",
"Standby"
]

I think that this list is missing some status and that causes this message?

The errors appear randomly :-(

Versions after pip3 install -r requirements.txt :
Requirement already satisfied: solaredge_modbus==0.7.0 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 1)) (0.7.0)
Requirement already satisfied: pymodbus>=2.4.0 in /usr/local/lib/python3.9/dist-packages (from solaredge_modbus==0.7.0->-r requirements.txt (line 1)) (3.6.4)

Any idea what status of the inverter could cause this?

Regards, Ed

OlmZone, 2024-05-31:
Hi,
I experience the same errors, that keep on going at random intervals:
 2024-05-31 09:36:06.614 Error: SolarEdge: Call to function 'onHeartbeat' failed, exception details:
2024-05-31 09:36:06.616 Error: SolarEdge: Traceback (most recent call last):
2024-05-31 09:36:06.616 Error: SolarEdge: File "/home/pi/domoticz/plugins/domoticz-solaredge-modbustcp-plugin/plugin.py", line 544, in onHeartbeat
2024-05-31 09:36:06.616 Error: SolarEdge: _plugin.onHeartbeat()
2024-05-31 09:36:06.616 Error: SolarEdge: File "/home/pi/domoticz/plugins/domoticz-solaredge-modbustcp-plugin/plugin.py", line 352, in onHeartbeat
2024-05-31 09:36:06.616 Error: SolarEdge: to_lookup = int(inverter_values[unit[Column.MODBUSNAME]])
2024-05-31 09:36:06.616 Error: SolarEdge: KeyError: 'status' 

(plugin version 1.1.1 and solaredge_modbus-0.7.0, not in docker-container.)
Although it seems to work pretty well, it looks ugly in my error log.
If I can help by providing more info or maybe testing, let me know.

Regards,
Jos

averter, 2024-06-30:
Hello both. I've had a similar issue in a different project. Perhaps my findings will help https://github.com/saidlm/Domoticz-Solax-plugin/issues/3

infoedkok, 2024-09-04:
Averter mentioned in this link https://github.com/saidlm/Domoticz-Solax-plugin/issues/3 a possible cause and solution:
_it has to do with pymodbus version. If the version starts with 3.3 or 3.0 it ought to be lowercase big and little, otherwise it needs to be uppercase BIG and LITTLE. See an implementation [line in HA code](https://github.com/wills106/homeassistant-solax-modbus/blob/4647ba453a4f088b66b8202f4ace42b8baad005f/custom_components/solax_modbus/__init__.py#L43)_

In this case, in __init__.py of the solaredge module, it is in uppercase. Pymodbus installed inside my domoticz is 2.4.3. 
So uppercase should be fine (?)

infoedkok, 2024-09-09:
Did some more debugging and I have an idea. It seems to me some sort of timing issue.

 In the domoticz.log I can see  regular OnHeartBeat loops where in milliseconds after the call "OnHeartBeat" the inverter values are ALL retrieved. 

The errors occures when the retrieving of the invertervalues happenes seconds after the OnHeartBeat call. Example:
2024-09-09 22:14:32.063  SolarEdgeMODBUS: onHeartbeat
.. lots of other stuff..
2024-09-09 22:15:16.672  SolarEdgeMODBUS: inverter values: {
    "commit_power_control_settings": 0,
    "restore_power_control_default_settings": 0,
    "reactive_power_config": 4,
    "reactive_power_response_time": 200,
    "advanced_power_control_enable": 1,
    "export_control_mode": 0,
    "export_control_limit_mode": 0,
    "export_control_site_limit": 0
}
So there is a "gap" of almost half a minute in this case!

Notice that the list of the values a very short. In a regular loop, this list a much longer.
And because the "status" value is not in the shortened list, it can not be matched within the defined tablestructure in the plugin (e.g line 201 for the THREE_PHASE_INVERTER .
As you can see, the "status" is the first record in this table structure.
Since it is not in the shortened list above, it is generating the error.

So the question is: why is the plugin sometimes waiting after the OnHeartBeat call?