nmakel/solaredge_modbus #106: Modbus reading meter 0kW and current L1 L2 L3 0A in the sleeping mode

twejrzan, 2024-12-06:
I have three phase inverter SolarEdge - when inverter is going to "Seeping mode" 
the meters.Meter1.power = -32768
also inverter current L1 L2 L3 are not read properly

herbi3, 2024-12-06:
Are you reading and using the scale factor for power and current?

twejrzan, 2024-12-06:
Yes, I am using the reading factor, but this is not a problem of scaling. For the "Production mode" it works correctly

twejrzan, 2024-12-06:
Maybe this a problem of new Solaredge firmware because few months ago it was everything OK

herbi3, 2024-12-06:
Do you have more data available? What’s the setup? Where are the CTs? - generally, the meter doesn’t care what the inverter is doing so I’m curious to see what the issue is here.I have three inverters and haven’t had this same issue. You can check it out hereGrafanapublic.winterfell.tv

twejrzan, 2024-12-06:
{
    "c_id": "SunS",
    "c_did": 1,
    "c_length": 65,
    "c_manufacturer": "SolarEdge",
    "c_model": "SE10K-RWS48BEN4",
    "c_version": "0004.0022.0039",
    "c_serialnumber": "7E068E9A",
    "c_deviceaddress": 1,
    "c_sunspec_did": 103,
    "c_sunspec_length": 50,
    "current": 0,
    "l1_current": 0,
    "l2_current": 0,
    "l3_current": 0,
    "current_scale": -2,
    "l1_voltage": 4072,
    "l2_voltage": 4074,
    "l3_voltage": 4079,
    "l1n_voltage": 2342,
    "l2n_voltage": 2352,
    "l3n_voltage": 2367,
    "voltage_scale": -1,
    "power_ac": 0,
    "power_ac_scale": 0,
    "frequency": 5003,
    "frequency_scale": -2,
    "power_apparent": 0,
    "power_apparent_scale": 0,
    "power_reactive": 0,
    "power_reactive_scale": 0,
    "power_factor": 0,
    "power_factor_scale": 0,
    "energy_total": 14780800,
    "energy_total_scale": 0,
    "current_dc": 0,
    "current_dc_scale": 0,
    "voltage_dc": 8,
    "voltage_dc_scale": -1,
    "power_dc": 0,
    "power_dc_scale": 0,
    "temperature": 1838,
    "temperature_scale": -2,
    "status": 2,
    "vendor_status": 0,
    "rrcr_state": 0,
    "active_power_limit": 100,
    "cosphi": 0,
    "commit_power_control_settings": 0,
    "restore_power_control_default_settings": 0,
    "reactive_power_config": 0,
    "reactive_power_response_time": 200,
    "advanced_power_control_enable": 0,
    "export_control_mode": 0,
    "export_control_limit_mode": 0,
    "export_control_site_limit": 0,
    "meters": {
        "Meter1": {
            "c_manufacturer": "SolarEdge",
            "c_model": "SE-RGMTR-1D-240C-A",
            "c_option": "Export+Import",
            "c_version": "0",
            "c_serialnumber": "0",
            "c_deviceaddress": 2,
            "c_sunspec_did": 201,
            "c_sunspec_length": 105,
            "current": -32768,
            "l1_current": -32768,
            "l2_current": -32768,
            "l3_current": -32768,
            "current_scale": -32768,
            "voltage_ln": -32768,
            "l1n_voltage": -32768,
            "l2n_voltage": -32768,
            "l3n_voltage": -32768,
            "voltage_ll": -32768,
            "l12_voltage": -32768,
            "l23_voltage": -32768,
            "l31_voltage": -32768,
            "voltage_scale": -32768,
            "frequency": -32768,
            "frequency_scale": -32768,
            "power": -32768,
            "l1_power": -32768,
            "l2_power": -32768,
            "l3_power": -32768,
            "power_scale": -32768,
            "power_apparent": -32768,
            "l1_power_apparent": -32768,
            "l2_power_apparent": -32768,
            "l3_power_apparent": -32768,
            "power_apparent_scale": -32768,
            "power_reactive": -32768,
            "l1_power_reactive": -32768,
            "l2_power_reactive": -32768,
            "l3_power_reactive": -32768,
            "power_reactive_scale": -32768,
            "power_factor": -32768,
            "l1_power_factor": -32768,
            "l2_power_factor": -32768,
            "l3_power_factor": -32768,
            "power_factor_scale": -32768,
            "export_energy_active": 0,
            "l1_export_energy_active": 0,
            "l2_export_energy_active": 0,
            "l3_export_energy_active": 0,
            "import_energy_active": 0,
            "l1_import_energy_active": 0,
            "l2_import_energy_active": 0,
            "l3_import_energy_active": 0,
            "energy_active_scale": 0,
            "export_energy_apparent": 0,
            "l1_export_energy_apparent": 0,
            "l2_export_energy_apparent": 0,
            "l3_export_energy_apparent": 0,
            "import_energy_apparent": 0,
            "l1_import_energy_apparent": 0,
            "l2_import_energy_apparent": 0,
            "l3_import_energy_apparent": 0,
            "energy_apparent_scale": -32768,
            "import_energy_reactive_q1": 0,
            "l1_import_energy_reactive_q1": 0,
            "l2_import_energy_reactive_q1": 0,
            "l3_import_energy_reactive_q1": 0,
            "import_energy_reactive_q2": 0,
            "l1_import_energy_reactive_q2": 0,
            "l2_import_energy_reactive_q2": 0,
            "l3_import_energy_reactive_q2": 0,
            "export_energy_reactive_q3": 0,
            "l1_export_energy_reactive_q3": 0,
            "l2_export_energy_reactive_q3": 0,
            "l3_export_energy_reactive_q3": 0,
            "export_energy_reactive_q4": 0,
            "l1_export_energy_reactive_q4": 0,
            "l2_export_energy_reactive_q4": 0,
            "l3_export_energy_reactive_q4": 0,
            "energy_reactive_scale": -32768
        }
    },
    "batteries": {
        "Battery1": {
            "c_manufacturer": "SolarEdge",
            "c_model": "SolarEdge Home Battery 9.2kWh",
            "c_version": "48V DCDC 2.3.3",
            "c_serialnumber": "7E068E9A",
            "c_deviceaddress": 112,
            "c_sunspec_did": 0,
            "rated_energy": 9200.0,
            "maximum_charge_continuous_power": 5000.0,
            "maximum_discharge_continuous_power": 5000.0,
            "maximum_charge_peak_power": 4800.0,
            "maximum_discharge_peak_power": 8150.0,
            "average_temperature": 14.0,
            "maximum_temperature": 0.0,
            "instantaneous_voltage": -0.030463047325611115,
            "instantaneous_current": 0.0,
            "instantaneous_power": 0.0,
            "lifetime_export_energy_counter": 0,
            "lifetime_import_energy_counter": 0,
            "maximum_energy": 9200.0,
            "available_energy": 8939.51953125,
            "soh": 97.0,
            "soe": 5.462052854454669e-07,
            "status": 10,
            "status_internal": 1,
            "event_log": 0,
            "event_log_internal": 0
        }
    }
}

twejrzan, 2024-12-06:
when I login directly to the inverter by SetApp it shows the data properly.

herbi3, 2024-12-06:
Yeah that’s not quite right. It’s showing a status of 10 for the battery too, never seen that but in saying that, I only have 400v batteries not 48v.Have your tried to read the data from another device to eliminate any weird settings or changes on your local machine?

twejrzan, 2024-12-06:
I use workaround solution: if the inverter status is 2 (Sleeping mode) I use OPENHAB Solaredge binding and it reads the values correctly, but the min refresh time is 1 min

twejrzan, 2024-12-06:
Battery status 10 means "Saving Energy";)

herbi3, 2024-12-06:
Ah right, mine only goes up to “6” which is holding.With the latest document released in April 2024 (as far as I can find at least) there wasn’t any significant changes to the modbus registers.I did however notice that, when the grid itself goes down and the backup interface kicks in to go off grid, that’s the only time I notice the meter scales go into negative.Is there anything else in the middle, a modbus proxy or reading direct? And are you able to try reading the data from a different machine with a clean solaredge modbus library?

twejrzan, 2024-12-06:
Thank you for your comments.
Unfortunatelly, this is the only Solaredge inverter I have.
I will try to observe the issue and let you know if find something.

herbi3, 2024-12-06:
Ah yeah, apologies, I mean to read the data from another computer or virtual environment with a clean solaredge modbus install

twejrzan, 2024-12-06:
I have just installed solaredge_modbus at my Windows PC. It gives the same data.