nmakel/solaredge_modbus #107: Error on running example.py

puschen66, 2024-12-26:
Hi,
I'm completely new here and I've a problem on running example.py
Who can help me?
Thanks in advance.

**I'm using:**
debian 12.8
python 3.11
solaredge_modbus 0.8.0
pymodbus 3.8.0 

**Command:**
/usr/bin/python3.11 example.py 192.168.178.50 1502

**Error-Message:**
Traceback (most recent call last):
  File "/user/python/SolarEdge/example.py", line 26, in <module>
    values = inverter.read_all()
             ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/solaredge_modbus/__init__.py", line 420, in read_all
    results.update(self._read_all(register_batch, rtype))
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/solaredge_modbus/__init__.py", line 356, in _read_all
    data = self._read_holding_registers(offset, length)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/solaredge_modbus/__init__.py", line 244, in _read_holding_registers
    result = self.client.read_holding_registers(address, length, slave=self.unit)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given

stefanosonzo, 2025-01-03:
try installing pymodbus version 3.6.2 with

  pip uninstall pymodbus

and then
  
  pip install pymodbus==3.6.8

puschen66, 2025-01-03:
Yes, that's it.
Thank you very much.

closed 2025-01-03 by puschen66