nmakel/solaredge_modbus #110: Duplicated registers in Inverter and StorEdge

milkotodorov, 2025-01-15:
The following registers in [`Inverter`](https://github.com/nmakel/solaredge_modbus/blob/master/src/solaredge_modbus/__init__.py#L559-L567) class and the ones in [`StorEdge`](https://github.com/nmakel/solaredge_modbus/blob/master/src/solaredge_modbus/__init__.py#L702-L716) class are duplicating each other (almost all of them).

Are there any good reason for that?

herbi3, 2025-01-15:
It is a fix to prior to the little endian map 

Now there's the little endian map for the items that require it under inverter.

herbi3, 2025-01-15:
The StorEdge class isn't needed now, as the registers that require little endian under the Inverter class now have a lookup map which automatically uses little endian for the registers that need it under Inverter class

milkotodorov, 2025-01-16:
I see...the StorEdge was like a workaround until there was a way that these registers be part of the `Inverter` class and now no longer needed.

We have however the `0xe000: export_control_mode`, `0xe001: export_control_limit_mode` and `0xe002: export_control_site_limit` not part of the `Inverter` class.

Why not make them part of the `Inverter` class and remove the `StorEdge` one?

herbi3, 2025-01-16:
It looks like they're still there?[image][image][image]

milkotodorov, 2025-01-16:
These ones: https://github.com/nmakel/solaredge_modbus/blob/master/src/solaredge_modbus/__init__.py#L703-L705 - they are not part of the `Inverter` class.

milkotodorov, 2025-01-16:
Actually they are there, but with different offsets: https://github.com/nmakel/solaredge_modbus/blob/master/src/solaredge_modbus/__init__.py#L555-L557

milkotodorov, 2025-01-16:
Then shall we remove the `StorEdge` class to not cause confusion, when not needed?

herbi3, 2025-01-16:
It has zero effect on functionality. You can use either one. It's easier to use the inverter one to call everything at once. This is someone else pull request and ultimately it can be removed should the author be ok with that

LotadaC, 2025-03-09:
Only the last nine registers in the StoreEdge-Class are the Store Edge Block
in inverter-class
https://github.com/nmakel/solaredge_modbus/blob/fd3ce7ae32a259ee371c672dac3bcd75bfe51258/src/solaredge_modbus/__init__.py#L559-L567
 the other three are part of the Export Limit Control Block, correct?
(but they need the same endianess?)

What about the "External Production Max Power" Register (0xe012)?
https://github.com/nmakel/solaredge_modbus/blob/master/documents/application_note_power_control_configuration-v1.3.pdf page 16