nmakel/solaredge_modbus #114: User Request: add function to change the active power limit

infoedkok, 2025-04-04:
So far, the integration can (only?) read the solar edge registers.

I like to change the active-power-limit register with a Domoticz script e.g from 100% to 80%. 
Ofcourse, already tried to update the device in domoticz with a script, but that does not work.

I already tested this function in home assistent and there it works fine.

Please add this for use in domoticz.

herbi3, 2025-04-04:
Instead of using something like 'inverter.read(xxx)' you can use 'inverter.write(command, value)

frepke, 2025-04-04:
> So far, the integration can (only?) read the solar edge registers.
>
> I like to change the active-power-limit register with a Domoticz script e.g from 100% to 80%.
> Ofcourse, already tried to update [...]

Just out of curiosity, what's the point of switching the active power limit to 80% ?

herbi3, 2025-04-04:
Agreed, the power limit % isn't ideal. Maybe the export or production limit is a better setting in my opinion

infoedkok, 2025-04-04:
Just an example. I want to calculate the value based on some other domoticz devices. Typically, the amount of produced solar energy related to the mean price and other power consumption at a given moment.

infoedkok, 2025-04-04:
The solar edge app supports "tarif optimalisation" based on dynamic prices, but that's not what I am looking for.

frepke, 2025-04-04:
> Just an example. I want to calculate the value based on some other domoticz devices. Typically, the amount of produced solar energy related to the mean price and other power consumption at a given moment.
> 

Okay, so it has nothing to do with a dynamic energy contract or so?

herbi3, 2025-04-04:
I've done something similar for wholesale energy here that changes every 5 minutes. It automatically charges the batteries from set times based on how much is needed via the duration, along with calculating the source of energy for consumption used.

This is built with the amber energy api for the pricing

Https://public.winterfell.tv/

frepke, 2025-04-04:
> The solar edge app supports "tarif optimalisation" based on dynamic prices, but that's not what I am looking for.

Okay, that's clear.

infoedkok, 2025-04-04:
Dynamic pricing is just one parameter I like to use in the calculation. Use case is to maximize returned solar power considering the use of power in my house.  E.g when  I charge my EV, running the dishwash, dryer etc during overproduction periodes.
And, I 've heard that some dynamic energy providers want to change the period of calculations: from every one hour to every 15 minutes. That requests an automatic procedure

herbi3, 2025-04-04:
For using the % value, that's only limiting the inverter to X percent of the inverters maximum rated output. 10kw inverter with an 80% limit will only allow 8kw to be produced. I could be missing something in your use case here, but currently I can't see what is useful by using the % setting.

Nonetheless, did the write command above work for you?

infoedkok, 2025-04-04:
@herbi3 I will have look! Thank you!